particle-iot/device-os #1793: [gen 3] The use of Serial2 on the Xenon prevents it from achieving lowest power sleep.

iitgrad, 2019-05-24:
I believe I have found a sleep power bug on the xenon when using Serial2 against 1.1.0-rc.2.  It is also in 1.2 beta 1.  If you fire up Serial 2 and then shut it down, the Xenon no longer can sleep at 43uA, it jumps to 95 uA. But this works properly with Serial and Serial1.

## Bug Report

### Expected Behavior
You should see approximately 2.3mA draw for the first 5 seconds, then 43uA during the first 10 second sleep, then after firing up Serial1 and shutting it back down, and returning back to sleep, you should be back to 43uA draw for 10 seconds. And finally after wake we fire up Serial2 and shut it down (note just like Serial1) and you should see another sleep draw of 43uA.

### Observed Behavior

The actual draw after Serial2 has been fired up and shutdown causes the Xenon to only achieve 95uA draw during sleep. 

### Steps to Reproduce

Code is supplied below to reproduce. Reproduces on 1.1.0-rc.2 and 1.2.0-beta 1.

### Test App

```c
SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

#include "Serial2/Serial2.h"

void setup() {
  delay(5000); // looks normal at around 2.3mA
  System.sleep(D2, RISING, 10);  // looks good at 43uA
  Serial1.begin(9600);
  Serial1.end();
  System.sleep(D2, RISING, 10);  // still looks good at 43uA
  Serial2.begin(9600);
  Serial2.end();
  System.sleep(D2, RISING, 100);  // sleep is up to 95uA
}

void loop() {
}
```

technobly, 2019-05-25:
@iitgrad Hi Kevin, thanks for the issue report.  I cannot confirm the issue.  I did some tests on my Xenon powered via LiPo battery only, routed through a EEVblog uCurrent Gold current measurement device.  I tested 1.1.0, 1.2.0-beta.1 and 1.2.1-rc.1 and all three are doing the same thing:
- Many times I see ~47uA for all 3 cases
- Many times I see the last Sleep call fail to stay asleep and the current rise to ~3-4mA.  
This last case seems related to #1789 so I'm inclined to say there's no specific problem with Serial2, but rather an issue with the sleep API.  There may be something unique with your Xenon as well, so perhaps trying this on multiple Xenons would be worthwhile.

iitgrad, 2019-06-07:
Hi @technobly.  We are doing much more testing here in the lab and with multiple Xenons.  So far we are replicating the sleep issue.  We will try to get more info to you today on exact configuration and video of what we are seeing.

awkimball, 2019-06-10:
Hi @technobly. After some additional testing on two different Xenon's on 1.1.0, 1.2.0-beta.1 and 1.2.1-rc.2 Kevin and I are still able to duplicate the sleep current issue. Like you said, we get an intermittent wakeup related to #1789 sometimes, however even when this wakeup doesn't happen we see ~43uA during the first sleep and anywhere from 50-250uA during the subsequent sleeps (after Serial1/2 have been enabled and disabled) [Video](https://drive.google.com/open?id=1woQsrBakMrdVk8I_-XcpRnxt650Rq9xb) For reference, I used the code snippet that is posted above and we have tried a number of supply voltages (3.7, 4.2, 6.0V)

avtolstoy, 2020-09-23:
Closing as Xenon is discontinued: https://docs.particle.io/reference/discontinued/mesh/

Other platforms should have significant improvements in sleep reliability in 2.x DeviceOS releases.

closed 2020-09-23 by avtolstoy