particle-iot/device-os #1805: Hard fault SOS in WiFi.localIP() when system thread is connecting to WiFi

elcojacobs, 2019-06-04:
<details>
  <summary><i>Getting the local IP address while the system thread is connecting to WiFi in the background causes a hard fault SOS</i></summary>
</details>

## Bug Report
It seems that the WiFi functions are not thread-safe. This very simple app will cause an instant hard fault SOS on a photon. Compiled with device-os 1.2.1-rc.2.

I suspect a pointer to a network interface becomes dangling in the user thread because of network changes in the system thread. Why are you not using smart pointers for most of this?

### Expected Behavior

Print 0.0.0.0 to the char[16] array.

### Observed Behavior

Hard fault SOS.
A noteworthy observation is that when the setup button is pressed during boot and released right afterward, the SOS does not occur.

### Steps to Reproduce

Compile app below for a photon. Flash and get an instant SOS.

### Test App

```c
/* Includes ------------------------------------------------------------------*/
#include "application.h"

SYSTEM_THREAD(ENABLED);
SYSTEM_MODE(SEMI_AUTOMATIC);

int buzz = A2;

void
printWiFiIp(char dest[16])
{
    IPAddress ip = spark::WiFi.localIP();
    snprintf(dest, 16, "%d.%d.%d.%d", ip[0], ip[1], ip[2], ip[3]);
}

void
setup()
{
    pinMode(buzz, OUTPUT);
    digitalWrite(buzz, HIGH);
    delay(200);
    digitalWrite(buzz, LOW);
}

void
loop()
{
    if (spark::WiFi.ready() && !spark::WiFi.listening()) {
        Particle.connect();
    } else if (!spark::WiFi.connecting()) {
        spark::WiFi.connect(WIFI_CONNECT_SKIP_LISTEN);
    }

    char ip[16];
    printWiFiIp(ip);
}
```

elcojacobs, 2019-06-04:
As a workaround, it might be a good idea to switch to system events so the WiFi functions are not used in the user thread.
Example:
```
/* Includes ------------------------------------------------------------------*/
#include "application.h"
#include <vector>

SYSTEM_THREAD(ENABLED);
SYSTEM_MODE(SEMI_AUTOMATIC);

int buzz = A2;

IPAddress ip = uint32_t(0);
bool connected = false;

void
printWiFiIp(char dest[16])
{
    snprintf(dest, 16, "%d.%d.%d.%d", ip[0], ip[1], ip[2], ip[3]);
}

std::vector<system_event_t> events;

void
handleNetworkEvent(system_event_t event, int param)
{
    switch (param) {
    case network_status_connected:
        ip = WiFi.localIP();
        connected = true;
        Particle.connect();
        break;
    default:
        ip = uint32_t(0);
        connected = false;
        break;
    }
    events.push_back(param);
}

void
setup()
{
    pinMode(buzz, OUTPUT);
    digitalWrite(buzz, HIGH);
    delay(200);
    digitalWrite(buzz, LOW);
    System.on(network_status, handleNetworkEvent);
    Serial.begin();
    WiFi.connect(WIFI_CONNECT_SKIP_LISTEN);
}

void
loop()
{
    char ipString[16];
    printWiFiIp(ipString);
    Serial.println();
    Serial.print("IP: ");
    Serial.println(ipString);
    delay(5000);
    for (auto& event : events) {
        Serial.print(uint32_t(event));
        Serial.print(',');
    }
}
```

I'll try porting my app to events to workaround this major bug.

elcojacobs, 2019-06-04:
As a possible workaround, I wanted to try to only fetch the IP on the system thread by using a system event handler.
But strangely enough, the network_status_connected event never occurs.
The events since boot are:
32,32,32,32,64,64,16384

So cloud_status (64) occurs, but 

```c
#include "application.h"
#include <vector>

SYSTEM_THREAD(ENABLED);
SYSTEM_MODE(SEMI_AUTOMATIC);

int buzz = A2;

IPAddress ip = uint32_t(0);
bool connected = false;

void
printWiFiIp(char dest[16])
{
    snprintf(dest, 16, "%d.%d.%d.%d", ip[0], ip[1], ip[2], ip[3]);
}

std::vector<system_event_t> events;

void
handleNetworkEvent(system_event_t event)
{
    switch (event) {
    case network_status_connected:
        ip = WiFi.localIP();
        connected = true;
        break;
    case network_status_powering_on:
    case network_status_on:
    case network_status_powering_off:
    case network_status_off:
    case network_status_connecting:
        ip = uint32_t(0);
        connected = false;
        break;
    }
    events.push_back(event);
}

void
setup()
{
    pinMode(buzz, OUTPUT);
    digitalWrite(buzz, HIGH);
    delay(200);
    digitalWrite(buzz, LOW);
    System.on(all_events, handleNetworkEvent);
    Serial.begin();
}

void
loop()
{
    if (WiFi.ready() && !WiFi.listening()) {
        Particle.connect();
    } else if (!WiFi.connecting()) {
        WiFi.connect(WIFI_CONNECT_SKIP_LISTEN);
    }

    char ipString[16];
    printWiFiIp(ipString);
    Serial.println();
    Serial.print("IP: ");
    Serial.println(ipString);
    delay(5000);
    for (auto& event : events) {
        Serial.print(uint32_t(event));
        Serial.print(',');
    }
}
```

Serial output:
```
[13:10:44:498] 32,32,32,32,64,64,16384,␍␊
[13:10:44:498] IP: 0.0.0.0␍␊
```

avtolstoy, 2020-03-03:
Resolved in https://github.com/particle-iot/device-os/pull/1949

closed 2020-03-03 by avtolstoy