particle-iot/device-os #1809: [Gen3] I2C reads back address byte

ScruffR, 2019-06-07:
As discussed in this thread
https://community.particle.io/t/i2c-invalid-reads/50214

and this test code
```
#define MASTER 0
#define speed 100000
#define SLAVE_ADDRESS 0x18

#if MASTER
void setup() {
    Serial.begin(115200);
    Particle.function("sendData", sendData);
    Wire.setSpeed(speed);
    Wire.begin();
}

int sendData(String cmd) {
   for (int i = 0; i < 1; ++i) {
        uint8_t count = Wire.requestFrom(SLAVE_ADDRESS, 6);
        // delay(10); // doesn't change anything
        while(Wire.available()) {
            char c = Wire.read();
            Serial.print(c);
        }
        Serial.println();
    }
    return 0;
}

#else 
void requestEvent() {
    Wire.print("456789");
}

void setup() {  
    Wire.setSpeed(speed);
    Wire.begin(SLAVE_ADDRESS);    
    Wire.onRequest(requestEvent);
}
#endif
```

I could confirm that with the second call of the `sendData` function the data received by the master (if master is Argon and slave is Photon- probably true for other Gen3 devices as master too) will be "prepended" with `SLAVE_ADDRESS << 1 & 0x01`.
When swapping roles (Photon=master, Argon=slave) the master receives the correct data.

perotom, 2019-06-10:
Possibly the same problem as here: https://github.com/particle-iot/nrf5_sdk/pull/9

YutingYou, 2019-06-10:
@perotom  Thanks for digging into this issue 👍. I also did some tests:

Test case 1: Argon(Master) - Xenon(Slave)
[image]

Test case 2: Photon(Master) - Photon(Slave)
[image]

Test case 3: Argon(Master) - Photon(Slave)
[image]

From the Logic Analyser(Test case 3), we can find there is no too long or too short clock pulse as is in the case you provided [i2c-clock-stretching-on-nrf52840](https://infocenter.nordicsemi.com/index.jsp?topic=%2Ferrata_nRF52832_Rev2%2FERR%2FnRF52832%2FRev2%2Flatest%2Fanomaly_832_149.html&anchor=anomaly_832_149), so it might be not the same issue, and software TWI driver `twi_sw_master.c` is deprecated, we're using the hardware TWI driver.

From Test case 3, we can see the extra byte is sent from Gen2 device, I'll continue to look into this issue, if you have any other new discoveries, you can leave it here. Thank you!

perotom, 2019-06-10:
From my side I can tell it happens in about 0.1% of requests, so you might need to write your own test script to see the effect (We did a loop where we read on byte every 100ms and compare it to the previously read value). The Electron doesn't suffer from this behaviour.

avtolstoy, 2019-10-03:
Resolved in #1829

closed 2019-10-03 by avtolstoy