particle-iot/device-os #1813: [Xenon] SEMI_AUTOMATIC mode blocks after calling Mesh.off()

ScruffR, 2019-06-09:
After the first publish the code never runs `loop()` again.

```
SYSTEM_MODE(SEMI_AUTOMATIC);

void setup() {
  Serial.begin();
}

void loop() {
  static uint32_t msLast = 0;
  Serial.print('.');  // get only printed on first call to loop()
  if (millis() - msLast < 1000) return;
  msLast = millis();

  Serial.println("Turning ON radio...");
  Mesh.on();
  Mesh.connect(); 
  if (waitFor(Mesh.ready, 10000)) {
    Particle.connect();
    if (waitFor(Particle.connected, 10000)) {
      Particle.publish("xenon-test", PRIVATE);
      Serial.println("published xenon-text");
      delay(100);
    }
    else
      Serial.println("Couldn't connect to cloud");
  }
  else
    Serial.println("Couldn't connect to mesh");
  
  Serial.println("Turning OFF radio...");
  Mesh.off();
  Serial.println("radio is off");
}
```

This works in `SYSTEM_MODE(AUTOMATIC)` tho'
Tested with 1.1.0 and 1.2.1-rc.2 (Xenon + Argon 1.1.0 gateway)

YutingYou, 2019-07-16:
Hi @ScruffR, thanks for reporting this issue, it has been fixed in #1857 .

avtolstoy, 2019-07-31:
Closing as PR #1857 has been merged.

closed 2019-07-31 by avtolstoy