particle-iot/device-os #1814: Mesh.on is a no-op

tve, 2019-06-09:
## Bug Report

### Expected Behavior

`Mesh.on()` should enable the mesh radio, especially after `Mesh.off()` has been called.

### Observed Behavior

`Mesh.on()` has no effect, because `OpenThreadNetif::powerUp()` does nothing. Yet, `OpenThreadNetif::powerDown()` calls `down()`. See https://github.com/particle-iot/device-os/blob/develop/hal/network/lwip/openthread/lwip_openthreadif.cpp#L868-L874

### Steps to Reproduce

Build 1.3.0_alpha for a mesh device with debug tracing enabled, call Mesh.off(), then Mesh.on(), observe that the thread interface does not get restarted.

tve, 2019-06-09:
After a pile of troubleshooting, I see that cal;ling `Mesh.connect()` will enable the mesh interface and end up connecting. This is confusing because users call `Mesh.off()` to turn the mesh radio off to save power and then expect `Mesh.on()` to "revert" the operation, i.e. to turn Mesh back on. Maybe things should be documented better if this is how you expect it to work? See https://community.particle.io/t/xenon-not-publishing-when-using-mesh-off-mesh-on/50208

kubark42, 2020-02-28:
Assuming that the method @tve linked to `Mesh.on()` is `powerUp()`, then the problem he reports still seems to be unresolved:

```
int OpenThreadNetif::powerUp() {
    return 0;
}
```
(https://github.com/particle-iot/device-os/blob/v1.5.0-rc.1/hal/network/lwip/openthread/lwip_openthreadif.cpp#L800)

Since Mesh is deprecated, it's unlikely this will ever be resolved. However, it would be really nice if the documentation at https://docs.particle.io/reference/device-os/firmware/xenon/#on- reflected this.

closed 2020-03-23 by avtolstoy

kubark42, 2020-03-30:
@avtolstoy when you get a chance could you update this ticket with why it was closed? In particilar, was it resolved, or is it being deprioritized for some other reason, e.g. because of mesh deprecation?

avtolstoy, 2020-03-30:
Thanks for the ping.

`Mesh.on()` being a no-op is not the problem really (e.g. Ethernet netif driver also doesn't actually do anything in `powerUp()`). In practice this is a duplicate of #1631 that we are tracking and is the current behavior on Gen 3 devices, where each network interface is managed manually. Please follow that issue for updates.

In order to replicate the behavior of Gen 2 `Network.on()` / `Network.off()` may be used.