particle-iot/device-os #1815: Question of "style" Particle.h vs. application.h

ScruffR, 2019-06-10:
e.g. the demo implementations in this sub-directory
https://github.com/particle-iot/device-os/tree/v1.3.0-alpha.1/user/tests/app/ble
or some code snippets in the docs and tutorials still use `#include "application.h"`.

Although `Particle.h` (formerly `Spark.h`) should have replaced `application.h` (as per #249) we keep finding internal/official code using `application.h` while we try to get the community to move from "deprecated"/outdated syntax to the "modern" terminology.

rickkas7, 2019-06-10:
This doesn't answer your question, however when I converted the test applications into examples in the docs, I changed everything to use Particle.h instead.

technobly, 2019-06-10:
I agree they should all be using `Particle.h`