particle-iot/device-os #1832: Issue with multiple sequential writes to EEPROM - system freeze

shanevanj, 2019-06-20:
<details>
  <summary><i>submission notes</i></summary>
</details>

## Bug Report

### Expected Behavior

Allow data to be written to the EEPROM using EEPROM.put() in any sequence or for any length of time (within the physical limits of the EEPROM)

### Observed Behavior

When using SYSTEM_THREAD(ENABLED) the app will randomly stop after 10 or more sequential writes of a simple struct using EEPROM.put(). 
The code did not stop at the same places.
The effect on the device varies from solid cyan to flashing cyan to white or even SOS 10.
Not predictable 

### Steps to Reproduce

See app below.
Argon and Xeon tested.
Firmware 1.1.0/1.2.1-rc.2/1.2.1-rc.3

### Workaround

Remove  SYSTEM_THREAD(ENABLED)
Use SYSTEM_MODE(SEMI_AUTOMATIC)

call Particle.connect() once mass EEPROM updates are completed

### Test App

```
SYSTEM_THREAD(ENABLED);

const int dbOffset = 50;
const int sizeOfAttributeName = 4;
const int largestAttributeValue = 20;
const int numOfAttributes = 75;

struct {  // common struct for attributes stored in EEPROM
  int attributeNum;
  char attributeValue[largestAttributeValue + 1];
  time_t lastChanged;
} attributes;

// ------------------------------------------------------------------------------
void eepromCreateDataTable() //set NVram to defaults
// ------------------------------------------------------------------------------
{
    
    Serial.printf("\nStarting EEPROM test");
    
    for ( int recno = 0; recno < numOfAttributes; recno++ ) { //
    
        attributes.attributeNum = recno;
        attributes.lastChanged = Time.now();
        snprintf(attributes.attributeValue, largestAttributeValue, "%05d", recno  );
    
        SINGLE_THREADED_BLOCK() {
             EEPROM.put((recno * sizeof(attributes)) + dbOffset, attributes);
        }
        Serial.printf("\nRecord %d added", recno);
    }
}

// ------------------------------------------------------------------------------
void selectAll()
// ------------------------------------------------------------------------------
{
    Serial.print("\n#\tName\tUpdated\t\tValue");
    
    for (int recno = 0; recno < numOfAttributes ; recno++) {

        EEPROM.get((recno * sizeof(attributes)) + dbOffset, attributes);
        
        Serial.printf("\n%d\t%d\t\t%lu\t%s", recno, attributes.attributeNum, attributes.lastChanged, attributes.attributeValue);
    }

}

// ------------------------------------------------------------------------------
void setup() 
// ------------------------------------------------------------------------------
{
    Serial.begin();   // open serial over USB
    while(!Serial.isConnected()) // wait for Host to open serial port
        Particle.process();
    
    eepromCreateDataTable();
    selectAll();
    
    Serial.print("\nTest completed");
    
}

// ------------------------------------------------------------------------------
void loop() 
// ------------------------------------------------------------------------------
{
    delay(5000);
    Serial.print(".");
}

```

### References

Links to the Community, Docs, Other Issues, etc..

---

ScruffR, 2019-06-20:
This is the respective community thread
https://community.particle.io/t/random-eeprom-write-put-failures-argon-xenon/50518

shanevanj, 2019-09-21:
**Update:** 
SYSTEM_THREAD is NOT enabled - 2150 iteration completed with no issue

With SYSTEM_THREAD(ENABLED) fails within first iteration

m-mcgowan, 2019-09-25:
Thanks for the well thought out bug report and test application. We will look into this!

shanevanj, 2019-10-23:
**Update** - tested with 1.4.2 on Argon
**TLDR**; fails instantly with SYSTEM_THREAD(ENABLED);

Updated test code - added iteration counter.
Test device ARGON, fresh particle update and flash tinker from CLI.

Test code:

`SYSTEM_THREAD(ENABLED);

const int dbOffset = 50;
const int sizeOfAttributeName = 4;
const int largestAttributeValue = 20;
const int numOfAttributes = 75;

struct {  // common struct for attributes stored in EEPROM
  int attributeNum;
  char attributeValue[largestAttributeValue + 1];
  time_t lastChanged;
} attributes;
// ------------------------------------------------------------------------------
void eepromCreateDataTable() //set NVram to defaults
// ------------------------------------------------------------------------------
{
    Serial.printf("\nStarting EEPROM test");
    for ( int recno = 0; recno < numOfAttributes; recno++ ) { //
        attributes.attributeNum = recno;
        attributes.lastChanged = Time.now();
        snprintf(attributes.attributeValue, largestAttributeValue, "%05d", recno  );
        EEPROM.put((recno * sizeof(attributes)) + dbOffset, attributes);
        Serial.printf("\nRecord %d added", recno);
    }
}
// ------------------------------------------------------------------------------
void selectAll()
// ------------------------------------------------------------------------------
{
    Serial.print("\n#\tName\tUpdated\t\tValue");
    for (int recno = 0; recno < numOfAttributes ; recno++) {
        EEPROM.get((recno * sizeof(attributes)) + dbOffset, attributes);
        Serial.printf("\n%d\t%d\t\t%lu\t%s", recno, attributes.attributeNum, attributes.lastChanged, attributes.attributeValue);
    }
}
// ------------------------------------------------------------------------------
void setup() 
// ------------------------------------------------------------------------------
{
    Serial.begin();   // open serial over USB
    while(!Serial.isConnected()) // wait for Host to open serial port
        Particle.process();
}
// ------------------------------------------------------------------------------
void loop() 
// ------------------------------------------------------------------------------
{
    static int testIteration;
    
    eepromCreateDataTable();
    selectAll();
    
    Serial.printlnf("\n\nTest cycle %d completed", ++testIteration);
    delay(5000);
}`

avtolstoy, 2019-10-23:
@shanevanj I just ran the test app on an Argon with 1.4.2 for about 100 iterations and wasn't able to reproduce the issue.

SOS 10 would indicate an assertion failure, which most likely comes from LittleFS and not our code (there are no assertions in `read()`/`write()` operations). There are just a handful of cases where LittleFS might throw an assertion failure during runtime and most of them are caused by bad block numbers, fs structure issues and caching issues. Would it be possible for you to use a debugger and capture the stacktrace? You would need to set a breakpoint on `panic_` function or just interrupt the execution while it's still in the panic handler.

I might also suggest wiping the whole external flash, by simply writing 4MB of `0xff` into it over DFU and seeing if that helps, DCT can be backed up, so keys and other things will stay intact, but you'll need to reprovision wifi credentials after this:
```console
$ dfu-util -d 2b04:d00c -a 1 -s 0x1 -U dct_backup.bin # backup DCT
$ dfu-util -d 2b04:d00c -a 2 -s 0x80000000 -D 4mb_ff.bin # erase external flash
$ dfu-util -d 2b04:d00c -a 1 -s 0x1 -D dct_backup.bin # restore DCT backup
```

avtolstoy, 2019-10-23:
The fault comes from SoftDevice:
```
#0  panic_ (code=code@entry=HardFault, extraInfo=extraInfo@entry=0x0, HAL_Delay_Microseconds=0x53d25 <HAL_Delay_Microseconds(uint32_t)>) at src/panic.c:41
#1  0x000495c0 in app_error_fault_handler (_id=<optimized out>, _pc=<optimized out>, _info=0) at src/nRF52840/core_hal.c:171
#2  0x00024c94 in ?? ()
```

Full stacktrace from all the threads:
```
Thread 9 (Thread 536902592):
#0  __ISB () at ../../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include/cmsis_gcc.h:418
#1  prvProcessTimerOrBlockTask (xListWasEmpty=1, xNextExpireTime=<optimized out>) at freertos/FreeRTOS/Source/timers.c:603
#2  prvTimerTask (pvParameters=<optimized out>) at freertos/FreeRTOS/Source/timers.c:551
#3  0x00000000 in ?? ()

Thread 8 (Thread 536977352):
#0  __ISB () at ../../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include/cmsis_gcc.h:418
#1  xQueueReceive (xQueue=0x200196f8, pvBuffer=pvBuffer@entry=0x20019fac, xTicksToWait=xTicksToWait@entry=4294967295) at freertos/FreeRTOS/Source/queue.c:1378
#2  0x0005bee6 in os_queue_take (queue=<optimized out>, item=item@entry=0x20019fac, delay=delay@entry=4294967295) at src/nRF52840/concurrent_hal.cpp:364
#3  0x0005b67e in BleObject::BleEventDispatcher::processBleEventFromThread (param=0x200196c8) at src/nRF52840/ble_hal.cpp:667
#4  0x000a4ac4 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 7 (Thread 537025152):
#0  __ISB () at ../../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include/cmsis_gcc.h:418
#1  xQueueSemaphoreTake (xQueue=0x2001cef8, xTicksToWait=xTicksToWait@entry=4294967295) at freertos/FreeRTOS/Source/queue.c:1569
#2  0x0005bfc4 in os_semaphore_take (semaphore=<optimized out>, timeout=timeout@entry=4294967295, reserved=reserved@entry=false) at src/nRF52840/concurrent_hal.cpp:457
#3  0x000618f4 in (anonymous namespace)::ot_process (arg=0x2001cf48) at network/openthread/ot_api.cpp:69
#4  0x000a4ac4 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 6 (Thread 537061232):
#0  __ISB () at ../../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include/cmsis_gcc.h:418
#1  xEventGroupWaitBits (xEventGroup=0x20029ce8, uxBitsToWaitFor=28, xClearOnExit=1, xWaitForAllBits=0, xTicksToWait=<optimized out>)
    at freertos/FreeRTOS/Source/event_groups.c:403
#2  0x00000032 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 5 (Thread 537040368):
#0  __ISB () at ../../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include/cmsis_gcc.h:418
#1  xQueueReceive (xQueue=0x20026468, pvBuffer=pvBuffer@entry=0x200295cc, xTicksToWait=xTicksToWait@entry=100) at freertos/FreeRTOS/Source/queue.c:1378
#2  0x0005bee6 in os_queue_take (queue=<optimized out>, item=item@entry=0x200295cc, delay=delay@entry=100) at src/nRF52840/concurrent_hal.cpp:364
#3  0x000528fe in particle::net::Esp32NcpNetif::loop (arg=0x20026210) at network/lwip/esp32/esp32ncpnetif.cpp:155
#4  0x000a4ac4 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 4 (Thread 536988240):
#0  __ISB () at ../../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include/cmsis_gcc.h:418
#1  xQueueReceive (xQueue=0x2001b168, pvBuffer=pvBuffer@entry=0x2001ca34, xTicksToWait=xTicksToWait@entry=100) at freertos/FreeRTOS/Source/queue.c:1378
#2  0x0004a8e8 in sys_arch_mbox_fetch (mbox=mbox@entry=0x2000c5d8 <tcpip_mbox>, msg=msg@entry=0x2001ca34, timeout_ms=timeout_ms@entry=100) at src/nRF52840/lwip/sys_arch.c:430
#3  0x00076ba6 in tcpip_timeouts_mbox_fetch (mbox=<optimized out>, msg=<optimized out>) at lwip/src/api/tcpip.c:104
#4  tcpip_thread (arg=<optimized out>) at lwip/src/api/tcpip.c:142
#5  0x000a4ac4 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 3 (Thread 536972744):
#0  0x000c8cd0 in delay_machine_code ()
#1  0x0009fafa in nrfx_coredep_delay_us (time_us=10) at ./nrf5_sdk/modules/nrfx/soc/nrfx_coredep.h:177
#2  nrfx_qspi_cinstr_xfer (p_config=0x20018b90, p_tx_buffer=<optimized out>, p_rx_buffer=0x20018b8f) at nrf5_sdk/modules/nrfx/drivers/src/nrfx_qspi.c:204
#3  0x0009fbcc in nrfx_qspi_mem_busy_check () at nrf5_sdk/modules/nrfx/drivers/src/nrfx_qspi.c:245
#4  0x00049050 in erase_common (start_addr=16384, num_blocks=1, len=<optimized out>) at src/nRF52840/exflash_hal.c:277
#5  0x000a6058 in lfs_bd_erase (lfs=0x2000a464 <(anonymous namespace)::s_instance+64>, block=<optimized out>) at littlefs/lfs.c:247
---Type <return> to continue, or q <return> to quit--- 
#6  lfs_dir_commit (lfs=lfs@entry=0x2000a464 <(anonymous namespace)::s_instance+64>, dir=dir@entry=0x20018ca4, regions=regions@entry=0x20018c58, count=count@entry=1)
    at littlefs/lfs.c:524
#7  0x000a62bc in lfs_dir_update (lfs=lfs@entry=0x2000a464 <(anonymous namespace)::s_instance+64>, dir=dir@entry=0x20018ca4, entry=entry@entry=0x20018c94, data=data@entry=0x0)
    at littlefs/lfs.c:661
#8  0x000a6336 in lfs_file_sync (lfs=0x2000a464 <(anonymous namespace)::s_instance+64>, file=0x20008710 <(anonymous namespace)::eeprom()::eeprom+4>) at littlefs/lfs.c:1559
#9  0x000a6356 in lfs_file_close (lfs=0x2000a464 <(anonymous namespace)::s_instance+64>, file=0x20008710 <(anonymous namespace)::eeprom()::eeprom+4>) at littlefs/lfs.c:1394
#10 0x0005658a in (anonymous namespace)::EepromFile::close (this=<optimized out>) at src/nRF52840/eeprom_hal.cpp:121
[... cut ...]
#1  0x000495c0 in app_error_fault_handler (_id=<optimized out>, _pc=<optimized out>, _info=0) at src/nRF52840/core_hal.c:171
#2  0x00024c94 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 1 (Thread 536903744):
#0  prvIdleTask (pvParameters=<optimized out>) at freertos/FreeRTOS/Source/tasks.c:3447
#1  0x000a4ac4 in ?? ()
```

My only guess would be that `nrfx_qspi_mem_busy_check()` somehow prevents SoftDevice from running correctly, perhaps due to the fact that we are busy looping. @YutingYou Any ideas?

avtolstoy, 2019-10-23:
Here is a different stack trace, where an assertion failure is trigger by Nordic SDK:
```
#0  0x00053d3c in HAL_Delay_Microseconds (uSec=<optimized out>) at src/nRF52840/delay_hal.cpp:43
#1  0x00031334 in panic_ (code=code@entry=AssertionFailure, extraInfo=extraInfo@entry=0x0, HAL_Delay_Microseconds=0x53d0d <HAL_Delay_Microseconds(uint32_t)>) at src/panic.c:92
#2  0x0009fb4c in nrfx_qspi_cinstr_xfer (p_config=0x20018d70, p_tx_buffer=<optimized out>, p_rx_buffer=0x20018d6f) at nrf5_sdk/modules/nrfx/drivers/src/nrfx_qspi.c:214
#3  0x0009fbb4 in nrfx_qspi_mem_busy_check () at nrf5_sdk/modules/nrfx/drivers/src/nrfx_qspi.c:245
#4  0x00049020 in erase_common (start_addr=466944, num_blocks=1, len=<optimized out>) at src/nRF52840/exflash_hal.c:278
#5  0x000a5ae6 in lfs_bd_erase (lfs=0x2000a564 <(anonymous namespace)::s_instance+64>, block=<optimized out>) at littlefs/lfs.c:247
#6  lfs_ctz_extend (off=0x20008838 <(anonymous namespace)::eeprom()::eeprom+44>, block=0x20008834 <(anonymous namespace)::eeprom()::eeprom+40>, size=466, head=113, 
    pcache=<optimized out>, rcache=0x2000a578 <(anonymous namespace)::s_instance+84>, lfs=0x2000a564 <(anonymous namespace)::s_instance+64>) at littlefs/lfs.c:1167
#7  lfs_file_write (lfs=0x2000a564 <(anonymous namespace)::s_instance+64>, file=file@entry=0x20008810 <(anonymous namespace)::eeprom()::eeprom+4>, 
    buffer=buffer@entry=0x2003e3b8, size=size@entry=32) at littlefs/lfs.c:1685
---Type <return> to continue, or q <return> to quit---
#8  0x000567ca in (anonymous namespace)::EepromFile::write (size=32, buffer=0x2003e3b8 "\r", offset=466, this=0x2000880c <(anonymous namespace)::eeprom()::eeprom>)
    at src/nRF52840/eeprom_hal.cpp:65
#9  HAL_EEPROM_Put (index=466, data=0x2003e3b8, length=32) at src/nRF52840/eeprom_hal.cpp:206
#10 0x000d412e in ?? ()
```

Full stack traces from all the threads:
```
Thread 9 (Thread 536902848):
#0  __ISB () at ../../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include/cmsis_gcc.h:418
#1  prvProcessTimerOrBlockTask (xListWasEmpty=1, xNextExpireTime=<optimized out>) at freertos/FreeRTOS/Source/timers.c:603
#2  prvTimerTask (pvParameters=<optimized out>) at freertos/FreeRTOS/Source/timers.c:551
#3  0x00000000 in ?? ()

Thread 8 (Thread 536977608):
#0  __SEV () at ../../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include/cmsis_gcc.h:406
#1  xQueueReceive (xQueue=0x200197f8, pvBuffer=pvBuffer@entry=0x2001a0ac, xTicksToWait=xTicksToWait@entry=4294967295) at freertos/FreeRTOS/Source/queue.c:1378
#2  0x0005bece in os_queue_take (queue=<optimized out>, item=item@entry=0x2001a0ac, delay=delay@entry=4294967295) at src/nRF52840/concurrent_hal.cpp:364
#3  0x0005b666 in BleObject::BleEventDispatcher::processBleEventFromThread (param=0x200197c8) at src/nRF52840/ble_hal.cpp:667
#4  0x000a4aac in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 7 (Thread 537025408):
#0  __ISB () at ../../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include/cmsis_gcc.h:418
#1  xQueueSemaphoreTake (xQueue=0x2001cff8, xTicksToWait=xTicksToWait@entry=4294967295) at freertos/FreeRTOS/Source/queue.c:1569
#2  0x0005bfac in os_semaphore_take (semaphore=<optimized out>, timeout=timeout@entry=4294967295, reserved=reserved@entry=false) at src/nRF52840/concurrent_hal.cpp:457
#3  0x000618dc in (anonymous namespace)::ot_process (arg=0x2001d048) at network/openthread/ot_api.cpp:69
#4  0x000a4aac in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 6 (Thread 537059704):
#0  __ISB () at ../../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include/cmsis_gcc.h:418
#1  xQueueSemaphoreTake (xQueue=xQueue@entry=0x2000a4dc <(anonymous namespace)::s_lfs_mutex+4>, xTicksToWait=xTicksToWait@entry=4294967295)
    at freertos/FreeRTOS/Source/queue.c:1569
#2  0x0007763e in xQueueTakeMutexRecursive (xMutex=0x2000a4dc <(anonymous namespace)::s_lfs_mutex+4>, xTicksToWait=xTicksToWait@entry=4294967295)
    at freertos/FreeRTOS/Source/queue.c:670
#3  0x0005ca1a in StaticRecursiveMutex::lock (timeout=0, this=0x2000a4d8 <(anonymous namespace)::s_lfs_mutex>) at ./src/nRF52840/static_recursive_mutex.h:45
#4  filesystem_lock (fs=fs@entry=0x2000a524 <(anonymous namespace)::s_instance>) at src/nRF52840/littlefs/filesystem.cpp:38
#5  0x000569b8 in particle::fs::FsLock::lock (this=<synthetic pointer>) at ./src/nRF52840/littlefs/filesystem.h:82
#6  particle::fs::FsLock::FsLock (fs=0x2000a524 <(anonymous namespace)::s_instance>, this=<synthetic pointer>) at ./src/nRF52840/littlefs/filesystem.h:74
#7  (anonymous namespace)::DcdFile::read (size=15, buffer=0x2002df30 "", offset=8119, this=0x20008850 <(anonymous namespace)::dcd()::dcd>) at src/nRF52840/dct_hal.cpp:44
#8  dct_read_app_data_copy (offset=offset@entry=8119, ptr=ptr@entry=0x2002df30, size=size@entry=15) at src/nRF52840/dct_hal.cpp:195
#9  0x00056cae in hal_get_device_secret (data=data@entry=0x2002ec14 '\377' <repeats 68 times>, "\060\343\003 \330", <incomplete sequence \366>, size=size@entry=32, 
    reserved=reserved@entry=0x0) at src/nRF52840/deviceid_hal.cpp:87
---Type <return> to continue, or q <return> to quit---
#10 0x00053a9a in fetch_system_properties (storage=0x2002ec10, keyCount=2, flags=<optimized out>) at src/nRF52840/ota_flash_hal.cpp:656
#11 0x00053b0a in add_system_properties (info=info@entry=0x2002df84, create=create@entry=true, additional=additional@entry=0) at src/nRF52840/ota_flash_hal.cpp:682
[... cut ...]
#7  lfs_file_write (lfs=0x2000a564 <(anonymous namespace)::s_instance+64>, file=file@entry=0x20008810 <(anonymous namespace)::eeprom()::eeprom+4>, 
    buffer=buffer@entry=0x2003e3b8, size=size@entry=32) at littlefs/lfs.c:1685
---Type <return> to continue, or q <return> to quit---
#8  0x000567ca in (anonymous namespace)::EepromFile::write (size=32, buffer=0x2003e3b8 "\r", offset=466, this=0x2000880c <(anonymous namespace)::eeprom()::eeprom>)
    at src/nRF52840/eeprom_hal.cpp:65
#9  HAL_EEPROM_Put (index=466, data=0x2003e3b8, length=32) at src/nRF52840/eeprom_hal.cpp:206
#10 0x000d412e in ?? ()

```

YutingYou, 2019-10-23:
The interrupt priority of QSPI is 7 which is the lowest priority, it shouldn't block Softdevice, I'll look into this issue.

YutingYou, 2019-10-31:
In `nrfx_qspi_mem_busy_check()`, the driver is going to read the status register from the external flash by using sending custom instruction API `nrfx_qspi_cinstr_xfer()`. From the stack trace log, we can find `nrfx_qspi_cinstr_xfer()` didn't behave as we expected, it is caused by timeout receiving the ready event(`NRF_QSPI_EVENT_READY`)

> The READY event will be generated when the custom instruction has been sent.

Given there are some bug fixes for QSPI driver in Nordic nrfx driver, I was going to find a way to reproduce this issue and try whether the new QSPI driver can fix it, however, it's hard to reproduce this issue on my side. I tried to immediately open its serial port, use monolithic/modular firmware, increase the packet size, remove the antenna, but got no luck.

@shanevanj If you got time, could you please help try the new QSPI driver I attached, I've confirmed it works on Device OS v1.4.2.
```
third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers/src/nrfx_qspi.c
third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers/include/nrfx_qspi.h
third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/hal/nrf_qspi.h
```

There are also other people encountered this issue:
https://devzone.nordicsemi.com/f/nordic-q-a/28947/what-does-qspi-activate-task-do

I've sent a ticket to Nordic, I'll reply to this thread once I have progress.

[qspi_driver_v1.8.1.zip](https://github.com/particle-iot/device-os/files/3793023/qspi_driver_v1.8.1.zip)

shanevanj, 2019-12-28:
Tested on 1.4.4 and all is working as expected.

Thanks everyone!

shanevanj, 2019-12-28:
I spoke too soon ........

XENON:
Standalone - no issue
When plugged into a ethernet wing - as soon as it connects to particle cloud - it freezes on the write operations

ARGON:
Freeze almost immedialy

shanevanj, 2019-12-29:
**Update:**

**Test APP (below):**

- removed SINGLE_THREADED_BLOCK and added in 2 Particle.publish() calls.
- SYSTEM_THREAD(ENABLED)

**XENON with Ethernet wing:**
Still failing once publish functions are called - so I suspect the SPI issues as @rickkas7 describes above are in play once the SPI ethernet is used. It seems the SINGLE _THREADED_BLOCK just made he issue appear sooner.

**ARGON:**
Takes longer to fail but locks up within 3 iterations

The console does not receive the publish data from either device at all.

**_Remove_** SYSTEM_THREAD(ENABLED) - all devices work and do not lockup, Published messages received in console.

`SYSTEM_THREAD(ENABLED);

const int dbOffset = 50;
const int sizeOfAttributeName = 4;
const int largestAttributeValue = 20;
const int numOfAttributes = 75;

struct
{ // common struct for attributes stored in EEPROM
  int attributeNum;
  char attributeValue[largestAttributeValue + 1];
  time_t lastChanged;
} attributes;

char buf[128];

// ------------------------------------------------------------------------------
void eepromCreateDataTable() //set NVram to defaults
// ------------------------------------------------------------------------------
{

  Serial.printf("\nStarting EEPROM test");

  for (int recno = 0; recno < numOfAttributes; recno++)
  { //

    attributes.attributeNum = recno;
    attributes.lastChanged = Time.now();
    snprintf(attributes.attributeValue, largestAttributeValue, "%05d", recno);

    EEPROM.put((recno * sizeof(attributes)) + dbOffset, attributes);

    Serial.printf("\nRecord %d added", recno);
  }
}

// ------------------------------------------------------------------------------
void selectAll()
// ------------------------------------------------------------------------------
{
  Serial.print("\n#\tName\t\tUpdated\t\tValue");

  for (int recno = 0; recno < numOfAttributes; recno++)
  {

    EEPROM.get((recno * sizeof(attributes)) + dbOffset, attributes);

    Serial.printf("\n%d\t%d\t\t%lu\t%s", recno, attributes.attributeNum, attributes.lastChanged, attributes.attributeValue);
  }
}

// ------------------------------------------------------------------------------
void setup()
// ------------------------------------------------------------------------------
{
  Serial.begin();               // open serial over USB
  while (!Serial.isConnected()) // wait for Host to open serial port
    Particle.process();

  Particle.publish("sysMsg", "EEPROM Test Started", PRIVATE);
}

// ------------------------------------------------------------------------------
void loop()
// ------------------------------------------------------------------------------
{
  static int iterationCount = 0;
  eepromCreateDataTable();
  selectAll();

  Serial.printlnf(buf,
                  "\n\n----------------------------\nIteration:%d\n----------------------------",
                  ++iterationCount);
  sprintf(buf,
          "Iteration:%d",
          ++iterationCount);

  Particle.publish("sysMsg", buf, PRIVATE);
  
  delay(5000);
}
`

shanevanj, 2019-12-29:
**UPDATE**

Test APP (below):

- removed SINGLE_THREADED_BLOCK and added in 2 Particle.publish() calls.
- SYSTEM_THREAD(ENABLED)

**XENON with Ethernet wing:**
Still failing once publish functions are called - so I suspect the SPI issues as @rickkas7 describes above are in play once the SPI ethernet is used. It seems the SINGLE _THREADED_BLOCK just made he issue appear sooner.

**ARGON**:
Takes longer to fail but locks up within 3 iterations

The console does not receive the publish data from either device at all.

**_Remove_** SYSTEM_THREAD(ENABLED) - all devices work and do not lockup, Published messages received in console.

APP:
`// SYSTEM_THREAD(ENABLED);

const int dbOffset = 50;
const int sizeOfAttributeName = 4;
const int largestAttributeValue = 20;
const int numOfAttributes = 75;

struct
{ // common struct for attributes stored in EEPROM
  int attributeNum;
  char attributeValue[largestAttributeValue + 1];
  time_t lastChanged;
} attributes;

char buf[128];

// ------------------------------------------------------------------------------
void eepromCreateDataTable() //set NVram to defaults
// ------------------------------------------------------------------------------
{

  Serial.printf("\nStarting EEPROM test");

  for (int recno = 0; recno < numOfAttributes; recno++)
  { //

    attributes.attributeNum = recno;
    attributes.lastChanged = Time.now();
    snprintf(attributes.attributeValue, largestAttributeValue, "%05d", recno);

    EEPROM.put((recno * sizeof(attributes)) + dbOffset, attributes);

    Serial.printf("\nRecord %d added", recno);
  }
}

// ------------------------------------------------------------------------------
void selectAll()
// ------------------------------------------------------------------------------
{
  Serial.print("\n#\tName\t\tUpdated\t\tValue");

  for (int recno = 0; recno < numOfAttributes; recno++)
  {

    EEPROM.get((recno * sizeof(attributes)) + dbOffset, attributes);

    Serial.printf("\n%d\t%d\t\t%lu\t%s", recno, attributes.attributeNum, attributes.lastChanged, attributes.attributeValue);
  }
}

// ------------------------------------------------------------------------------
void setup()
// ------------------------------------------------------------------------------
{
  Serial.begin();               // open serial over USB
  while (!Serial.isConnected()) // wait for Host to open serial port
    Particle.process();

  Particle.publish("sysMsg", "EEPROM Test Started", PRIVATE);
}

// ------------------------------------------------------------------------------
void loop()
// ------------------------------------------------------------------------------
{
  static int iterationCount = 0;
  eepromCreateDataTable();
  selectAll();

  Serial.printlnf("\n\n----------------------------\nIteration:%d\n----------------------------",
                  ++iterationCount);
  sprintf(buf,
          "Iteration:%d",
          ++iterationCount);

  Particle.publish("sysMsg", buf, PRIVATE);

  delay(5000);
}
`

avtolstoy, 2019-12-30:
We are still looking into this issue and it is not yet resolved as you have confirmed as well.

I believe we are affected by nRF52840 anomaly 215: https://infocenter.nordicsemi.com/index.jsp?topic=%2Ferrata_nRF52840_Rev2%2FERR%2FnRF52840%2FRev2%2Flatest%2Fanomaly_840_215.html&anchor=anomaly_840_215

The fix is not so trivial unfortunately and we might have to rework a significant portion of DeviceOS dealing with module info. Resolving this is a priority and I'll update this issue once we have some concrete progress.

closed 2020-02-04 by avtolstoy