particle-iot/device-os #1834: Enabling H/W Watchdog and sleeping, then using System.reset cause the Xenon to enter DFU mode randomly

iitgrad, 2019-06-22:
</details>

## Bug Report
When setting the hardware watchdog timer for a long period and then going to sleep for a relatively short time, then waking for a short time and then doing System.reset the device randomly enters DFU mode. This can take 10 or 15 minutes to replicate. If the hardware watchdog timer is not enabled, continuous sleep and reset work normally.

### Expected Behavior
It should never enter DFU mode on it's own.

What should it do?
Just sleep and reset forever.

### Observed Behavior
Enters DFU mode randomly

### Steps to Reproduce

Compile and flash the code below.   
How did you make it do that, and which system/user firmware versions were tested?

### Versions
Tested on v1.1.0-rc.2, and v1.3.0-alpha.1

### Test App

```c
SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

#define WDT_REG 0x40010000
#define WDT_CRV_REG 0x40010504
#define WDT_RREN_REG 0x40010508

#define WDT_RELOAD 0x6E524635

void initializeWatchDog() {
    *(uint32_t *) WDT_RREN_REG = 0x00000001;
    *(uint32_t *) WDT_CRV_REG = (uint32_t) (200000 * 32.768);
    *(uint32_t *) WDT_REG = 0x00000001;
}

STARTUP(initializeWatchDog())

void setup() {
  pinMode(D2, INPUT_PULLDOWN);
}

void loop() {

  System.sleep(D2, RISING, 1*15);  
  delay(2000);
  System.reset();

}
```

### References

Links to the Community, Docs, Other Issues, etc..

---

avtolstoy, 2019-06-24:
@iitgrad Thanks for reporting this. While we do not officially support HW watchdog, I agree that the device shouldn't enter DFU on its own.

We have one of the safety mechanisms that relies on the hardware watchdog temporarily disabled in the bootloader, so things at the moment get out of sync. If the device is reset three times by the watchdog after an OTA update, it will end up in DFU. As a temporary workaround, I suggest adding the following code into your `initializeWatchDog()`:

```cpp
auto flags = (platform_system_flags_t*)(0x2003f3c0UL);
flags->IWDG_Enable_SysFlag = 0xffff;
```

iitgrad, 2019-06-24:
Thanks, @avtolstoy.  I also noticed in testing last night that if you pet the watchdog after each reset, right after setting it up it appears to never enter DFU mode. Is this explainable based on your comment above?  In either case, I'll be sure and add your code, not pet the watchdog, and test today to see what happens.

avtolstoy, 2019-06-24:
Most likely. You may also try powering down your device after an OTA update before running your application using HW watchdog, and seeing if that also resolves the issue. Powering down should also clear up `IWDG_Enable_SysFlag` flag set after OTA update.

iitgrad, 2019-06-24:
@avtolstoy Thanks. also, just to clarify this is a production device going into a ground application. It ties into a completely separate piece of hardware to send data via Sigfox, a 902 Mhz transmission. I'm using the Xenon for local control when above ground using BLE for settings, config, etc in the field.  But in all of this testing, I have never done an OTA update.  I have been using PW and compiling and flashing locally.  The normal application will sleep for 30 minutes, wake, do a measurement and then go back to sleep.  The product must run underground for 2 years.  With the Xenon sleep current and my battery pack, we should have ample mAh to go 2+ years.  

I do a reset after each wake-up and measure because I find that when using Serial2, even after doing Serial2.end() I never can get back to the low 40uA sleep current.  There is another bug report on this. (#1793) So, I just reset.

avtolstoy, 2019-06-24:
> But in all of this testing, I have never done an OTA update.

I haven't tried to reproduce this yet, so perhaps there are other paths that trigger the same behavior/functionality.

> I do a reset after each wake-up and measure because I find that when using Serial2, even after doing Serial2.end() I never can get back to the low 40uA sleep current.

As for this issue, I think we've been unable to confirm that this is related to `Serial2` and suspect this is most likely caused by race-conditions in sleep/networking code specifically. We are currently working on a solution for that and it will most likely go into one of the next releases.

iitgrad, 2019-06-24:
I have duplicated that this fix does not appear to solve the issue. 
```
auto flags = (platform_system_flags_t*)(0x2003f3c0UL);
flags->IWDG_Enable_SysFlag = 0xffff;
```
Here was my new code with the suggested fix
```SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

#define WDT_REG 0x40010000
#define WDT_CRV_REG 0x40010504
#define WDT_RREN_REG 0x40010508
#define WDT_RR0_REG 0x40010600

#define WDT_RELOAD 0x6E524635

void initializeWatchDog() {
    auto flags = (platform_system_flags_t*)(0x2003f3c0UL);
    flags->IWDG_Enable_SysFlag = 0xffff;
    *(uint32_t *) WDT_RREN_REG = 0x00000001;
    *(uint32_t *) WDT_CRV_REG = (uint32_t) (200000 * 32.768);
    *(uint32_t *) WDT_REG = 0x00000001;
}

STARTUP(initializeWatchDog())

void setup() {
  pinMode(D2, INPUT_PULLDOWN);
}

void loop() {

  System.sleep(D2, RISING, 1*15);  //***D2 is tied to GND
  delay(2000);
  System.reset();

}
```
This is currently being testing on 1.1.0-rc.2

avtolstoy, 2019-06-24:
@iitgrad Thanks for trying the suggested workaround. Perhaps there is something else in play; we'll try reproducing later on and I'll report back on the findings.

I would also suggest upgrading to 1.3.0-rc.1, which contains a [fix](https://github.com/particle-iot/device-os/pull/1803) for STOP sleep mode.

iitgrad, 2019-06-24:
If this helps, the following code works just fine.
```
SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

#define WDT_REG 0x40010000
#define WDT_CRV_REG 0x40010504
#define WDT_RREN_REG 0x40010508
#define WDT_RR0_REG 0x40010600

#define WDT_RELOAD 0x6E524635

void initializeWatchDog() {
    *(uint32_t *) WDT_RREN_REG = 0x00000001;
    *(uint32_t *) WDT_CRV_REG = (uint32_t) (200000 * 32.768);
    *(uint32_t *) WDT_REG = 0x00000001;
    *(uint32_t *) WDT_RR0_REG = WDT_RELOAD;
}

STARTUP(initializeWatchDog())

void setup() {
  pinMode(D2, INPUT_PULLDOWN);
}

void loop() {

  System.sleep(D2, RISING, 1*15);  //***D2 is tied to GND
  delay(2000);
  System.reset();

}
```
Basically added this single line after setting up the watchdog.
```
 *(uint32_t *) WDT_RR0_REG = WDT_RELOAD;
```