particle-iot/device-os #1840: [Gen 3] Xenon sometimes does not come online

gusgonnet, 2019-07-02:
<details>

I have a mesh network composed of an Argon and a Xenon.

The Argon is online all the time running an empty setup() and empty loop(). For the purpose of this test, is only serving as a mesh gateway.

The Xenon is running a small firmware, no sensors attached to it. It's alone in a breadboard with a usb cable connected to it for power, nothing else. The firmware makes the Xenon sleep 5 seconds, and it waits one second to go back to sleep.

</details>

## Bug Report

### Expected Behavior

The Xenon should wake up and connect to the Particle Cloud on it's own.

### Observed Behavior

The Xenon sometimes spends a long time(\*) blinking green or rapidly blinking green and never comes online. Oftentimes, a reset fixes this state.
At some point, it seems the Xenon times out and goes back to sleep. Sometimes when it wakes up is able to come online. Other times, it's not able to come online and the stuck cycle repeats.

(\*) long time: cannot say for real how long, since I wasn't glued to the Xenon. Half an hour is possible here, maybe more.

### Steps to Reproduce

Argon running empty setup() and emtpy loop().
Xenon running the test app below.
Both devices are running Device OS v1.3.0.rc-1.

### Test App

```c
void setup() {
}

void loop() {
  delay(1000);
  System.sleep(D7,RISING,5);
}

```

### References

May be similar to issues reported here:
https://community.particle.io/t/many-unresponsive-xenons-with-ethernet/49645

gusgonnet, 2019-09-26:
cannot reproduce anymore in Device OS 1.4.0

@avtolstoy : Do we close this one?
Thanks

avtolstoy, 2019-09-26:
Sure :+1:

closed 2019-09-26 by gusgonnet