particle-iot/device-os #1856: compiling with 1.2.1 white light bricks p1

markterrill, 2019-07-15:
<details>
  <summary><i>Flashing a P1 with an image created by `particle compile p1` using particle cli 1.43.0 appears to create an image (1.2.1?) that leaves unit with white breathing light. </i></summary>

</details>

## Bug Report

### Expected Behavior
Actually work. Same code using 1.1.1 works. 

What should it do?
Fire up past the initial white pulsing light

### Observed Behavior

What did it actually do?

Stays on white pulse light indefinitely. Pressing setup button to go dark blue blinking mode blinks quickly blue then red

### Steps to Reproduce

How did you make it do that, and which system/user firmware versions were tested?

particle compile p1 [firmware];
particle update; #particle-cli 1.43.0
particle flash --usb [generated .bin file]

### Test App
Ask me for a binary if you want.

markterrill, 2019-07-15:
To 'undo' the mess:
```
particle flash --usb particlefirmware/p1-system-part1\@1.1.1.bin 
particle flash --usb particlefirmware/p1-system-part2\@1.1.1.bin 
particle flash --usb firmware/p1_firmware_1561425116735_20190625*p1* .  # a 1.1.1 compiled firmware
```

pps, the coreid was: 4e004d001851353530333932

markterrill, 2019-07-15:
My best guess is your new bootloader process isn't working. https://github.com/particle-iot/device-os/pull/1788

markterrill, 2019-07-15:
Lastly, this is particle-cli, but may be related. Fairly often 'particle update' fails.

> Marks-2017MBP:sdb-firmwareflash markterrill$ particle update

> Your device is ready for a system update.
> This process should take about 30 seconds. Here it goes!

! An error occurred while attempting to update the system firmware of your device:

dfu-util: 
File too short for DFU suffix
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!

dfu-util: 
Error during download get_status

> Please visit our community forums for help with this error:
https://community.particle.io/
Cannot read property 'code' of undefined

markterrill, 2019-07-15:
Yeah my best bet is that the particle update with the 1.2.1 new bootloader process is simply not working. Some system info:

   Android SDK Tools : 26.1.1 (/Users/markterrill/Library/Android/sdk/)
   ios-deploy        : 1.9.4
   ios-sim           : 8.0.1
   NodeJS            : v8.11.2 (/Users/markterrill/.nvm/versions/node/v8.11.2/bin/node)
   npm               : 6.1.0
   OS                : macOS Mojave
   Xcode             : Xcode 10.2.1 Build version 10E1001

technobly, 2019-07-15:
Have you tried running `particle update` again? If you flash tinker will it behave normally? `particle flash --usb tinker` Also please try running `particle serial inspect` and copy the output here. I would like to try you binary but would like to compile it as well, can you share the source?

markterrill, 2019-07-16:
Hi @technobly I did run particle update, and even particle doctor, including the doctor app. The only fix to get past steady white light (and red if you pressed setup) was to directly usb flash 1.1.1 parts.
particle serial inspect gave nothing, i think it even had a json parse error.

cli 1.42 works perfectly along with 1.1.1.

Can give access to the repo. Don't judge the messy code too harshly, was cobbled together 4 years ago when I didn't know what a pointer was. Recently refactored everything to try out esp32 but the particle code remains messy.

markterrill, 2019-07-16:
Hi @technobly , sent you an invite to the repo. Just learnt that unless I setup a github organisation I can only provide full collaborator access, so don't break anything ;)

technobly, 2019-07-16:
Hi @markterrill I checked my email(s) and spam folders and don't see an invite.  Before you potentially resend the invite, could you update the CLI to 1.43.1 and retest? That might fix your observed outcome as noted above, and should definitely fix the `particle update` part of your process. Thanks!

markterrill, 2019-07-24:
I'm going to stick with 1.1.1 for the time being. It's working, that's all I need.

technobly, 2019-07-24:
Ok I’m going to assume it’s fixed as noted in CLI 1.43.1 and close this issue.

closed 2019-07-24 by technobly