particle-iot/device-os #1859: BLE.scan() timeout decreases by itself after ~60 seconds of operation

dominicgermana, 2019-07-17:
## Bug Report
BLE.scan() timeout decreases from the default value of 5000 ms to a varying value less than 1000 ms without the user changing the scan parameters.

### Expected Behavior
BLE.scan should scan for 5000 ms (or the user defined timeout, once [setScanParameters is fixed](https://github.com/particle-iot/device-os/issues/1855)) every time the BLE.scan() method is called.

### Observed Behavior
Upon resetting the Xenon, BLE.scan() successfully scans for 5000 ms each time it is called, inside loop(), for somewhere in the ballpark of 60 - 80 seconds (time to failure varies).  Thereafter, the scan time decreased drastically to an average of about 400 ms, but varying from about 100 ms to 1000 ms.

The reported timeout value (from BLE.getScanParameters) did not change, it was always 500 throughout the correct and then incorrect behavior.

Here is an excerpt of the serial output during the changeover from success to failure mode:

> 0
> Parameters:
> 1
> 12
> 160
> 80
> 500
> 1
> 0
> Start time: 50460
> Finish time: 55461
> Scan duration (ms): 5001
> Found 5 results.
> 0
> Parameters:
> 1
> 12
> 160
> 80
> 500
> 1
> 0
> Start time: 55462
> Finish time: 60462
> Scan duration (ms): 5000
> Found 5 results.
> 0
> Parameters:
> 1
> 12
> 160
> 80
> 500
> 1
> 0
> Start time: 60464
> Finish time: 61390
> Scan duration (ms): 926
> Found 3 results.
> 0
> Parameters:
> 1
> 12
> 160
> 80
> 500
> 1
> 0
> Start time: 61393
> Finish time: 61702
> Scan duration (ms): 309
> Found 1 results.
> 0
> Parameters:
> 1
> 12
> 160
> 80
> 500
> 1
> 0
> Start time: 61703
> Finish time: 61945
> Scan duration (ms): 242
> Found 1 results.

### Steps to Reproduce
This occurred on a Xenon running standalone (not part of a mesh network), running System firmware version 1301 (deviceOS@1.3.0-rc.1), and Bootloader version 311.  Code compiled and flashed locally in Workbench.

At first I was using the callback overload of BLE.scan() and I thought perhaps there was a threading issue causing the behavior, so I tried the vector overload and then finally the array overload (as in the Test App below).  The behavior persisted with all 3 overloads of BLE.scan().

### Test App
```c
#include "Particle.h"

SYSTEM_THREAD(ENABLED);  // Run this user application immediately upon booting
SYSTEM_MODE(MANUAL);  // Run without requiring cloud (internet) connection

SerialLogHandler handler (LOG_LEVEL_ALL);
const size_t SCAN_RESULT_MAX = 30;
BleScanResult scanResults[SCAN_RESULT_MAX];

void setup() {
  Serial.begin(57600);
}

void loop() {
  BleScanParams params;
  Serial.println(BLE.getScanParameters(&params));
  Serial.println("Parameters:");
  Serial.println(params.version);
  Serial.println(params.size);
  Serial.println(params.interval);
  Serial.println(params.window);
  Serial.println(params.timeout);
  Serial.println(params.active);
  Serial.println(params.filter_policy);  // Default value  

  auto start_time = millis();
  Serial.printlnf("Start time: %d", start_time);
  int count = BLE.scan(scanResults, SCAN_RESULT_MAX);
  auto finish_time = millis();
  Serial.printlnf("Finish time: %d", finish_time);
  Serial.printlnf("Scan duration (ms): %d", finish_time - start_time);

  Serial.printlnf("Found %d results.", count);
}
```

### References

During my testing I tried setting the scan timeout and parameters, and I can confirm the same exact failure and error messages reported in issue #1855.

XuGuohui, 2019-07-18:
I'm working on this branch: https://github.com/particle-iot/device-os/tree/fix%2Fble%2Fv1.3.1-rc.1, and it doesn't produce this issue. Here is the log:

>Start time: 440582
0000440583 [hal.ble] TRACE: hal_ble_gap_start_scan().
0000440583 [hal.ble] TRACE: | interval(ms)   window(ms)   timeout(ms) |
0000440584 [hal.ble] TRACE:   100.000        50.000      5000
Finish time: 445585
Scan duration (ms): 5003
Found 20 results.
>
>Start time: 445586
0000445586 [hal.ble] TRACE: hal_ble_gap_start_scan().
0000445587 [hal.ble] TRACE: | interval(ms)   window(ms)   timeout(ms) |
0000445588 [hal.ble] TRACE:   100.000        50.000      5000
Finish time: 450589
Scan duration (ms): 5003
Found 20 results.
>
>Start time: 450590
0000450590 [hal.ble] TRACE: hal_ble_gap_start_scan().
0000450591 [hal.ble] TRACE: | interval(ms)   window(ms)   timeout(ms) |
0000450592 [hal.ble] TRACE:   100.000        50.000      5000
Finish time: 455593
Scan duration (ms): 5003
Found 19 results.
>
>Start time: 455594
0000455594 [hal.ble] TRACE: hal_ble_gap_start_scan().
0000455595 [hal.ble] TRACE: | interval(ms)   window(ms)   timeout(ms) |
0000455596 [hal.ble] TRACE:   100.000        50.000      5000
Finish time: 460597
Scan duration (ms): 5003
Found 18 results.
>
>Start time: 460598
0000460598 [hal.ble] TRACE: hal_ble_gap_start_scan().
0000460599 [hal.ble] TRACE: | interval(ms)   window(ms)   timeout(ms) |
0000460600 [hal.ble] TRACE:   100.000        50.000      5000
Finish time: 465601
Scan duration (ms): 5003
Found 18 results.
>
>Start time: 465601
0000465602 [hal.ble] TRACE: hal_ble_gap_start_scan().
0000465602 [hal.ble] TRACE: | interval(ms)   window(ms)   timeout(ms) |
0000465603 [hal.ble] TRACE:   100.000        50.000      5000
Finish time: 470604
Scan duration (ms): 5003
Found 18 results.
>
>Start time: 470605
0000470606 [hal.ble] TRACE: hal_ble_gap_start_scan().
0000470606 [hal.ble] TRACE: | interval(ms)   window(ms)   timeout(ms) |
0000470607 [hal.ble] TRACE:   100.000        50.000      5000
Finish time: 475608
Scan duration (ms): 5003
Found 18 results.

Here is the user application:
```
#include "Particle.h"

SYSTEM_THREAD(ENABLED);  // Run this user application immediately upon booting
SYSTEM_MODE(MANUAL);  // Run without requiring cloud (internet) connection

SerialLogHandler handler (LOG_LEVEL_ALL);
const size_t SCAN_RESULT_MAX = 30;
BleScanResult scanResults[SCAN_RESULT_MAX];

void setup() {
    Serial.begin(57600);
}

void loop() {
  auto start_time = millis();
  Serial.printlnf("Start time: %d", start_time);
  int count = BLE.scan(scanResults, SCAN_RESULT_MAX);
  auto finish_time = millis();
  Serial.printlnf("Finish time: %d", finish_time);
  Serial.printlnf("Scan duration (ms): %d", finish_time - start_time);

  Serial.printlnf("Found %d results.", count);
  Serial.println(" ");
}
```

Let's keep tuning on this issue.

XuGuohui, 2019-07-18:
Running fix/1.3.1-rc.1, it works well for a long time:

>Start time: 5414605
0005414605 [hal.ble] TRACE: hal_ble_gap_start_scan().
0005414606 [hal.ble] TRACE: | interval(ms)   window(ms)   timeout(ms) |
0005414607 [hal.ble] TRACE:   100.000        50.000      5000
Finish time: 5419608
Scan duration (ms): 5003
Found 9 results.
>
>Start time: 5419609
0005419609 [hal.ble] TRACE: hal_ble_gap_start_scan().
0005419610 [hal.ble] TRACE: | interval(ms)   window(ms)   timeout(ms) |
0005419611 [hal.ble] TRACE:   100.000        50.000      5000
Finish time: 5424612
Scan duration (ms): 5003
Found 9 results.
>
>Start time: 5424613
0005424613 [hal.ble] TRACE: hal_ble_gap_start_scan().
0005424614 [hal.ble] TRACE: | interval(ms)   window(ms)   timeout(ms) |
0005424615 [hal.ble] TRACE:   100.000        50.000      5000
Finish time: 5429616
Scan duration (ms): 5003
Found 9 results.

dominicgermana, 2019-07-18:
Thank you very much for confirming the problem Xu!  Is it possible for me to download binaries from somewhere and install 1.3.1-rc.1 using Particle CLI, or would I need to build it from source?  Do you know when 1.3.1-rc.1 is expected to be released?  Thanks again!

XuGuohui, 2019-07-19:
You need to build it from source. And 1.3.1-rc.1 is supposed to be released within 3 weeks.

closed 2019-08-08 by XuGuohui