particle-iot/device-os #1861: Implement selectMeshAntenna() in next release

l-gothberg, 2019-07-20:
<details>
  <summary><i>Implement selectExternalMeshAntenna()</i></summary>

## Feature/Enhancement Request
Selection of the external mesh antenna should be a one time event, not a thing that needs to be repeated on every device startup.  Can you please fully implement the feature that is annotated in the below documentation to eliminate the workaround?  Also, defaulting to the external antenna is probably wise since any production use is probably going to rely on better reception than the internal antenna can provide.

### References
https://docs.particle.io/reference/device-os/firmware/xenon/#antenna-selection
</details>

XuGuohui, 2019-08-30:
We do have an open PR that implements this API: https://github.com/particle-iot/device-os/pull/1693 , @avtolstoy do you think we should resolve it in v1.4.0 or later?

l-gothberg, 2019-08-31:
YES!

Regards,
Leo G.

~~~~~
"Non faciat malum, ut inde veniat bonum."

On Thu, Aug 29, 2019 at 8:46 PM XuGuohui <notifications@github.com> wrote:

> We do have an open PR that implements this API: #1693
> <https://github.com/particle-iot/device-os/pull/1693> , @avtolstoy
> <https://github.com/avtolstoy> do you think we should resolve it in
> v1.4.0 or later?
>
> —
> You are receiving this because you authored the thread.
> Reply to this email directly, view it on GitHub
> <https://github.com/particle-iot/device-os/issues/1861?email_source=notifications&email_token=ACOGKCSKRJLEOT72RXWQ3GTQHCJZDA5CNFSM4IFPZ6S2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD5QOQQY#issuecomment-526444611>,
> or mute the thread
> <https://github.com/notifications/unsubscribe-auth/ACOGKCQBPZ66CACVGASL66DQHCJZDANCNFSM4IFPZ6SQ>
> .
>

closed 2019-10-04 by sergeuz

sergeuz, 2019-10-04:
@l-gothberg We've just merged a [fix](https://github.com/particle-iot/device-os/pull/1933) for this, I'd appreciate if you could test it. Thanks!

PR #1933 (2019-10-04): Implement Mesh.selectAntenna()
### Problem

Even though the antenna is shared between mesh and BLE on gen3 devices, for parity with `BLE` there should be a method in the `Mesh` class that would allow setting the antenna used for mesh radio. This setting should be stored persistently, as otherwise an endpoint mesh device won't be able to connect to the cloud in safe mode.

### Solution

Implement `Mesh.selectAntenna()` and use DCT to store the antenna setting persistently.

This PR includes commits from https://github.com/particle-iot/device-os/pull/1693 and fixes https://github.com/particle-io [... cut]