particle-iot/device-os #1866: BLE scan returns incorrect data (fix/ble/v1.3.1-rc.1)

dominicgermana, 2019-07-28:
## Bug Report
BLE.scan() returns incorrect data.
I built branch fix/ble/v1.3.1-rc.1 from source so I could get the fixes for issue #1855 and #1859 
The good news is that both those issues are fixed!
The bad news is that I observed a much worse bug - the scan data returned is incorrect.

### Expected Behavior

BLE.scan() should return the correct address and advertisingData.

### Observed Behavior

The address and advertisingData.length() are incorrect. In my usage case, I'm scanning and looking for results that have a particular address and advertising data length.  With deviceOS v1.3.0-rc.1, this is the correct output from the test code below:

> 00 15 88 06 98 E8 
> Data length: 22 
> 00 15 88 06 72 D5 
> Data length: 22 

With fix/ble/v1.3.1-rc.1, this is the incorrect output:

> 98 E8 00 1F 00 16 
> Data length: 31 
> 72 D5 00 1F 00 16 
> Data length: 31 

### Steps to Reproduce

Using a standalone Xenon running 1.3.0-rc.1, I get the correct scan results, but with the problems reported in issues #1855 (scan parameters can't be set) and #1859 (scan timeout decreases drastically after running for ~60 seconds).
Those issues were reported fixed in branch fix/ble/v1.3.1-rc.1, so I checked out that branch, installed the dependencies, and built this version of the deviceOS from source.
I flashed the resulting bin file over USB (DFU) and was happy to see that the issues were indeed fixed.  Unfortunately, the scan result data is now incorrect.  The device address and advertisingData length are both wrong.  I have scanned the test peripherals I was using previously with other systems, so I know that the address is supposed to be 0x00 15 88 xx xx xx and the data length is 22, as correctly returned using version 1.3.0-rc.1.

### Test App

```c
#include "Particle.h"

SYSTEM_THREAD(ENABLED);  // Run this user application immediately upon booting
SYSTEM_MODE(MANUAL);  // Run without requiring cloud (internet) connection

SerialLogHandler handler (LOG_LEVEL_ALL);

const size_t SCAN_RESULT_MAX = 30;
BleScanResult scanResults[SCAN_RESULT_MAX];

void setup() {
  Serial.begin(57600);
}

void loop() {
  int count = BLE.scan(scanResults, SCAN_RESULT_MAX);

  for (int i = 0; i < count; i++) {
    Serial.printf("%02X ", scanResults[i].address[5]);
    Serial.printf("%02X ", scanResults[i].address[4]);
    Serial.printf("%02X ", scanResults[i].address[3]);
    Serial.printf("%02X ", scanResults[i].address[2]);
    Serial.printf("%02X ", scanResults[i].address[1]);
    Serial.printlnf("%02X ", scanResults[i].address[0]);
    Serial.printlnf("Data length: %02d ", scanResults[i].advertisingData.length());
  }
}
```

### References
Issues #1855 and #1859
Branch https://github.com/particle-iot/device-os/tree/fix/ble/v1.3.1-rc.1

XuGuohui, 2019-07-29:
The address you got previously is wrong, this is actually fixed in fix/ble/v1.3.1-rc.1, see: https://community.particle.io/t/bleaddress-issues-and-general-reproducible-bluetooth-connectivity-issues/51132/5

The advertising data length I got is correct in fix/ble/v1.3.1-rc.1. There is probably something wrong with the advertising data length in 1.3.0-rc.1. Here is the test application I'm using:
```
#include "Particle.h"

#define SCAN_RESULT_COUNT       30
#define BLE_ADV_DATA_MAX        31

SYSTEM_MODE(MANUAL);

SerialLogHandler log(115200, LOG_LEVEL_ALL);

BleScanResult results[SCAN_RESULT_COUNT];

void setup() {

}

void loop() {
    int count = BLE.scan(results, SCAN_RESULT_COUNT);

    if (count > 0) {
        uint8_t buf[BLE_MAX_ADV_DATA_LEN];
        size_t len;

        Serial.printf("%d devices are found:", count);
        for (int i = 0; i < count; i++) {
            BleAddress address = results[i].address;
            Serial.printf("MAC: %s\r\n", address.toString().c_str());

            len = results[i].advertisingData(buf, sizeof(buf));
            if (len > 0) {
                Serial.printf("Advertising data length: %02d\r\n", results[i].advertisingData.length());
                for (size_t j = 0; j < len; j++) {
                    Serial.printf("0x%02x,", buf[j]);
                }
                Serial.println("\r\n");
            }

            len = results[i].scanResponse(buf, sizeof(buf));
            if (len > 0) {
                Serial.printf("Scan response data length: %02d\r\n", results[i].scanResponse.length());
                for (size_t j = 0; j < len; j++) {
                    Serial.printf("0x%02x,", buf[j]);
                }
                Serial.println("\r\n");
            }
        }
    }

    delay(5000);
}
```

And here is the corresponding log:

>MAC: 8C:85:90:B5:24:FA
Advertising data length: 14
0x02,0x01,0x06,0x0a,0xff,0x4c,0x00,0x10,0x05,0x0b,0x1c,0xc0,0x0f,0x48,
>
>MAC: 13:03:A7:6F:3F:02
Advertising data length: 31
0x1e,0xff,0x06,0x00,0x01,0x09,0x20,0x02,0x17,0x98,0xf3,0x82,0xfe,0x6b,0xdf,0x82,0x88,0x1a,0xb1,0xab,0xf5,0xd5,0x2d,0xb3,0xce,0xa2,0xfb,0xfb,0xb5,0x1c,0xcf,
>
>MAC: 7B:60:86:EE:9A:15
Advertising data length: 14
0x02,0x01,0x1a,0x0a,0xff,0x4c,0x00,0x10,0x05,0x03,0x18,0x0c,0xb2,0x41,
>
>MAC: 5C:1E:CA:D0:80:7A
Advertising data length: 14
0x02,0x01,0x1a,0x0a,0xff,0x4c,0x00,0x10,0x05,0x01,0x1c,0x23,0xfe,0xf9,
>
>MAC: 52:63:C3:A0:35:59
Advertising data length: 14
0x02,0x01,0x06,0x0a,0xff,0x4c,0x00,0x10,0x05,0x07,0x1c,0xfe,0x6d,0x93,
>
>MAC: 6E:DD:92:0E:44:B2
Advertising data length: 23
0x02,0x01,0x06,0x13,0xff,0x4c,0x00,0x0c,0x0e,0x00,0x4d,0x76,0x9d,0x2e,0x6f,0x0b,0xa4,0x3e,0x08,0x65,0x80,0x25,0x98,
>
>MAC: 7C:C5:6D:F7:72:50
Advertising data length: 14
0x02,0x01,0x1a,0x0a,0xff,0x4c,0x00,0x10,0x05,0x1b,0x1c,0x85,0x10,0x20,
>
>MAC: 8C:85:90:94:9F:3E
Advertising data length: 11
0x02,0x01,0x06,0x07,0xff,0x4c,0x00,0x10,0x02,0x0b,0x08,
>
>MAC: 4C:97:DE:46:DF:79
Advertising data length: 14
0x02,0x01,0x1a,0x0a,0xff,0x4c,0x00,0x10,0x05,0x03,0x1c,0xa6,0x59,0xe0,
>
>MAC: 75:84:1E:7A:51:64
Advertising data length: 14
0x02,0x01,0x1a,0x0a,0xff,0x4c,0x00,0x10,0x05,0x13,0x1c,0x6c,0x9a,0xe9,
>
>MAC: 68:10:55:3D:A1:90
Advertising data length: 14
0x02,0x01,0x1a,0x0a,0xff,0x4c,0x00,0x10,0x05,0x17,0x1c,0xc3,0x57,0xd3,
>
>MAC: 57:FE:90:EB:17:A0
Advertising data length: 14
0x02,0x01,0x1a,0x0a,0xff,0x4c,0x00,0x10,0x05,0x03,0x1c,0x62,0x37,0xee,
>
>MAC: FC:EC:DA:4A:6F:86
Advertising data length: 31
0x02,0x01,0x06,0x11,0x06,0x17,0x45,0x56,0x10,0x73,0xed,0x09,0xa6,0xb3,0x4e,0x7c,0xe5,0x52,0xaa,0x3c,0xbc,0x09,0x16,0x2a,0x25,0xfc,0xec,0xda,0x4a,0x6f,0x85,
>
>Scan response data length: 16
0x0f,0x08,0x55,0x6e,0x69,0x46,0x69,0x20,0x43,0x6c,0x6f,0x75,0x64,0x4b,0x65,0x79,
>
>MAC: 4E:34:EE:FF:5F:BE
Advertising data length: 14
0x02,0x01,0x1a,0x0a,0xff,0x4c,0x00,0x10,0x05,0x03,0x1c,0xcd,0x7c,0x22,
>
>MAC: 51:31:03:D6:00:6E
Advertising data length: 14
0x02,0x01,0x1a,0x0a,0xff,0x4c,0x00,0x10,0x05,0x13,0x1c,0xb0,0x2b,0x52,
>
>MAC: 78:F9:9A:E6:25:3A
Advertising data length: 14
0x02,0x01,0x1a,0x0a,0xff,0x4c,0x00,0x10,0x05,0x01,0x90,0x49,0xfd,0x5f,

dominicgermana, 2019-07-29:
My apologies Xu, I made a mistake in my testing.  My user application had been compiled against deviceOS@1.3.0-rc.1.
Once I correctly compiled my code against the fix/BLE/v1.3.1-rc.1 branch, following [these instructions](https://docs.particle.io/support/particle-tools-faq/workbench/#working-with-a-custom-device-os-build), I found that BLE.scan() does indeed return the correct address and advertisingData.
Thank you!

closed 2019-07-29 by dominicgermana