particle-iot/device-os #1874: BLE.setAdvertisingInterval() does not work as expected

ScruffR, 2019-08-03:
## Bug Report

This may be relatied to issue #1855 and hence already fixed for a future release but it should be double checked.
Neither `BLE.setAdvertisingInterval()` nor setting the interval via `BLE.setAdvertisingParams()` seem to have any effect despite reporting success (return value `0`).

### Expected Behavior

When seeting the interval scanning devices should also see the updated advertising interval.

### Observed Behavior

Testing with the Android nRF connect app the reported advertising interval always stays the default about 100ms (BTW LightBlue app does not seem to report the interval hence I will stick with nRF connect even if some devs have more faith in that then in nRFs own app ;-) )

### Test App

Tested on Xenon 1.3.0-rc.1
```cpp
SYSTEM_MODE(MANUAL)

const int msInterv = 200;
BleAdvertisingData adv;
BleAdvertisingParams par;

void setup()
{
  Serial.begin();
  delay(5000);
  adv.appendLocalName("advTest");
  BLE.getAdvertisingParameters(&par);
  par.interval = msInterv * 0.625;
  Serial.printlnf("setAdvertisingParams   %d", BLE.setAdvertisingParameters(&par));
  Serial.printlnf("setAdvertisingInterval %d", BLE.setAdvertisingInterval(msInterv * 0.625));
  BLE.advertise(&adv);
}
```

### References

Discussed here
https://community.particle.io/t/having-trouble-setting-the-advertising-interval-for-ibeacon/51497

XuGuohui, 2019-08-03:
This has been fixed in fix/ble/v1.3.1-rc.1.

JeffreyKidwell, 2019-08-05:
@XuGuohui when is this release scheduled to be released.  Our team is on a bit of a time crunch that requires this functionality?

jaredwolff, 2019-08-06:
@JeffreyKidwell you can compile it yourself. If you already have Workbench installed you can:

1. Clone this repo
1. Checkout @XuGuohui's branch ( `git checkout fix/ble/v1.3.1-rc.1` )
1. Chang dir to which ever device you want to build for. In my case I was building for a Xenon: `cd ./device-os/modules/xenon/system-part1`
1. Export your build environment. In OSX it's  `export PATH=/Users/jaredwolff/.particle/toolchains/gcc-arm/5.3.1/bin:$PATH`
1. Run a `make`
It will run though to build your system part. Look for the `.bin` file once its finished. Write that file to your device using `particle flash`

XuGuohui, 2019-08-06:
@JeffreyKidwell 
https://github.com/particle-iot/device-os/issues/1855#issuecomment-518281235

closed 2019-08-08 by XuGuohui