particle-iot/device-os #1880: Options for BLE.connect don't apply.

jaredwolff, 2019-08-06:
## Bug Report

### Expected Behavior

When running `BLE.connect()` with parameters, the parameters should apply.

### Observed Behavior

When running  `BLE.connect()`with parameters (in particular the timeout parameter), `BLE.connect()` still only attempts to connect at the default timeout.

### Steps to Reproduce

Running `fix/ble/v1.3.1-rc.1` with this hash`91a3339cb29f4dcdae4139ebbb826494ec4f5494` (Latest as of 8/6)

### Test App

```c

void onConnectedCallback(const BlePeerDevice& peer, void* context) {

    // Set the flag
    connected = true;

    Log.info("connected!");
}

void scanResultCallback(const BleScanResult *scanResult, void *context) {

    // Connect to the device.
    found = true;
    foundAddress = scanResult->address;
    BLE.stopScanning();

}

void setup() {
    (void)logHandler; // Does nothing, just to eliminate warning for unused variable

    timer_millis = 0;

    delay(200);

   BLE.setScanTimeout(1000);

    BLE.onConnected(onConnectedCallback, NULL);

}

void loop() {
    // Scan for devices

    if( !found ) {
        BLE.scan(scanResultCallback, NULL);
        Log.info("Scan complete");
    } else if (found && !connected) {
        Log.info("connecting.. %02X:%02X:%02X:%02X:%02X:%02X",
            foundAddress[0], foundAddress[1], foundAddress[2],
            foundAddress[3], foundAddress[4], foundAddress[5]);

        BlePeerDevice peer = BLE.connect(foundAddress,24,0,1100);

        if( peer.connected() ) {
            Log.info("connected!");
            connected = true;
        }
        delay(5000);
    }

}
```

`BlePeerDevice peer = BLE.connect(foundAddress,24,0,1100);` should timeout after 11 s, correct?

### References

None at the moment.

XuGuohui, 2019-08-06:
The timeout specified in the argument list is not for establishing a BLE connection. It is one of the connection parameters that being applied after connection established. It is used to evaluate whether a connection is assumed to be disconnected. For example, the timeout value is set to 5 seconds, after the connection being established, if the peer device didn't respond to the handshake by going out of range or powering down the peer device, then after 5 seconds, local device will assume the conection is lost and a disconnected event is issued.

rickkas7, 2019-08-06:
I did not realize that's how that worked. I will clarify that in the documentation (ch36736).

jaredwolff, 2019-08-06:
@XuGuohui good to know! I'm working with a device that is advertising about every 10 seconds (otherwise sleeping). Is there a way to increase the window that BLE.connect() attempts to connect to the device?

XuGuohui, 2019-08-07:
>Is there a way to increase the window that BLE.connect() attempts to connect to the device?

I'm not sure what the `window` you mentioned is. Do you want the `BLE.connect()` keeps trying connecting to a device for a while before it exits?

jaredwolff, 2019-08-08:
I just looked into the Nordic API a little to see how they do things. It looks like they have a `connect_if_match` flag & the `nrf_ble_scan_filter_set` function. That way you can filter by UUID and connect instantaneously. 

Currently, I'm manually filtering by UUID, stopping the scan and then attempting to connect. So I set a flag and copy the address to a global variable. Then in the main function I attempt to connect
```
                // Connect to the device.
                found = true;
                foundAddress = scanResult->address;
                BLE.stopScanning();
```
(Inside scanResultCallback)

```
    // Scan for devices
    if( !found ){
        BLE.scan(scanResultCallback, NULL);
    } else if (!BLE.connected() && found ){
        Log.info("connecting..");
        BLE.connect(foundAddress);
    }
```
(Inside `loop()`)

I disable scanning and attempt to connect but it never does. Tried for over 40 seconds. 

So my main concern was that the connect function quits before ever seeing the device I want to connect to. The device is only accessible for a short amount of time in between advertising packets (those packets are about 10 seconds apart)

Hoooopefully this makes sense. :)

XuGuohui, 2019-08-09:
From the timestamp in the log message, the connect function seams returned immediately without actually starting the connecting process. Please check the returned value of the `connect()` function to see what is going on. With regards of the scanning filter, I was considering to implement that to simplify the development experience. It should be included in future releases.

jaredwolff, 2019-08-09:
Alright I'll close this out for now. I think we've diverted enough from the original idea of this bug report.

closed 2019-08-09 by jaredwolff