particle-iot/device-os #1883: Photon softAP: suppress SSID broadcast

studiohsoftware, 2019-08-11:
## Feature/Enhancement Request

I want to use web pages in softAP instead of connecting to external Wifi. SSID is broadcast in softAP mode, and is not secured (unsecure, insecure?)

I added a new option to the hal_system_config_t enum called SYSTEM_CONFIG_SOFTAP_DISABLE_BROADCAST.

So now I can call this from within the ino in Worbench:
System.set(SYSTEM_CONFIG_SOFTAP_DISABLE_BROADCAST,"1");
or
System.set(SYSTEM_CONFIG_SOFTAP_DISABLE_BROADCAST,"0");

List of files altered:
core_hal.h
ota_flash_hal.cpp
ota_flash_hal_stm32f2xx.cpp
softap.cpp
softap.h
wlan_hal.cpp

Any thoughts on me doing a pull request? Does anyone want this? 

More detail here:
https://community.particle.io/t/photon-softap-either-suppress-ssid-broadcast-or-password-protect/51561

m-mcgowan, 2019-08-13:
Can you please explain in more detail how this is insecure or the issue you are trying to resolve?

studiohsoftware, 2019-08-13:
I am opening up an existing musical device with a wireless rest API. Venue may not have wireless LAN, so hosting URLs in softAP mode is desirable. I want to offer the musician the option to hide the device so that audience members will be less likely to attach to it.