particle-iot/device-os #1884: Missing definitions for PROGMEM related functions

shanevanj, 2019-08-14:
<details>
  <summary><I>The 'n' versions of string library functions strcpy_P and strcmp_P are not defined in https://github.com/particle-iot/device-os/blob/develop/wiring/inc/spark_wiring_arduino.h</i></summary> This is missing in Device OS 1.2.1 tree and 1.3.0 tree
</details>

## Bug Report

strncpy_P strncmp_P etc are not defined in the above header and causes compile failures when these functions are called in app or library code

### Expected Behavior
definitions for the n versions of

`
strcpy_P
strcmp_P
`

need to be added as follows 

`
#ifndef strncpy_P
#define strncpy_P strncpy
#endif

#ifndef strncmp_P
#define strncmp_P strncmp
#endif
`

avtolstoy, 2019-08-14:
This is a duplicate of #1753. These functions are defined in https://github.com/particle-iot/device-os/blob/develop/wiring/inc/avr/pgmspace.h.

closed 2019-08-14 by avtolstoy

shanevanj, 2019-08-14:
Nope including Arduino.h does not resolve the issue for a multiple app in the WebIDE. 

Inserting the following in the .h file where the ArduinoJson.h is first used provides a workaround

#ifndef strncpy_P
#define strncpy_P strncpy
#endif

#ifndef strncmp_P
#define strncmp_P strncmp
#endif

#define ARDUINOJSON_ENABLE_PROGMEM 0

avtolstoy, 2019-08-14:
DeviceOS provides a more complete Arduino compatibility layer only if `Arduino.h` (or in this case `avr/pgmspace.h`) is included.

I've just tried building `ProgmemExample` and adding `#include <Arduino.h>` before including `ArduinoJson.h` resolves the issue:

```cpp
#include <Arduino.h>
#include <ArduinoJson.h>

void setup() {
#ifdef PROGMEM  // <- check that Flash strings are supported

  DynamicJsonDocument doc(1024);

  // You can use a Flash String as your JSON input.
  // WARNING: the strings in the input will be duplicated in the JsonDocument.
  deserializeJson(doc, F("{\"sensor\":\"gps\",\"time\":1351824120,"
                         "\"data\":[48.756080,2.302038]}"));
  JsonObject obj = doc.as<JsonObject>();

  // You can use a Flash String to get an element of a JsonObject
  // No duplication is done.
  long time = obj[F("time")];

  // You can use a Flash String to set an element of a JsonObject
  // WARNING: the content of the Flash String will be duplicated in the
  // JsonDocument.
  obj[F("time")] = time;

  // You can set a Flash String to a JsonObject or JsonArray:
  // WARNING: the content of the Flash String will be duplicated in the
  // JsonDocument.
  obj["sensor"] = F("gps");

  // It works with serialized() too:
  obj["sensor"] = serialized(F("\"gps\""));
  obj["sensor"] = serialized(F("\xA3gps"), 3);

  // You can compare the content of a JsonVariant to a Flash String
  if (obj["sensor"] == F("gps")) {
    // ...
  }

#else

#warning PROGMEM is not supported on this platform

#endif
}

void loop() {
  // not used in this example
}
```

shanevanj, 2019-08-14:
Agreed - in a single file test it seems to work i.e. using the JsonParserExample.ino example works. when expand this to a multiple file app with an independent .cpp and .h set of files with the Arduino.h included in the additional .h file it fails - I will try and post an example showing this

avtolstoy, 2019-08-14:
The key point here is that either the library should correctly be including `Arduino.h` and `avr/pgmspace.h`, or any source or header file including such libraries' headers should be including `Arduino.h` before them.

shanevanj, 2019-08-14:
Thanks for this feedback - it solves the issue. The placement is key to success - why is the Arduino.h just included by default by the preprocessor etc as it seems a crucial puzzle piece?

avtolstoy, 2019-08-14:
Feel free to check out the documentation on the matter: https://docs.particle.io/reference/device-os/firmware/photon/#arduino-compatibility

TL;DR we don't want to break existing libraries adapted for Particle DeviceOS.

Alternatively, instead of including `Arduino.h` you may simply add `#define PARTICLE_WIRING_ARDUINO_COMPATIBILTY 1` (again before any other includes).

shanevanj, 2019-08-14:
Thanks for your patience and advice!