particle-iot/device-os #1885: System event firmware_update reports incorrectly firmware_update_failed since 1.2.1

nzottmann, 2019-08-14:
## Bug Report
Somewhere between deviceOS 1.1.0 and 1.2.1 a bug appeared which causes the system event `firmware_update` to report even successful updates as failed.

### Expected Behavior

System event `firmware_update` should report `firmware_update_complete` if successful and `firmware_update_failed` if not. 

### Observed Behavior

- 1.2.1 and 1.3.0-rc.1
  It reports always `firmware_update_failed`, even for successful updates.
- 0.9.0 and 1.1.0
  Correct behaviour, reports `firmware_update_complete` for successful updates.

### Steps to Reproduce

Boron 2G/3G, deviceOS 1.2.1 or 1.3.0-rc.1
Cloud flash via VS Code or cloud flash via CLI

### Test App

```c
SerialLogHandler logHandler(LOG_LEVEL_WARN, {{ "app", LOG_LEVEL_INFO }});

void firmwareUpdateHandler(system_event_t event, int param)
{
  if(param == firmware_update_begin) Log.info("firmware_update_begin");
  if(param == firmware_update_complete) Log.info("firmware_update_complete");
  if((uint32_t)param == firmware_update_failed) Log.info("firmware_update_failed");
}

void setup() {
  System.on(firmware_update, firmwareUpdateHandler);
}

void loop() { }
```

### References
https://docs.particle.io/reference/device-os/firmware/boron/#system-events-reference

JDSymington, 2019-10-13:
I'm having the same problem with my Electron 3G-U270 on deviceOS 1.4.1-rc1.
Updates always report `firmware_update_failed` even when it was successful.

gkujawsk, 2019-11-20:
Still the same behavior observed on Electron 3G-U270 on 1.4.2.

avtolstoy, 2020-08-28:
This is resolved as of 1.5.1.

closed 2020-08-28 by avtolstoy