particle-iot/device-os #1888: onDataReceived callback not firing on central

sviraaj, 2019-08-16:
<details>
  <summary><i>submission notes</i></summary>

```
**Important:** Please sanitize/remove any confidential info like usernames, passwords, org names, product names/ids, access tokens, client ids/secrets, or anything else you don't wish to share.

Please choose one of the following templates **Bug Report** or **Feature/Enhancement Request** and delete the other.

You may also delete this submission notes header if you'd like. Thank you for contributing!
```
</details>

## Bug Report

### Expected Behavior

What should it do?

### Observed Behavior

What did it actually do?

### Steps to Reproduce

How did you make it do that, and which system/user firmware versions were tested?

### Test App

```c
void setup() {
  // A minimal test case to reproduce the issue is super helpful!
}

void loop() {

}
```

### References

Links to the Community, Docs, Other Issues, etc..

---

## Feature/Enhancement Request

Please describe your use case in as much detail as possible. Background information and detailing what the feature will enable you to do, is especially helpful.

sviraaj, 2019-08-17:
Tested the HTS service client with nRF SDK v15.3 without device OS. It seems to work as expected. I will try to dig up and see, if I can find something on the device OS. (Hope, I am not making some obvious mistake.)

XuGuohui, 2019-08-19:
I built your application against both 1.3.0 and the latest `develop` branch and it worked as expected. Since I don't have a temperature sensor in hand, I used another Xenon as a peripheral that has a characteristic with `indicate` property. Here is the sample code making a Xenon as the peripheral:
```c
#include "application.h"

SYSTEM_MODE(MANUAL);

Serial1LogHandler log(115200, LOG_LEVEL_ALL);

const char* serviceUuid = "6E400001-B5A3-F393-E0A9-E50E24DCCA9E";
uint16_t txUuid = 0x2A1C;

BleCharacteristic txCharacteristic("tx",
                                   BleCharacteristicProperty::INDICATE,
                                   BleUuid(txUuid),
                                   BleUuid(serviceUuid));

void setup() {
    LOG(TRACE, "Application started.");

    BLE.addCharacteristic(txCharacteristic);

    BleAdvertisingData data;
    data.appendServiceUUID(0x1809);
    BLE.advertise(&data);
}

void loop() {
    if (BLE.connected()) {
        txCharacteristic.setValue("Hello");
        delay(5000);
    }
}
```
The corresponding log on the central device:
>0000008131 [app] INFO: successfully connected!
0000008142 [app] INFO: prop 32
0000010480 [wiring.ble] TRACE: onDataNotified, connection: 0, attribute: 24
0000010481 [app] INFO: temperature[0]=48
0000010481 [app] INFO: temperature[1]=65
0000010487 [app] INFO: temperature[2]=6c
0000010493 [app] INFO: temperature[3]=6c
0000010493 [app] INFO: temperature[4]=6f
0000015565 [wiring.ble] TRACE: onDataNotified, connection: 0, attribute: 24
0000015566 [app] INFO: temperature[0]=48
0000015566 [app] INFO: temperature[1]=65
0000015569 [app] INFO: temperature[2]=6c
0000015578 [app] INFO: temperature[3]=6c
0000015578 [app] INFO: temperature[4]=6f
0000020650 [wiring.ble] TRACE: onDataNotified, connection: 0, attribute: 24
0000020651 [app] INFO: temperature[0]=48
0000020651 [app] INFO: temperature[1]=65
0000020661 [app] INFO: temperature[2]=6c
0000020662 [app] INFO: temperature[3]=6c
0000020662 [app] INFO: temperature[4]=6f

I guess your Xenon were running a system firmware that is upper than 1.3.0, while your application is compiled against 1.3.0. In that case, BLE would not work properly as the BLE event dispatching mechanism has been reworked in the latest develop branch. Kindly refer to the description of this PR: https://github.com/particle-iot/device-os/pull/1847

sviraaj, 2019-08-19:
@XuGuohui Thanks for your reply. I am building the system firmware+app from the modules subdir and flashing using the command,
`PARTICLE_SERIAL_DEV=/dev/ttyACM0 make all program-dfu APP=<APP-name>`
Tried clean build as well. Also tried from develop branch(which has the fixes for 1.3.1-rc.1 patch). Same issue persists. My peripheral is running "hts peripheral" from nrf5_sdk examples. Unfortunately I do not have another xenon with me right now to try with it as peripheral. Am I missing something?
I enabled logs from hal BLE. Although I am able to see other hal.ble logs, I am not seeing any logs from event handlers such as processGattClientEvents (using SerialLogHandler). Thanks.

XuGuohui, 2019-08-20:
To enable log in ISR, you need to uncomment this macro: https://github.com/particle-iot/device-os/blob/develop/wiring/src/spark_wiring_logging.cpp#L35 and rebuild your system firmware. I'm afraid that the peripheral device was not sending indication. It would be appreciated if you can post the log here.

sviraaj, 2019-08-20:
Hi @XuGuohui ,
I tried using the peripheral with nrfToolbox and I get the indications successfully.
Here is the log,
```
0000000227 [system] TRACE: Last reset reason: 120 (data: 0x00)                                                                    
0000000233 [hal.ble] INFO: hal_ble_stack_init().                                                                                 
0000000234 [hal.ble] INFO: APP RAM start: 0x20005e00                      
0000000239 [hal.ble] INFO: SoftDevice RAM end: 0x20005928                                                                       
0000000247 [hal.ble] INFO: BLE advertising interval: 100.000ms, timeout: 0ms.
0000000253 [net.ifapi] TRACE: LwIP started
0000000262 [ot.api] INFO: OpenThread version: OPENTHREAD/0.01.00; Xenon; Aug 20 2019 12:55:10
0000000264 [ot.api] INFO: Max transmit power: 8
0000000272 [net.th] INFO: Creating new LwIP OpenThread interface
0000000273 [net.ifapi] INFO: Netif th1 added
0000000276 [net.en] INFO: Creating Wiznet LwIP interface
0000000286 [net.ifapi] INFO: Netif en2 added
0000000286 [hal] TRACE: Heap: 69/164 Kbytes used
0000000294 [net.th] TRACE: OpenThread state changed: e0200
0000000296 [hal.ble] INFO: hal_ble_stack_init().
0000000305 [hal.ble] INFO: hal_ble_gatt_server_add_service().
0000000306 [hal.ble] INFO: Service handle: 14.
0000000309 [hal.ble] INFO: hal_ble_gatt_server_add_characteristic().
0000000317 [hal.ble] INFO: Characteristic value handle: 16.
0000000320 [hal.ble] INFO: Characteristic cccd handle: 0.
0000000328 [hal.ble] INFO: hal_ble_gatt_server_set_characteristic_value().
0000000339 [hal.ble] INFO: hal_ble_gatt_server_add_characteristic().
0000000339 [hal.ble] INFO: Characteristic value handle: 18.
0000000350 [hal.ble] INFO: Characteristic cccd handle: 19.
0000000350 [hal.ble] INFO: hal_ble_gatt_server_add_characteristic().
0000000361 [hal.ble] INFO: Characteristic value handle: 21.
0000000362 [hal.ble] INFO: Characteristic cccd handle: 0.
0000000372 [hal.ble] INFO: hal_ble_set_callback_on_periph_link_events().
0000000375 [system.nm] INFO: State changed: NONE -> DISABLED
0000000576 [comm] INFO: channel inited
0000000576 [hal.ble] INFO: hal_ble_stack_init().
0000000577 [hal.ble] INFO: hal_ble_set_callback_on_periph_link_events().                                               [31/93990]
0000000580 [app] INFO: BLE scan!
0000000591 [hal.ble] INFO: hal_ble_gap_start_scan().
0000000591 [hal.ble] INFO: | interval(ms)   window(ms)   timeout(ms) |
0000000602 [hal.ble] INFO:   100.000        50.000      5000
0000005605 [hal.ble] INFO: BLE GAP event: Scanning timeout
0000005605 [app] INFO: 5 devices found
0000005605 [app] INFO: Connect to rssi=-69 address=6B:36:A9:9A:7A:FE
0000005616 [hal.ble] INFO: hal_ble_gap_connect().
0000005634 [hal.ble] INFO: BLE GAP event: connected.
0000005635 [hal.ble] INFO: BLE role: 2, connection handle: 0
0000005635 [hal.ble] INFO: | interval(ms)  latency  timeout(ms) |
0000005646 [hal.ble] INFO:   36*1.25          0       500*10
0000005646 [hal.ble] INFO: Attempts to exchange ATT_MTU if needed.
0000005657 [wiring.ble] TRACE: New peripheral is connected.
0000005658 [wiring.ble] TRACE: Start discovering services.
0000005662 [hal.ble] INFO: hal_ble_gatt_client_discover_all_services().
0000005726 [hal.ble] INFO: BLE GATTC event: services discovered.
0000005816 [hal.ble] INFO: BLE GATTC event: services discovered.
0000005817 [wiring.ble] TRACE: Start discovering characteristics.
0000005822 [hal.ble] INFO: hal_ble_gatt_client_discover_characteristics().
0000005906 [hal.ble] INFO: BLE GATTC event: characteristics discovered.
0000006041 [hal.ble] INFO: BLE GATTC event: characteristics discovered.
0000006131 [hal.ble] INFO: BLE GATTC event: descriptors discovered.
0000006221 [hal.ble] INFO: BLE GATTC event: descriptors discovered.
0000006222 [hal.ble] INFO: hal_ble_gatt_client_discover_characteristics().
0000006311 [hal.ble] INFO: BLE GATTC event: characteristics discovered.
0000006401 [hal.ble] INFO: BLE GATTC event: descriptors discovered.
0000006401 [hal.ble] INFO: hal_ble_gatt_client_discover_characteristics().
0000006481 [hal.ble] INFO: Request to change ATT_MTU from 23 to 247
0000006482 [hal.ble] INFO: sd_ble_gattc_exchange_mtu_request() failed: 17
0000006491 [hal.ble] INFO: BLE GATTC event: characteristics discovered.
0000006581 [hal.ble] INFO: BLE GATTC event: characteristics discovered.
0000006716 [hal.ble] INFO: BLE GATTC event: descriptors discovered.
0000006717 [hal.ble] INFO: hal_ble_gatt_client_discover_characteristics().
[... 3 lines cut ...]
0000006987 [hal.ble] INFO: hal_ble_gatt_client_discover_characteristics().
0000007076 [hal.ble] INFO: BLE GATTC event: characteristics discovered.
0000007166 [hal.ble] INFO: BLE GATTC event: characteristics discovered.
0000007256 [hal.ble] INFO: BLE GATTC event: descriptors discovered.
0000007302 [hal.ble] INFO: Request to change ATT_MTU from 23 to 247
0000007303 [hal.ble] INFO: sd_ble_gattc_exchange_mtu_request() failed: 17
0000007346 [hal.ble] INFO: BLE GATTC event: descriptors discovered.
0000007436 [hal.ble] INFO: BLE GATTC event: descriptors discovered.
0000007437 [hal.ble] INFO: hal_ble_gatt_client_configure_cccd().
0000007526 [hal.ble] INFO: BLE GATT Client event: write with response completed.
0000007526 [hal.ble] INFO: hal_ble_gatt_client_configure_cccd().
0000007616 [hal.ble] INFO: BLE GATT Client event: write with response completed.
0000007616 [app] INFO: successfully connected!
0000007620 [wiring.ble] INFO: got char 0x2a1c
0000007628 [app] INFO: t prop 32
0000008121 [hal.ble] INFO: Request to change ATT_MTU from 23 to 247
0000008201 [hal.ble] INFO: BLE GAP event: exchange MTU response.
0000008201 [hal.ble] INFO: Effective ATT MTU: 23.
0000010676 [hal.ble] INFO: BLE GAP event: connection parameter update request.
0000010991 [hal.ble] INFO: BLE GAP event: connection parameters updated.
0000010991 [hal.ble] INFO: | interval(ms)  latency  timeout(ms) |
0000010997 [hal.ble] INFO:   800*1.25          0       400*10
```

Thanks

sviraaj, 2019-08-20:
Part of the issue I am facing is because of the following. I have noticed the following code.
https://github.com/particle-iot/device-os/blob/ed8030ebcf9723a23355dacc35b407dfe7a44f89/wiring/src/spark_wiring_ble.cpp#L1201
```
    if (enable) {
        if ((impl()->properties() & BleCharacteristicProperty::NOTIFY) == BleCharacteristicProperty::NOTIFY) {
            config.cccd_value = BLE_SIG_CCCD_VAL_NOTIFICATION;
        } else {
            config.cccd_value = BLE_SIG_CCCD_VAL_INDICATION;
        }
    } else {
        config.cccd_value = BLE_SIG_CCCD_VAL_DISABLED;
    }
```
The and operator on line 1201 is overloaded to perform or operation instead. Should the central not configure cccd for notify even though a device with multiple properties have notify as one of them? This check fails in the scenario where a characteristic has both lets say notify and read. The issue definitely is for me that cccd writes are not received on the peripheral. For notify this is the problem I have. For another characteristic which has indication property set, it looks like the peripheral is not getting a callback when cccd is set. I will confirm after some testing and close the issue if no other issue persists.

XuGuohui, 2019-08-21:
I'd say good catch! I was wondering whether it can also resolve the issue if you change this line: https://github.com/particle-iot/device-os/blob/develop/wiring/inc/spark_wiring_ble.h#L70 to
`static_cast<std::underlying_type<BleCharacteristicProperty>::type>(lhs) &`. Could you help verify it? Thanks in advance.

sviraaj, 2019-08-21:
@XuGuohui Yes, that does it as well. I will close the issue once the other part is solved as well. Thanks.

XuGuohui, 2019-08-21:
>For another characteristic which has indication property set, it looks like the peripheral is not getting a callback when cccd is set.

Is this the other part of issue you mentioned?

sviraaj, 2019-08-21:
Yes. For 2 different services each having a characteristic which can notify/indicate, characteristic cccd write for one is not being received. Although, I think this time it might be an issue on the peripheral (I am not sure). But, I want to be sure before I close this. Is that okay @XuGuohui ?

XuGuohui, 2019-08-22:
That's totally fine. It would still be possible that the second issue is encountered by the central. It's good to dig out what is going wrong.

sviraaj, 2019-08-31:
It seems, the other issue which I mentioned, was due the fact that the security on the peripheral was "just works" rather than open. @XuGuohui Is this expected? If yes, this issue can be closed. Thanks.

XuGuohui, 2019-09-01:
Currently our BLE doesn't support paring and bonding, so if the peripheral required paring as the premise to exchange data, then it would occure the second issue you mentioned.

closed 2019-09-01 by XuGuohui