particle-iot/device-os #1892: Electron 3G U270 does not report RSSI since 1.3.1-rc.1

ahk97, 2019-08-22:
## Bug Report

### Expected Behavior
CellularSignal sig = Cellular.RSSI();
sig.getStrength()
sig.getQuality()
sig.getAccessTechnology()

Should report meaningful values > 0

### Observed Behavior
In some cases (possibly when connected to 2G network) 
sig.getStrength=-1
sig.getQuality=-1
sig.getAccessTechnology=0

### Steps to Reproduce

Was working fine with FW 1.2.1
Broken with 1.3.1-rc.1

ahk97, 2019-08-22:
Could this be related to this fix?
https://github.com/particle-iot/device-os/pull/1886

ahk97, 2019-09-12:
I did some more testing. Issue is not related to the modem type, I can see it with the SARA G350 and the U270. Could be some timeout issue when getting the RSSI.
Here's what the code does: Every 60 Minutes this piece of code gets called from the loop:
```
			if (Particle.connected()) {
				CellularSignal sig = Cellular.RSSI();
       				 snprintf(INFOchar, 620, "FW: %s FreeMem: %d Up: %ds %.0f%%/%.0f%%/%d", (const char*)System.version(), System.freeMemory(), millis() / 1000, sig.getStrength(),sig.getQuality(),sig.getAccessTechnology());
				result = Particle.publish("Info", INFOchar, 300, PRIVATE);
			}
```
With FW 1.3.1 or 1.4.0
about 3 out of 10 times, the CellularSignal values come back as -1%/-1%/0.
With FW 1.2.1 it works correctly 100% of the time.

When I change the code to:
```
			if (Particle.connected()) {
				CellularSignal sig = Cellular.RSSI();
				if (sig.getStrength() < 0) {
            				delay(200);
			    		sig = Cellular.RSSI();
				}
				if (sig.getStrength() < 0) {
            				delay(200);
			    		sig = Cellular.RSSI();
				}
      				  snprintf(INFOchar, 620, "FW: %s FreeMem: %d Up: %ds %.0f%%/%.0f%%/%d", (const char*)System.version(), System.freeMemory(), millis() / 1000, sig.getStrength(),sig.getQuality(),sig.getAccessTechnology());
				result = Particle.publish("Info", INFOchar, 300, PRIVATE);
			}
```

The publish results almost always show a valid signal strength...

technobly, 2019-09-14:
@ahk97 Thank you very much for the detailed report.  I have identified the root cause here and will be creating a fix soon.  The issue is that 2G/3G modems have low power mode enabled by default and will enter this state after sitting idle for about 10 seconds.  When that happens, any AT command will initially be delayed while the modem transitions from idle to active.  The AT/OK check that was added has a 1 second delay, and will timeout.  This unfortunately prevents the system from getting the signal information after it's been sitting idle for a while, and is also why repeating the request when it fails causes it to succeed.

**As a workaround**, you may set the power level to high continuously to prevent the modem from entering a low power state.  *FYI: this API is NOT considered public and may change at any time.*

```c++
#include "cellular_hal.h"

SerialLogHandler logHandler(LOG_LEVEL_ALL);

SYSTEM_MODE(SEMI_AUTOMATIC)

#define FWVERSION (1)
char INFOchar[1024];

void setup() {
    // volatile setting, must be called before modem is initialized (Cellular.on() / Cellular.connect() / Particle.connect() )
    // FYI: this API is NOT considered public and may change at any time.
    cellular_set_power_mode(0, NULL); // forces high power mode continuously
    // cellular_set_power_mode(1, NULL); // forces low power mode when idle for about 10 seconds (this is the default as well so doesn't need to be called explicitly)
    
    Particle.connect();
}

void loop() {
    if (Particle.connected()) {
        CellularSignal sig = Cellular.RSSI();
        snprintf(INFOchar, 620, "V: %s FW: %s FreeMem: %d Up: %ds %.0f%%/%.0f%%/%d", FWVERSION, (const char*)System.version(), System.freeMemory(), millis() / 1000, sig.getStrength(),sig.getQuality(),sig.getAccessTechnology());
        Particle.publish("Info", INFOchar, 300, PRIVATE);
    }
    delay(20000);
}
```

ahk97, 2019-09-15:
I tried setting the high power mode and this does work fine.
The electrons I have tested, reported the RSSI every hour without problems.
Thanks.

closed 2019-09-17 by technobly