particle-iot/device-os #1900: Electron HTTP posts (via TCP) fail on newer OS versions

ColinVT, 2019-09-03:
## Bug Report

I outlined the issues in [this thread](https://community.particle.io/t/http-maybe-tcp-broken-in-1-3-1-rc-1-on-electron-and-on-all-versions-for-boron/51903/7) and it seems this issue is present in both threaded and non-threaded operation. To summarize the issue: longer HTTP messages fail to complete on newer deviceOS versions on the Electron, and don't seem to work on _any_ OS version on the Boron.

### Expected Behavior

Complete an HTTP post to the specified endpoint every ~30 seconds.

### Observed Behavior

The device fails to complete the post, and eventually the device will lose cloud connection, crash, etc.

### Steps to Reproduce
##### **Working example**:
Run test app on an electron running deviceOS release 1.1.0 (or below). The code runs successfully and posts complete as expected.

#### **Failure example**:
Run test app on an electron running deviceOS release 1.3.1-rc.1 (also observed on 1.2.1) or on a Boron with any deviceOS version. (Failure occurs whether logging is defined in HTTPClient.cpp or not).

### Test App
NOTE: Make sure to include the HTTPClient library in the project along with the below main.ino.

#### **main.ino**
```c
// This #include statement was automatically added by the Particle IDE.
#include <HttpClient.h>

STARTUP(startupFunction());
#if PLATFORM_ID == PLATFORM_ELECTRON_PRODUCTION
STARTUP(cellular_credentials_set("broadband", "", "", NULL));
#endif

SYSTEM_MODE(SEMI_AUTOMATIC);

extern void startupFunction();

static char json_str[2048];
int timeAlive_s = 0;
int lastPost_s = 0;

HttpClient http;

http_header_t headers[] = {
  { "Accept" , "*/*"},
  { NULL, NULL } // NOTE: Always terminate headers will NULL
};

http_request_t request;
http_response_t response;

void startupFunction() {
  System.enableFeature(FEATURE_RESET_INFO);
}

void setup()
{
  // SLAVE_SERIAL_CONSOLE.begin(57600);     //initialize uart port, disabled for demonstration purposes
  Serial.begin(115200);    //initialize debug port
  // SLAVE_SERIAL_CONSOLE.setup();

  #if PLATFORM_ID == PLATFORM_BORON
  Cellular.setActiveSim(INTERNAL_SIM);
  Cellular.clearCredentials();
  #endif

  Serial.println("Calling Cellular.connect.");

  Cellular.connect();

  while (!Cellular.ready()) {
    Serial.print(".");
    delay(333);
  };

  Serial.println("!");
  delay(100);

  // Connect to cloud. This is necessary because manual mode is turned on.
  Serial.print("Calling Particle.connect");

  Particle.connect();

  while (!Particle.connected()) {
    Serial.print(".");
    delay(333);
  };
  Serial.println("!");

  memset(json_str, 'A', 2048);    //Fill with 'A's to simulate data
  Serial.println("Waiting");
}

void loop()
{
  // Update clocks
  uint32_t loopTime_ms = millis();
  // Update the time alive counter, and test for flag updates
  if ((uint32_t)timeAlive_s*1000 <= loopTime_ms) {
    timeAlive_s++;
    Serial.print(".");
    if(timeAlive_s >= 30 && (timeAlive_s - 30) >= lastPost_s){
      Serial.println("!");
      Serial.println("Posting");
      http_send_data();
      lastPost_s = timeAlive_s;
      Serial.println("Waiting");
    }
  }
  Particle.process();
}

void http_send_data(){
  request.hostname = "YOUR_HOSTNAME";
  request.port = 80;
  request.path = "YOUR_PATH";
  request.body = json_str;

  http.post(request, response, headers);
}
```

ColinVT, 2019-09-11:
Can someone at particle check if this is resolved by [this fix](https://github.com/particle-iot/device-os/pull/1911)?

ColinVT, 2019-09-16:
> Can someone at particle check if this is resolved by [this fix](https://github.com/particle-iot/device-os/pull/1911)?

Just checked, this is still broken on 1.4.0.

mstan, 2019-09-16:
Hi Colin,

A couple of clarifying questions.

A) Is this issue occurring on 3G or LTE versions of your devices, or both?
B) When testing the fix in PR #1911 -- did you attempt to update both your Device OS AND user application? The fix in PR #1911 requires the user application be recompiled -- targeting v1.4.0 -- as it modifies a user application library as well.

ColinVT, 2019-09-16:
A.) Yes, however on the Electron, the issue is not present on devices running 1.0.1 and earlier, whereas it is present on the Boron for all deviceOS versions (at least 0.9.0 onwards, I haven't tested with any versions prior to that).

B.) Yes, I completed the update via USB using boron-system-part1@1.4.0.bin, boron-softdevice@1.4.0.bin, and boron-bootloader@1.4.0.bin, and then compiled the code using `particle compile boron --target 1.4.0` which completed successfully, after which I flashed that binary to the device. The device connects to particle cloud and I checked particle console and confirmed the deviceOS version had been updated to 1.4.0, however the error still occurs.

ColinVT, 2019-09-16:
For the endpoint, [here is a good option](https://webhook.site) so in the example code you would have `request.hostname = "webhook.site";` and `request.path = "/xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx";` (insert the path assigned to you when you visit webhook.site from your browser). By setting the message length to 256 instead of 2048 (change in two places in code), the code runs correctly, and you will see the posts appearing in your browser window. If you set it back to 2048 and change **nothing else**, you will see that the code fails and the requests do not appear in your browser window (unless running on an electron with 1.0.1 or earlier, in which case it works correctly at 2048 and even higher than that). As a test to 100% confirm that this is an issue of deviceOS, compile the example code for 1.0.1 and run it on an Electron on 1.0.1 and leave the message length at 2048, the posts will complete successfully,. Now compile the **exact same code** for 1.3.1-rc.1 and run on an Electron on 1.3.1-rc.1 and you will see the posts fail, and this is the same behavior that the Boron exhibits on all deviceOS versions.

ColinVT, 2019-09-17:
Update here:
Yesterday I tested only the Boron on 1.4.0 since previously the Electron and boron were failing in the exact same way (when the electron was on version 1.3.1), and because the Boron still failed exactly the same way, I assumed the electron would as well, **however** it seems that the electron posts correctly at 2048 characters on 1.4.0! (although it seems the first post of each boot fails with error `HttpClient Connection failed`). Unfortunately the Boron still behaves the exact same and it is still not possible to post all but pretty short HTTP posts. I am updating the title of the issue to reflect that it _seems_ the issue has been resolved on the Electron.

ColinVT, 2019-09-17:
### To make this absolutely as simple as possible to replicate, **here is a (literally) less-that-5-minute procedure to replicate this error with nothing but a Boron and a computer:**

1.) Open [this link](http://webhook.site/#!/8c2d48c8-51f0-41d8-a02d-8af007651ce4/8db7e18e-f00b-434f-bb2f-399a60449fd9/1) in your browser. This is the endpoint specified in the code included, or you can go to webhook.site to get your own unique endpoint path which you would paste into the code (line 81!) and recompile.

2.) Compile the [attached code](https://github.com/particle-iot/device-os/files/3623055/BoronHttpFailExample.zip) as-is (`MESSAGE_SIZE` is defined as 256) and run on a Boron, you will see the posts appear in the webhook.site browser window. (You can also monitor serial for more details via httpClient's logging)

3.) Now edit main.ino and change `MESSAGE_SIZE 256` to `MESSAGE_SIZE 2048`, recompile & flash and you will see that the posts are failing. (You can also monitor serial for more details via httpClient's logging)

[BoronHttpFailExample.zip](https://github.com/particle-iot/device-os/files/3623055/BoronHttpFailExample.zip)

technobly, 2019-09-17:
@ColinVT Thank you for the succinct details here, they will really help in resolving this issue!  I had this on my list today to reproduce and just now got to it.  Electron LTE / E Series on 1.4.0 does not have any issue I can see posting to an Echo server `httpbin.org/post` with 2048 bytes of data, and receiving that data back.

With Boron though, I have to build as a monolithic app to enable debug logging and I can see it trying to send the data, but then fails to be respond and subsequently gets power cycled by the system.
```
0000172071 [lwip] TRACE: No response to 10 echo-requests
0000172071 [lwip] TRACE: Serial link appears to be disconnected.
```
To compile as monolithic
```
main $ make clean all -s PLATFORM=boron COMPILE_LTO=n MODULAR=n DEBUG_BUILD=y APPDIR=~/code/apps/issue-1900 program-dfu
```

The app I used is basically the same one you provided with a little tweak to get it to compile locally
```c++
#include "Particle.h"
// This #include statement was automatically added by the Particle IDE.
#include <HttpClient.h>

STARTUP(System.enableFeature(FEATURE_RESET_INFO);)
// #if PLATFORM_ID == PLATFORM_ELECTRON_PRODUCTION
// STARTUP(cellular_credentials_set("broadband", "", "", NULL));
// #endif

SYSTEM_MODE(SEMI_AUTOMATIC);

SerialLogHandler logHandler(LOG_LEVEL_ALL);

static char json_str[2048];
int timeAlive_s = 0;
int lastPost_s = 0;

HttpClient http;

http_header_t headers[] = {
  { "Accept" , "*/*"},
  { NULL, NULL } // NOTE: Always terminate headers will NULL
};

http_request_t request;
http_response_t response;

void http_send_data(){
  request.hostname = "httpbin.org";
  request.port = 80;
  request.path = "/post";
  request.body = json_str;

  http.post(request, response, headers);
}

void setup()
{
  // SLAVE_SERIAL_CONSOLE.begin(57600);     //initialize uart port, disabled for demonstration purposes
  Serial.begin(115200);    //initialize debug port
  // SLAVE_SERIAL_CONSOLE.setup();

#if PLATFORM_ID == PLATFORM_BORON
  // Cellular.setActiveSim(INTERNAL_SIM);
  // Cellular.clearCredentials();
#endif

  Serial.println("Calling Cellular.connect.");

  Cellular.connect();

  while (!Cellular.ready()) {
    Serial.print(".");
    delay(333);
  };

  Serial.println("!");
  delay(100);

  // Connect to cloud. This is necessary because manual mode is turned on.
  Serial.print("Calling Particle.connect");

  Particle.connect();

  while (!Particle.connected()) {
    Serial.print(".");
    delay(333);
  };
  Serial.println("!");

  memset(json_str, 'A', 2048);    //Fill with 'A's to simulate data
  Serial.println("Waiting");
}

void loop()
{
  // Update clocks
  uint32_t loopTime_ms = millis();
  // Update the time alive counter, and test for flag updates
  if ((uint32_t)timeAlive_s*1000 <= loopTime_ms) {
    timeAlive_s++;
    Serial.print(".");
    if(timeAlive_s >= 30 && (timeAlive_s - 30) >= lastPost_s){
      Serial.println("!");
      Serial.println("Posting");
      http_send_data();
      lastPost_s = timeAlive_s;
      Serial.println("Waiting");
    }
  }
  Particle.process();
}
```

ColinVT, 2019-09-17:
Awesome! Glad it was able to be reproduced and is on the radar! This is the only thing blocking us from moving forward with the plan to upgrade our fleet, so it will be a huge relief once it is addressed, thank you!!

ColinVT, 2019-09-30:
@technobly have you heard anything back about this from u-Blox?