particle-iot/device-os #1904: Where to find actual API description for 1.4.0 version? Some methods stops to work, example was taken from official API reference

Saynok, 2019-09-04:
heartRateMeasurementCharacteristic = peer.getCharacteristicByUUID(BleUuid(0x1530));

Error:  no matching function for call to 'particle::BlePeerDevice::getCharacteristicByUUID(particle::BleUuid)'

rickkas7, 2019-09-04:
Instead of returning the characteristic, it now is passed in the first parameter:

```
peer.getCharacteristicByUUID(heartRateMeasurementCharacteristic, BleUuid(0x2a37));
```

The API docs and examples will be updated 1.4.0 is released (not just rc version). The definition is here:

https://github.com/particle-iot/device-os/blob/develop/wiring/inc/spark_wiring_ble.h#L480

closed 2019-09-04 by rickkas7

Saynok, 2019-09-04:
thanks a lot! 
I have new problem: got no data by notifications.

Saynok, 2019-09-04:
Found   int subscribe(bool enable) const;
Thanks again :)

XuGuohui, 2019-09-05:
@Saynok If your device connect to peer device using the default parameter `connect(const BleAddress& addr, bool automatic = true)`, it will automatically subscribe the characteristic's notification and indication if presented. No?