particle-iot/device-os #1906: Xenon: No recorded vitals for this device but vitals do report. Since 1.3.0-rc1

kamotswind, 2019-09-04:
##  Bug Report
Tested on 1.3.0-rc1 and 1.4.0-rc1. The Particle Console reports "No recorded vitals for this device." but vitals are being reported. Clicking the refresh button on the vitals section causes the device to send vitals, [diagnostics.txt](https://github.com/particle-iot/device-os/files/3576753/diagnostics.txt), but the site says the device failed to report vitals after the timeout period.

This may just be a Particle Console issue instead of a firmware issue but I think something changed in firmware since it had been working on 0.8.0-rc27. 

### Expected Behavior

Display vitals/diagnostic data.

### Observed Behavior

Particle Console reports "No recorded vitals for this device."

### Steps to Reproduce

Described above. Tested with 1.3.0-rc1 and 1.4.0-rc1 along with 0.8.0-rc27 (I think) where it did work.

### Test App

Any application with cloud enabled.

### References
[diagnostics.txt](https://github.com/particle-iot/device-os/files/3576753/diagnostics.txt)

zfields, 2019-09-05:
Thank you for reporting! We have an internal ticket tracking this item, and we have linked your issue! I'll put a note to provide visibility here once this issue has been root caused and corrected!

kamotswind, 2019-09-06:
@zfields Sounds good, thank you! Please let me know if you need any more information/troubleshooting from my end. I did install 1.4.0-rc1 on an Electron and Photon to test and they are not showing the same issue. It looks to be only impacting Xenon devices.

zfields, 2019-09-06:
That matches our current understanding as well. Thank you again for the diligence!

zfields, 2019-09-16:
Update: We have confirmed device vitals are being generated and reported by the device. At this time, it appears those values are not being rendered by the console. We are continuing to investigate.

closed 2020-03-03 by avtolstoy