particle-iot/device-os #1919: retained int value is not set and retrieved correctly

shanevanj, 2019-09-19:
<details>
  <summary><i>retained int returns wrong value on first use and after system reset</i></summary>

</details>

## Bug Report

### Expected Behavior

Serial output:
Starting retained var test
savedValue=0
savedValue=1
savedValue=2
savedValue=3
savedValue=4

etc...

### Observed Behavior

Serial output:
Starting retained var test
savedValue=-778261199
savedValue=-778261198
savedValue=-778261197
savedValue=-778261196
savedValue=-778261195
savedValue=-778261194
savedValue=-778261193
savedValue=-778261192
savedValue=-778261191

### Steps to Reproduce

Firmware 1.4.0
flash Test app

### Test App

```

STARTUP(System.enableFeature(FEATURE_RETAINED_MEMORY)); 
retained int savedValue = 0; 

void setup() {
    Serial.begin(115200);
    
    while (!Serial.isConnected())
    Particle.process();
    Serial.printf(
      "\nStarting retained var test");
} 

void loop() { 
    Serial.printlnf("savedValue=%d", savedValue++); 
    delay(1000);
}

```

### References

Links to the Community, Docs, Other Issues, etc..

---

## Feature/Enhancement Request

retain an int value for use between soft restarts

shanevanj, 2019-09-19:
Community thread

> https://community.particle.io/t/retained-memory-tips/38137

technobly, 2019-11-22:
Thanks @shanevanj for the report.  I have confirmed this is not initializing as it should on Argon and with the latest firmware 1.4.2.  We will look into it.

PR #2001 (2020-01-27): [Gen 3] Fix retained values not initializing correctly [ch42294]
<details>
  <summary><i>submission notes</i></summary>

```
**Important:** Please sanitize/remove any confidential info like usernames, passwords, org names, product names/ids, access tokens, client ids/secrets, or anything else you don't wish to share.

Please Read and Sign the Contributor License Agreement ([Info here](https://github.com/spark/firmware/blob/develop/CONTRIBUTING.md)).

You may also delete this submission notes header if you'd like. Thank you for contributing!
```
</details>

### Problem

Variables that are defin [... cut]
closed 2020-01-27 by keeramis