particle-iot/device-os #1922: TCP/UDP Stuttering

tntman42, 2019-09-21:
## Bug Report

### Expected Behavior

Near real-time communication with a Linux computer (Ubuntu 16.04 LTS) using either UDP or TCP.

This is also happening at the same time the Argon board is running multiple threads to get data from an IMU and write pixel data to an OLED. The threading was necessary to keep things running in reasonable amounts of time.

### Observed Behavior

The argons would recieve packets in "bursts", meaning they would get all the packets from a certain period of time all at once. This was not good for getting nearly real-time data.

### Steps to Reproduce

Firmware Versions:
1.0.0 - 1.2.0

We set up a client of either TCP or UDP, and also enabled threading, and attempting to connect to the server

### Test App

Argon Code:

```c
#include <neopixel.h>
#include <MPU6050/MPU6050.h>
#include <Adafruit_mfGFX.h>
#include <Adafruit_SSD1351_Photon.h>
#include <I2Cdev.h>
SYSTEM_THREAD(ENABLED)

#include <stdlib.h>
#include <string.h>
#include <math.h>
#include <Wire.h>
#include "spark_wiring.h"
#include <Particle.h>

#include "screen.h"
#include "drivetrain.h"
#include "easy_tcp.h"
#include "imu_calibrate.h"
#include "charger_status.h"
#define INTEGRATED_DEBUG 1
#define COMMAND_DEBUG 0
// // FOR AN ARGON BOARD
// #define mosi D12 //blue - DIN - MO on Argon Board
// #define sclk D13 //yellow
// #define cs A5    //orange
// #define dc D4    //green
// #define rst D5   //white

// IMPORTANT: Set pixel COUNT, PIN and TYPE
#define PIXEL_PIN D7
#define PIXEL_COUNT 1
#define PIXEL_TYPE WS2812B

#define NETWORK_LED 0

Adafruit_NeoPixel strip(PIXEL_COUNT, PIXEL_PIN, PIXEL_TYPE);

typedef struct command {
    char str[64];
} command;

Thread oledThread; //("testThread", threadFunction);
void threadOled(void);
DiffDrive diff_drive;
IMUCalibrate imu;
Screen screenObject;
[... 764 lines cut ...]
// waiting for all client handling to die
#if DEBUG_CPP
        puts("Waiting for threads to join");
#endif
	for (pthread_t tid : threads)
	{
		pthread_join(tid, NULL); // waiting for threads to die
	}

	close(socket_descriptor);
	return 0;
}
#endif

```

### References

This is the exact same issue as [https://community.particle.io/t/udp-delays-on-argon/50917/11](https://community.particle.io/t/udp-delays-on-argon/50917/11)

---

## Feature/Enhancement Request

We were hoping that the UDP/TCP could be looked into, and there could be some kind of resolution to the issue. The actual usage does not have to be "thread safe" but it would be nice if threads could be near by.

nhewitt99, 2019-10-16:
Is anybody available to comment on this problem? @rickkas7 was able to confirm the issue [in the post linked above](https://community.particle.io/t/udp-delays-on-argon/50917/11), and since then the same issue with UDP has been echoed by more users in that forum thread. Concise test code is available there to replicate the issue without the overhead of the source code above. Thanks!

elcojacobs, 2019-11-13:
I have been struggling with networking issues on the Photon for a long time. A new theory I have based on latest tests is that the lack of any delays in my code might cause the system thread to get very little execution time and does not handle the network traffic properly.

Try if adding a 1 ms delay in your app loop helps, because it yields a thread switch.

technobly, 2019-11-22:
Thanks everyone for the wealth of info here, including Rick's tests in the community.  We will look into this.

pierrep, 2019-12-03:
Just wanting to say I'm REALLY looking forward to a fix for this, it's affecting my development significantly. Thanks for looking into it.