particle-iot/device-os #1923: Scan failure/timeout results in hung BLE state

sviraaj, 2019-09-23:
<details>
  <summary><i> Setup: Xenon in central mode. Connecting multiple peripherals to xenon.
When scan fails due to data notified/indicated before scan timeout, further scans or BLE data processing is not possible, due to scan hang. This is attributed to https://github.com/particle-iot/device-os/blob/2896ead2f717e5cdea2f733e502593064ee8c125/hal/src/nRF52840/ble_hal.cpp#L1553 where scan semaphore is released only in success case. IMO, the scan semaphore should be released even during failure.</i></summary>

</details>

## Bug Report

### Expected Behavior

Scan failure should not result in scan being blocked. It should release semaphore and we should be able to retry again.

### Observed Behavior

Scan blocks due to semaphore never being released.

### Steps to Reproduce
Can  be reproduced in the case where we are scanning for device and at the same time data is notified by another peripheral before scan timeout.

### Test App

### References
https://github.com/particle-iot/device-os/blob/2896ead2f717e5cdea2f733e502593064ee8c125/hal/src/nRF52840/ble_hal.cpp#L1553

XuGuohui, 2019-09-27:
I don't think we should release the scanning semaphore in case of any GAP error, despite of the reason. I have been running these test application under this PR: https://github.com/particle-iot/device-os/pull/1929 for a long period and I can't reproduce the issue as you described. 

I do see some topics on the Nordic forum about the GAP timeout reason and sometimes the SoftDevice may return `BLE_GAP_TIMEOUT_SRC_CONN` as the reason of BLE scanning timeout. But anyway, it should be the issue of the SoftDevice and it's probably fixed by updating the SoftDevice.

sviraaj, 2019-09-27:
@XuGuohui Okay, that sounds reasonable. But I think there should be some timeout on the scan semaphore in case it is not released for any reason. This will prevent the scan call from blocking indefinitely. The semaphore timeout can be greater than the scan timeout. Let me know what you think. I was observing BLE_GAP_TIMEOUT_SRC_CONN when the scan was ongoing and there is a data notify from other peripheral at the same time, before scan timed out. Thanks.

XuGuohui, 2019-09-27:
I'd say setting a timeout for taking the scan semaphore is not applicable to the case that if user application sets the scan timeout to be indefinite. Maybe changing the BLE scanning behavior to be unblocking is an alternative solution. In that case, that's a new feature instead of an issue and should it should be in a separated issue or PR whatever.

Let's just keep this issue open as a reminder, until we completely fix it.

XuGuohui, 2020-10-14:
@sviraaj We're now revisiting this issue and I'm trying to replicate it. I have an Argon that acts as a BLE central and a Boron that acts as a BLE peripheral. The peripheral will keep sending data to the central as long as it is connected. The central keeps scanning nearby BLE devices and connects to the peripheral once it is discovered and it continue scanning BLE devices while receiving data that is notified from the peripheral. I had run the test for a mount of time but the central didn't get stuck as there is no failure encountered. So I was wondering how easily I can replicate the issue so that I can find the root cause to the issue.

sviraaj, 2020-10-14:
Hi @XuGuohui . It has been a very long time, sorry if I am unable to recollect everything. 
To reproduce this, you will need multiple peripherals (not just one). The scenario is something like this.
1. Scanning -> Scan timeout -> Connect peripheral 1.
2. **Scanning -> Data notification from connected peripheral 1 (before scan timeout has occurred**) -> BLE_GAP_TIMEOUT_SRC_CONN -> scanSemaphore_ not released. (No other event after this, such as scanTimeout)
3. BLE.Scan() hang waiting for release of scanSemaphore_ on next scan loop.

With more peripherals, it is easier to reproduce (Easier to hit the scenario outlined above). I had tried this is with up to 8 peripherals together.

Sample reference code (Just the loop, I had used something very similar to this):
    ```
    if (BLE.connected())
    {
        int i = 0;
        for (auto& p : BLE.getAllPeers())
        {
            Log.info("%d: " + p.address().toString(true), p.connHandle());
        }
    }
    if (BLE.getAllPeers().size() < MAX_BLE_CONNECTIONS)
    {
        Log.info("BLE scan!");
        // Only scan for 500 milliseconds
        BLE.setScanTimeout(50);
        // We are not connected to a sensor, scan for one
        Vector<BleScanResult> scanResults = BLE.scan();
        int count = scanResults.size();

        if (scanResults.size())
        {
            Log.info("%d devices found", scanResults.size());
        }

        for (int ii = 0; ii < count; ii++)
        {
            // uint8_t buf[BLE_MAX_ADV_DATA_LEN];
            size_t svcCount;
            BleUuid foundServiceUuid;
            bool already_connected = false;

            /* Check if this device previously connected and is still a valid device */
            for (auto& p : BLE.getAllPeers())
            {
                if ((scanResults[ii].address == p.address()) && (p.connHandle() != BLE_INVALID_CONN_HANDLE))
                {
                    Log.warn("Tried connecting to already connected device. Something wrong!");
                    already_connected = true;
                    break;
                }
            }

            if (already_connected)
            {
                continue;
            }

            // We're looking for devices with specific service.
            svcCount = scanResults[ii].scanResponse.serviceUUID(&foundServiceUuid, 1);
            if (svcCount > 0 && foundServiceUuid == ServiceUuid)
            {

                Log.info("Connect to rssi=%d address=%02X:%02X:%02X:%02X:%02X:%02X ",
                         scanResults[ii].rssi, scanResults[ii].address[0],
                         scanResults[ii].address[1], scanResults[ii].address[2],
                         scanResults[ii].address[3], scanResults[ii].address[4],
                         scanResults[ii].address[5]);

                peer = BLE.connect(scanResults[ii].address);
                if (peer.connected())
                {
                    bool char_avail = false;
                    Log.info("successfully connected!");

                    // Get the Data measurement characteristic
                    char_avail =
                        peer.getCharacteristicByUUID(dataMeasurementCharacteristic, DataUuid);
                    // dataMeasurementCharacteristic.onDataReceived(onDataReceived, NULL);

                    // Get the cfg characteristic
                    char_avail = peer.getCharacteristicByUUID(cfgCharacteristic, CfgUuid);
                }
                else
                {
                    Log.info("connection failed");
                }
            }
            delay(1000);
        }
    }
```

Let me know if I can add anything else. Thanks.

XuGuohui, 2020-10-14:
>I had tried this is with up to 8 peripherals together.

This should not be achievable technically unless you modify the device-os source code to support this amount of links. The device-os currently support 3 peripheral links simultaneously, otherwise, the RAM for BLE link informations will probably corrupted.

sviraaj, 2020-10-15:
> > I had tried this is with up to 8 peripherals together.
> 
> This should not be achievable technically unless you modify the device-os source code to support this amount of links. The device-os currently support 3 peripheral links simultaneously, otherwise, the RAM for BLE link informations will probably be corrupted.

I did change the softdevice_ram_size to support 8 peripherals.[](
(https://github.com/particle-iot/device-os/commit/00304d54f4be3870daad0023b1a138ec32fa1683))

XuGuohui, 2020-10-15:
A pull request has been created to try fixing the issue: https://github.com/particle-iot/device-os/pull/2220

avtolstoy, 2020-12-03:
Closing because #2220 has been merged.

closed 2020-12-03 by avtolstoy