particle-iot/device-os #1924: Resetting the BLE notify callback in central mode upon new peripheral connection

sviraaj, 2019-09-23:
<details>
  <summary><i>submission notes</i></summary>

</details>

## Bug Report

Setup: Multiple peripheral to single central (xenon)

Whenever there is a new peripheral connection, the callback for the same characteristic of previous peripheral (peer) is set to null which results in the app callback not being called for those peripherals. Is there a reason why inheritedCallback resets the callbackptr (https://github.com/particle-iot/device-os/blob/2896ead2f717e5cdea2f733e502593064ee8c125/wiring/src/spark_wiring_ble.cpp#L805)?

### Expected Behavior

The app callback should be called for all peripherals upon receiving data notify for characteristic.

### Observed Behavior

The callback is invoked only for the last peripheral connected.

### Steps to Reproduce
Connect multiple peripherals, upon data notify, the application callback will not be invoked in case of the peripherals which connected earlier than the last peripheral. This is due to the inheritCallback (https://github.com/particle-iot/device-os/blob/2896ead2f717e5cdea2f733e502593064ee8c125/wiring/src/spark_wiring_ble.cpp#L805) setting the callback to null.

Using the following possible workarounds currently:
1. Execute the below code upon every connection before getCharacteristic
`temperatureMeasurementCharacteristic.onDataReceived(onDataReceivedTemp, NULL);`

2. Do not reset the callback ptr in inheritCallback (https://github.com/particle-iot/device-os/blob/2896ead2f717e5cdea2f733e502593064ee8c125/wiring/src/spark_wiring_ble.cpp#L805).

### Test App
#include "application.h"

SYSTEM_MODE(MANUAL);

SerialLogHandler logHandler(LOG_LEVEL_ALL);

const size_t SCAN_RESULT_MAX = 30;

BleCharacteristic temperatureMeasurementCharacteristic;

BleScanResult scanResults[SCAN_RESULT_MAX];
BlePeerDevice peer;

void onDataReceivedTemp(const uint8_t* data, size_t len, const BlePeerDevice& peer, void* context);

void setup()
{
    (void)logHandler; // Does nothing, just to eliminate the unused variable warning

    temperatureMeasurementCharacteristic.onDataReceived(onDataReceivedTemp, NULL);
}

void loop()
{
    if (BLE.connected())
    {
    }
    //else
    {
        Log.info("BLE scan!");
        int count = BLE.scan(scanResults, SCAN_RESULT_MAX);

        if (count)
        {
            Log.info("%d devices found", count);
        }

        for (int ii = 0; ii < count; ii++)
        {
            uint8_t buf[BLE_MAX_ADV_DATA_LEN];
            size_t len;

            // We're looking for devices that have a temperature service (0x1809)
            len = scanResults[ii].advertisingData.get(
                BleAdvertisingDataType::SERVICE_UUID_16BIT_COMPLETE, buf, BLE_MAX_ADV_DATA_LEN);
            if (len > 0)
            {
                for (size_t jj = 0; jj < len; jj += 2)
                {
                    if (*(uint16_t*)&buf[jj] == BLE_SIG_UUID_HEALTH_THERMONETER_SVC)
                    { // 0x1809
                        // Found a device with a temperature service

                        Log.info("Connect to rssi=%d address=%02X:%02X:%02X:%02X:%02X:%02X ",
                                 scanResults[ii].rssi, scanResults[ii].address[0],
                                 scanResults[ii].address[1], scanResults[ii].address[2],
                                 scanResults[ii].address[3], scanResults[ii].address[4],
                                 scanResults[ii].address[5]);

                        peer = BLE.connect(scanResults[ii].address);
                        if (peer.connected())
                        {
                            Log.info("successfully connected!");

                            // Get the temperature measurement characteristic
                            temperatureMeasurementCharacteristic = peer.getCharacteristicByUUID(BleUuid(0x2A1C));
                            //peer.getCharacteristicByUUID(temperatureMeasurementCharacteristic, BleUuid(0x2A1C));
                            Log.info("prop %d", temperatureMeasurementCharacteristic.properties());                   
                        }
                        else
                        {
                            Log.info("connection failed");
                        }
                    }
                }
            }
        }
    }
}

void onDataReceivedTemp(const uint8_t* data, size_t len, const BlePeerDevice& peer, void* context)
{
    for (int i = 0; i < len; i++)
    {
        Log.info("temperature[%d]=%x", i, data[i]);
    }
}
### References

https://github.com/particle-iot/device-os/blob/2896ead2f717e5cdea2f733e502593064ee8c125/wiring/src/spark_wiring_ble.cpp#L805

---

XuGuohui, 2019-09-23:
I think this will be resolved in this PR:https://github.com/particle-iot/device-os/pull/1915/files#diff-93f9dd9012bb15fde13e60440775ab25L805.

sviraaj, 2019-09-23:
@XuGuohui Yes, I think that should do it. Thanks. 👍

PR #1915 (2019-09-27): BLE: fixes that setValue() doesn't work with BleTxRxType::AUTO if characteristic has WRITE property.
### Problem

https://github.com/particle-iot/device-os/issues/1913

### Solution

Send characteristic value, which property is `WRITE`, when `BleTxRxType::AUTO` is used by default.

### Steps to Test

1. Build and flash the test applications `user/tests/wiring/ble_central_peripheral` to two Gen3 devices respectively.
2. Simply connect them to serial terminals and type `t` on the terminal to start the test.

### Example App

N/A

### References

- Closes #1913
- Closes #1924

### Completeness

- [... cut]
closed 2019-09-27 by XuGuohui