particle-iot/device-os #1925: Missing API for BLE max packet size

bbarwik, 2019-09-23:
## Feature request

In system/src/ble_control_request_channel.cpp there's a function
```
int BleControlRequestChannel::gattParamChanged(const hal_ble_link_evt_t& event) {
    if (event.conn_handle == curConnHandle_) {
        maxPacketSize_ = event.params.att_mtu_updated.att_mtu_size - BLE_ATT_OPCODE_SIZE - BLE_ATT_HANDLE_SIZE;
        DEBUG("maxPacketSize_: %d", maxPacketSize_);
    }
    return 0;
}
```

It would be useful to have possibility to get access to maxPacketSize_ from BLE api, because knowlage about max packet is necessary to achieve high throughput.
Also information about current connection interval would be useful.
```int BleObject::ConnectionsManager::getPpcp(hal_ble_conn_params_t* ppcp) const {
    CHECK_TRUE(ppcp, SYSTEM_ERROR_INVALID_ARGUMENT);
    ble_gap_conn_params_t bleGapConnParams = {};
    int ret = sd_ble_gap_ppcp_get(&bleGapConnParams);
    CHECK_NRF_RETURN(ret, nrf_system_error(ret));
    *ppcp = toHalConnParams(&bleGapConnParams);
    return SYSTEM_ERROR_NONE;
}
```
This function should be somehow available in BLE api.

XuGuohui, 2019-09-27:
Currently all the features you mentioned are now available in BLE HAL and ready to be exposed for the BLE wiring APIs:
- Setting desire ATT MTU
- Getting connection parameters
- Callback on ATT MTU changed

These probably should be available since v1.5.0-rc.1.

shaneqld, 2021-08-02:
@XuGuohui Are the changes you mentioned available in any release yet? I'm looking for a way to negotiate a larger MTU. Thanks!

XuGuohui, 2021-08-03:
@shaneqld Sorry about that the mentioned features haven't been included in any release yet. I'll raise this again internally.

XuGuohui, 2021-08-05:
I'm sorry that we've decided to de-prioritize this feature request, but this request is queued up for sure.