particle-iot/device-os #1932: attachInterrupt() fails for pin D7 on Boron

Otterverse, 2019-10-01:
## Bug Report

### Expected Behavior

attachInterrupt(D7, myFunc, RISING) == true;

### Observed Behavior

attachInterrupt(D7, myFunc, RISING) == false;

### Steps to Reproduce

Using firmware 1.4.0, attempt to attach an interrupt to D7 on a Boron device and it will return false.

I'm almost positive this is due to legacy (electron specific if my research is correct) code in spark_wiring_interrupts.cpp which has this catch in the attachInterrupt() definition:

```
#if Wiring_Cellular == 1
  /* safety check that prevents users from attaching an interrupt to D7
   * which is shared with BATT_INT_PC13 for power management */
  if (pin == D7) return false;
#endif 
```
And while the docs for the Electron clearly label this pin as off limits for that device, the docs for Boron state:

```
All A and D pins (including TX, RX, and SPI) on Gen 3 (mesh) devices can be used for interrupts...
```
I'm fairly certain Wiring_Cellular still gets set on the Boron though.

avtolstoy, 2019-10-01:
Thanks for submitting the issue! Yes, this is an oversight and D7 should be available for `attachInterrupt` on Gen3 devices. This should be fixed before 1.4.1 default.

Otterverse, 2019-10-01:
Wow! That is the fastest response I've ever had to a bug report! Thank you, I'll look forward to the update.

technobly, 2019-10-01:
D7 is excluded on Gen 2 because of the shared interrupt with USART 3 or PMIC used for connection the the modem on Electron. Just wanted to double check that D7 is not shared on Gen 3 in any similar way.

closed 2019-10-14 by artyLee