particle-iot/device-os #1934: Feature request: Put modem in standby without sleeping MCU

JDSymington, 2019-10-06:
### Request:
On some of the sleep modes available for the electron it is possible to set the SLEEP_NETWORK_STANDBY flag. This will put the modem in standby mode before putting the MCU to sleep.

It would be really handy to put the modem in standby mode but keep the MCU active. This would drastically reduce power consumption and when data is ready to be sent a connection to the operator can quickly be made.

### Some background information:
I built a solar powered weather station also measuring wind speed. Anemometer rotations are counted using a counter IC. Every 5 seconds the Electron is woken from sleep, reads and clears the counter and goes back to sleep. After 5 minutes of collecting data, the average, maximum and minimum wind speed are calculated. The modem is turned on and the results are published to a web server. Then the modem is turned off and the Electron goes back to sleep. The cycle repeats.

The problem with this approach it that it takes at least 20 seconds to connect to the cellular network and a lot of data is used in the process.

The alternative is not to sleep at all, but this is really power hungry which is a problem for a solar powered system.

### My  proposed solution:
Put the modem in standby mode. When the Electron is woken up every 5 seconds to read the counter the modem should remain in standby mode. When 5 minutes passed, the modem should come out of standby mode, publish data and then go back to standby.

Currently the modem only remains in standby while the Electron is sleeping.

Any advice on how to achieve this would be appreciated.

technobly, 2019-10-06:
@JDSymington have you proposed this idea on the [Particle community](https://community.particle.io/) yet?  I would encourage you to do so to engage the "hive mind" there, and it's really a better place to get more feedback on these types of application use case requests.  That said I'll take a stab at this, but I can't promise I will be able to respond quickly in the future ^_^

The Gen 2 Electron will enter a low power state automatically after about 10 seconds of being idle.  It kind of sounds like you just want to use SYSTEM_MODE(SEMI_AUTOMATIC) so that you don't accidentally wake up the modem every 5 seconds when you wake from sleep.  Use SLEEP_NETWORK_STANDBY to prevent the modem from being turned off, it will enter a low power state automatically, and when you need to publish after 5 minutes, just do a Particle.connect() and publish your data and go back to sleep.

JDSymington, 2019-10-06:
@technobly thanks for your quick reply. I haven't proposed it on the community yet. Will do so now.

I'm already using SYSTEM_MODE(SEMI_AUTOMATIC). The problem is that even when using this, the modem is still taken out of standby as soon as the Electron wakes up. As far as I know SYSTEM_MODE(SEMI_AUTOMATIC) only prevents the modem from switching itself on.

technobly, 2019-10-06:
Hmm, I wouldn't expect it to wake up the modem... but maybe there is something that waking from sleep mode is doing to the RTS/CTS/DTR lines that causes that to happen.  It should not be sending any AT commands, which would also cause a wake up.  Another thing to try is SYSTEM_MODE(MANUAL) but this causes more code to be written on your part to make the connection and background processing (just something to be aware of).  Perhaps you can take a look at the low level AT command logging `SerialLogHandler logHandler(LOG_LEVEL_ALL);` with system binaries from github or CLI that have `DEBUG_BUILD=y` enabled and see if there is in fact no AT commands every 5 seconds when you wake.

JDSymington, 2019-10-06:
I wrote a simple piece code that illustrates the problem.

```C++
SYSTEM_MODE(MANUAL);
STARTUP(cellular_credentials_set("INTERNET", "", "", NULL));
SerialLogHandler logHandler(LOG_LEVEL_ALL);

void setup()
{
  Serial.begin();

  if (!Cellular.ready())
    Cellular.connect();

  if (!waitFor(Cellular.ready, 180000))
    System.reset();

}

void loop()
{
  System.sleep(0, FALLING, SLEEP_NETWORK_STANDBY, 10);
  delay(1000);
  Serial.println("Awake.");
  delay(8000);
  Particle.process();
  Serial.println("Going to sleep.");
  delay(1000);
}
```

Basically I'm putting the system in manual mode and turning the modem on. Then the electron is going into a loop where it sleeps for 10 seconds with the network in standby mode. It then wakes up for 10 seconds. 

What I expect to see is that the RGB LED stays off when the electron wakes up. Instead it starts blinking green which means the modem is not on standby anymore. I also verified this by measuring power consumption with an oscilloscope. 

I'll try to figure out how to use 
> DEBUG_BUILD=y

I think you're right about the AT command.

technobly, 2019-10-07:
>I'll try to figure out how to use
>
>>DEBUG_BUILD=y
>
>I think you're right about the AT command.

@JDSymington For Electron, you just need to grab the binaries from the github release or use the CLI `particle update` (1.4.0 currently).  Both of these have DEBUG_BUILD=y by default.  If you OTA update system firmware via flashing a new app it will deliver binaries without DEBUG_BUILD enabled.  

If you are seeing blinking green, this is definitely a bug that needs to be fixed.  I will try them all eventually when fixing a bug, but what Electron are you using U260/U270/U201/G350 ?

JDSymington, 2019-10-07:
I was still on 1.2.1 due to some problems I had with 1.4.0. It seems that AT commands aren't reported with 1.2.1 and debug enabled.

I flashed the 1.4.0 binaries you linked to and had the same problem with the modem coming out of standby when the electron wakes up.

This is the debug log:

```
Serial monitor opened successfully:
Awake.
   142.094 AT read  +   17 "\r\n+UUSORD: 0,33\r\n"
Socket 0: handle 0 has 33 bytes pending
   142.105 AT send      17 "AT+USORF=0,1024\r\n"
   142.118 AT read  +   72 "\r\n+USORF: 0,\"34.194.48.89\",5684,33,\"\x17\xfe\xfd\x00\x01\x00\x00\x00\x00\x007\x00\x14\x00\x01\x00\x00\x00\x00\x007\xdd\x00\xc9\x97\xd3@p8\x1b\x1e\x9aV\"\r\n"       
   142.121 AT read OK    4 "OK\r\n"
   142.121 AT send       4 "AT\r\n"
   142.124 AT read OK    6 "\r\nOK\r\n"
   142.124 AT send      14 "AT+USORF=0,0\r\n"
   142.130 AT read  +   15 "\r\n+USORF: 0,0\r\n"
Socket 0: handle 0 has 0 bytes pending
   142.131 AT read OK    6 "\r\nOK\r\n"
0000142131 [system] TRACE: received 33
0000142131 [comm.coap] TRACE: recieved ACK for message id=2e
0000142132 [comm.protocol] TRACE: Reply recieved: type=2, code=0
0000142132 [comm.protocol] INFO: message id 46 complete with code 0.00
0000142132 [comm.protocol] INFO: rcv'd message type=13
0000147104 [comm.coap] TRACE: sending message id=2f
socketSendTo(0,34.194.48.89,5684,,41)
   147.104 AT send      35 "AT+USOST=0,\"34.194.48.89\",5684,41\r\n"
   147.113 AT read  >    3 "\r\n@"
   147.113 AT send      41 "\x17\xfe\xfd\x00\x01\x00\x00\x00\x00\x008\x00\x1c\x00\x01\x00\x00\x00\x00\x008\xbf\x14\xe0\xcbo\xc5\x9e\x87j\xe9\xb2\x02\x89k62\xa6\xd9[\xe1"
   147.249 AT read  +   16 "\r\n+USOST: 0,41\r\n"
   147.250 AT read OK    6 "\r\nOK\r\n"
   147.250 AT send       4 "AT\r\n"
   147.254 AT read OK    6 "\r\nOK\r\n"
   147.254 AT send      14 "AT+USORF=0,0\r\n"
   147.261 AT read  +   15 "\r\n+USORF: 0,0\r\n"
Socket 0: handle 0 has 0 bytes pending
   147.262 AT read OK    6 "\r\nOK\r\n"
Going to sleep
```

If this happens to be a bug, how would you take the modem out of standby mode if it doesn't happen automatically when the electron wakes up? 

I'm using the 3G-U270.

Please let me know if there is anything else I can test to help find the root of the problem.

technobly, 2019-12-05:
@JDSymington I'm running your app and not seeing the same thing on Electron LTE with Device OS 1.4.0 and 1.4.4.
```
...
[ Modem::join ] = = = = = = = = = = = = = = = =
     1.185 AT send      14 "AT+CGPADDR=1\r\n"
     1.195 AT read  +   29 "\r\n+CGPADDR: 1,10.***.**.***\r\n"
     1.197 AT read OK    6 "\r\nOK\r\n"
0000001197 [system] INFO: ARM_WLAN_WD 2
0000001197 [system] INFO: CLR_WLAN_WD 1, DHCP success
Serial connection closed.  Attempting to reconnect...
Serial monitor opened successfully:
Awake.
Going to sleep.
Serial connection closed.  Attempting to reconnect...
Serial monitor opened successfully:
Awake.
Going to sleep.
```

Also your app doesn't try to connect to the Particle Cloud, but your logs indicate what looks like a ping to the cloud (rcv'd message type=13) so I'm wondering if your device is running a different app.

I'm going to close this for now as an unconfirmed bug.  Please feel free to re-open with more info later if you decide I'm wrong :D

closed 2019-12-05 by technobly

JDSymington, 2019-12-06:
@technobly I'm running the app on my 3G-U270 now on 1.4.4.

I can confirm that I now see the same result as you.
Also now the RGB is breathing green instead of blinking green when the Electron wakes up.

I'm certain that I wasn't running another app as I've been seeing this same behavior for the past year on separate apps. I'm wondering if something didn't change in the firmware that unintentionally fixed the problem.

I'll implement the SLEEP_NETWORK_STANDBY flag in my main code and do some further testing.

Thank you for the help.