particle-iot/device-os #1937: Argon unreliable timeout with TCPClient

hunex91, 2019-10-10:
## Bug Report

### Expected Behavior

If the given server is offline or not accessible the `TCPClient.connect()` should fail within a well defined interval. 

### Observed Behavior

In case of a Photon this is standard 5000 ms, but with the Argon it can be between 10 sec and even a minute.

### Steps to Reproduce

Run the following code on an Argon and on a Photon. We tested on the Argon with the Device OS 1.4.1-rc.1 and with prior versions.
We also tested with the [HttpClient](https://github.com/amcewen/HttpClient) and that also gives similar results.

### Test App

```c
byte server[] = {192, 168, 1, 234};
http_request_t request;
http_response_t response;

int32_t timeStart;
int32_t timeEnd;

TCPClient client;

void setup() {
    Serial.begin(9600);
}

void loop() {
    Serial.println();
 
    timeStart = millis();
    uint32_t g = WiFi.ping(server);
    timeEnd = millis();

    Serial.printlnf("Ping=%u, Ellapsed time=%u ms", g, timeEnd - timeStart);

    timeStart = millis();
    g = client.connect(server, 80);
    timeEnd = millis();

    Serial.printlnf("TCP Connect=%u, Ellapsed time=%u ms", g, timeEnd - timeStart);
    
    request.ip = server;
    request.port = 80;

    timeStart = millis();
    http.post(request, response);
    timeEnd = millis(); 

    Serial.printlnf("HTTP Post=%u, Ellapsed time=%u ms", response.status, timeEnd - timeStart);
}
```

### Sample output

<details>
  <summary><i>Output with an Argon and Photon</i></summary>

```
--------
-- ARGON
--------

-- The server is online

Ping=4294967176, Ellapsed time=0 ms
TCP Connect=1, Ellapsed time=295 ms
HTTP Post=400, Ellapsed time=654 ms

Ping=4294967176, Ellapsed time=0 ms
TCP Connect=1, Ellapsed time=45 ms
HTTP Post=400, Ellapsed time=655 ms

-- The server is offline

Ping=4294967176, Ellapsed time=0 ms
TCP Connect=0, Ellapsed time=18098 ms
HTTP Post=4294967295, Ellapsed time=18500 ms

Ping=4294967176, Ellapsed time=0 ms
TCP Connect=0, Ellapsed time=18500 ms
HTTP Post=4294967295, Ellapsed time=18500 ms

TCP Connect=0, Ellapsed time=30249 ms
HTTP Post=4294967295, Ellapsed time=29999 ms

Ping=4294967176, Ellapsed time=0 ms
TCP Connect=0, Ellapsed time=29999 ms
HTTP Post=4294967295, Ellapsed time=29999 ms

Ping=4294967176, Ellapsed time=0 ms
TCP Connect=0, Ellapsed time=30000 ms
HTTP Post=4294967295, Ellapsed time=29999 ms

---------
-- PHOTON
---------

-- The server is online

Ping=5, Ellapsed time=26 ms
TCP Connect=1, Ellapsed time=15 ms
HTTP Post=400, Ellapsed time=414 ms

Ping=5, Ellapsed time=66 ms
TCP Connect=1, Ellapsed time=8 ms
HTTP Post=400, Ellapsed time=423 ms

Ping=5, Ellapsed time=39 ms
TCP Connect=1, Ellapsed time=9 ms
HTTP Post=400, Ellapsed time=412 ms

Ping=5, Ellapsed time=28 ms
TCP Connect=1, Ellapsed time=8 ms
HTTP Post=400, Ellapsed time=423 ms

Ping=5, Ellapsed time=47 ms
TCP Connect=1, Ellapsed time=25 ms
HTTP Post=400, Ellapsed time=745 ms

-- The server is offline

Ping=0, Ellapsed time=5000 ms
TCP Connect=0, Ellapsed time=5000 ms
HTTP Post=4294967295, Ellapsed time=5000 ms

Ping=0, Ellapsed time=5000 ms
TCP Connect=0, Ellapsed time=5000 ms
HTTP Post=4294967295, Ellapsed time=5000 ms

Ping=0, Ellapsed time=5000 ms
TCP Connect=0, Ellapsed time=5000 ms
HTTP Post=4294967295, Ellapsed time=5000 ms

Ping=0, Ellapsed time=5000 ms
TCP Connect=0, Ellapsed time=5000 ms
HTTP Post=4294967295, Ellapsed time=5000 ms
```
</details>

### References

https://docs.particle.io/reference/device-os/firmware/argon/#ping-
https://docs.particle.io/reference/device-os/firmware/argon/#connect--3
https://github.com/amcewen/HttpClient

---

technobly, 2019-11-22:
It's unclear to me why there is a variable time delay here, but I know on LTE devices such as the Boron or Electron, this timeout is 120 seconds.  That is to say, it's not guaranteed to be really short.  This timeout is also 120 seconds by default in Linux: https://pracucci.com/linux-tcp-rto-min-max-and-tcp-retries2.html

The Photon may have a hard coded 5s timeout that is not waiting not nearly long enough.

technobly, 2019-11-22:
I'll see if we can confirm what is going on here, but I'm inclined to say you should expect up to a 120 second delay worst case.