particle-iot/device-os #1942: Firmware 0.7.0 release has incorrect module number

jlkalberer, 2019-10-13:
<details>
  <summary><i>The firmware release on Github seems to be an incorrect build. It seems like the module metadata is incorrect.</i></summary>

[This release](https://github.com/particle-iot/device-os/releases/tag/v0.7.0) and possibly other RCs for 0.7.0 have a `moduleVersion` of 101 when they should be 204.
</details>

## Bug Report

### Expected Behavior

The release on Github should have a module version of 204. [It seems like there is a commit to fix this](https://github.com/particle-iot/device-os/commit/ef5ddd3eea209fa015a27b839a0b21165e0e762b#diff-8c4a1b61839ec4967050014823514dc7)

### Observed Behavior

[When I use `HalModuleParser` the version is 101](https://github.com/particle-iot/binary-version-reader).

### Steps to Reproduce

[I can reproduce this in spark-protocol](https://github.com/Brewskey/spark-protocol) by running `yarn update-firmware`. The `settings.json` file has all the parsed modules.

### Test App

```js
const parser = new HalModuleParser();
parser.parseFile(
  'bootloader-0.7.0-p1.bin',
  (result:) => {
    console.log(result);
  },
 )
```

rickkas7, 2019-10-13:
The bootloader version is not the same as the device OS system binary version.

https://github.com/particle-iot/device-os/blob/develop/system/system-versions.md

The bootloader is version 101 for 0.7.0. 
The system-part1 and system-part2 should be version 207 for the final release of 0.7.0.

jlkalberer, 2019-10-13:
@rickkas7 - I've updated my comment a bit.

The issue is that 1.4.0 and newish firmware have a `depModuleVersion` of 204. This module build doesn't exist anywhere in the releases AFAIK.

avtolstoy, 2019-10-14:
@jlkalberer Thanks for reporting the issue. It seems that we did not publish any binaries for 0.7.0-rc.5 (which has the `204` system parts): https://github.com/particle-iot/device-os/releases/tag/v0.7.0-rc.5

Just so that we are on the same page:
```
      "depModuleFunction": 4, // MODULE_FUNCTION_SYSTEM_PART
      "depModuleIndex": 2, // system-part2
      "depModuleVersion": 204, // version 204 (0.7.0-rc.5)
```

This release does exist in the database, so when performing cloud OTA everything should work out as intended, however the main purpose why `system-part1` has a strict requirement on `system-part2` of a particular version is to prevent the device from directly upgrading from < 0.7.0 releases to prevent the device getting bricked. I agree though, that we should bump it to 207 (0.7.0 final) and we'll do that before 1.4.1 final.

For your particular case:

> I've added a hack to update depModuleVersion to 207 but it doesn't seem safe.

It is safe to do so or even simply having 0.7.0 final (207) binaries in place of 0.7.0-rc.5 (204).

jlkalberer, 2019-10-14:
Thanks, for clearing that up.

Any way you can publish those binaries?
I maintain the latest version of the Particle local cloud and I'd like to revert my hack in case it introduces some bugs.

avtolstoy, 2019-11-15:
> Any way you can publish those binaries?
> I maintain the latest version of the Particle local cloud and I'd like to revert my hack in case it introduces some bugs.

I just double-checked and we've pulled that release (0.7.0-rc.5) due a critical issue with it. You should probably be using 0.7.0 final in its place.

We've also fixed the module dependency in 1.4.1: https://github.com/particle-iot/device-os/blob/develop/modules/shared/system_module_version.mk#L9

I'm going to close this issue, please feel free to reopen it if any additional problems arise.

closed 2019-11-15 by avtolstoy