particle-iot/device-os #1955: Xenon: Serial/Serial1 missing bytes

jaredwolff, 2019-11-04:
## Bug Report

When trying to run a simple command/response protocol with a separate device using Serial1, I've run into some issues when Serial as a debug mechanism. 

### Expected Behavior

When connected to the device using the VCP, (9600 BAUD) a single byte `0x03` is issued over Serial. That gets forwarded on to Serial1. A response from Serial1 is then forwarded on to Serial. A response should be that byte `0x03` plus an additional 3 (usually `0x00`). No matter how many times the command is sent.

### Observed Behavior

When sending, some bytes are missed on the return. See below:

[image]

### Steps to Reproduce

I used two Xenons in Manual mode. DeviceOS 1.4.2. I also went back and tried on an earlier version (1.3.1) but no change.

### Test App

## The Device Connected via VCP
```c
// setup() runs once, when the device is first turned on.
void setup() {

  Serial.begin();
  Serial1.begin(9600);

}

// loop() runs over and over again, as quickly as it can execute.
void loop() {

  // Process from PC
  while(Serial.available()) {
    char c = Serial.read();
    Serial1.write(c);
  }

  // Echo on the other end.
  while(Serial1.available()) {
    char c = Serial1.read();
    Serial.write(c);
    // Serial1.write(c);
    // Serial1.write((uint8_t)0);
    // Serial1.write((uint8_t)0);
    // Serial1.write((uint8_t)0);
  }

}
}

void loop() {

}
```

## The "Remote" Device

```
// setup() runs once, when the device is first turned on.
void setup() {

  Serial.begin();
  Serial1.begin(9600);

}

// loop() runs over and over again, as quickly as it can execute.
void loop() {

  // Process from PC
  // while(Serial.available()) {
  //   char c = Serial.read();
  //   Serial1.write(c);
  // }

  // Echo on the other end.
  while(Serial1.available()) {
    char c = Serial1.read();
    // Serial.write(c);
    Serial1.write(c);
    Serial1.write((uint8_t)0);
    Serial1.write((uint8_t)0);
    Serial1.write((uint8_t)0);
  }

}
```

### References

[Community forum post.](https://community.particle.io/t/xenon-serial-missing-data/53040/7)

jaredwolff, 2019-11-04:
It also seems like the buffer isn't filling up. `Serial1.available()` returns 1 even if I haven't read bytes.  Even if there are several bytes to be received.

jaredwolff, 2019-11-04:
Even removed `Serial` from the whole equation. Using the debugger `Serial1.available()` returns > 1 but the buffer doesn't accumulate. It seems to reset. 

```// setup() runs once, when the device is first turned on.
void setup() {

  // Serial.begin();
  Serial1.begin(9600);
  timer.start();

  pinMode(D7,OUTPUT);

}

// loop() runs over and over again, as quickly as it can execute.
void loop() {

  if( sendByte ) {
    sendByte = false;
    Serial1.write(3);
  }

  // Echo on the other end.
  if(bytes = Serial1.available()) {
    if(digitalRead(D7)) {
      digitalWrite(D7,LOW);
    } else {
      digitalWrite(D7,HIGH);
    }
  }

}
```

jaredwolff, 2019-11-04:
Here's the test setup:

[image]

Cheong2K, 2019-11-04:
Hi @jaredwolff, we are looking into this issue, we will get back to you as soon as possible.

Cheong2K, 2019-11-06:
@jaredwolff, have you got this resolved? I could not replicate this issue with my units so far.

jaredwolff, 2019-11-06:
Nope. I only made this ticket after many attempts of fixing it. Can you describe your test setup so I can try to replicate it working on your end?

Cheong2K, 2019-11-06:
My setup is exactly the same as yours with the code you provided. I tried with manual mode as you said, also I tried with system thread enabled, I sent from my PC every 1ms, it works for me.

Would you mind to check again with the GND wire or change the wires with another set to see?

jaredwolff, 2019-11-06:
Did you use the first set of code? The only way I got it to happen was by sending 0x03 manually as fast as I could click the mouse using Cool Term. 

As mentioned the signal looked ok even when some bytes were missing. So is it possible ‘Serial’ is dropping those bytes?

The second set of code was removing that element. I.e. it’s not the failure case. What I did see was Serial.available() not incrementing as more bytes are received and not read. Is this expected?

ScruffR, 2019-11-08:
I also found some odd behaviour where the first byte transferred always seems a bit off
[image]

This is the trace created by this code
```
uint8_t data[4] = { 0x00, 0x00, 0x00, 0x00 };
void setup() {
  Serial1.begin(250000, SERIAL_8N1);
}
void loop() {
  Serial1.write(data, sizeof(data));
  delay(1000);
}
```
Note the extended stop bit of the initial byte.

avtolstoy, 2020-02-07:
This issue also potentially affects USB control requests. The easy way of reproducing is initializing `SerialLogHandler` in the application, NOT opening the serial port (or opening and closing) on the host, letting the device log for some time some substantial amount of data and then running some USB control requests e.g. `particle usb dfu`. The request will timeout. Running `particle usb dfu` again and while waiting for the requests to be handled opening the serial port, will allow the request to get through.

ScruffR, 2020-02-07:
I don't agree with the change of the topic title that limits the issue to USB Serial since the timing on Serial1 still seems off too.

avtolstoy, 2020-02-07:
@ScruffR I don't agree that there is anything wrong with the image from the scope :smile: UART is asynchronous and the stop bit configuration specifies the minimum length of the stop bit. Would you expect something like this to behave any differently or break UART communications (the delays are arbitrary)?
```c++
Serial1.write(0x00);
delay(10);
Serial1.write(0x00);
delay(1000);
Serial1.write(0x00);
delay(1);
Serial1.write(0x00);
```

As far as why this exaclty happens for your example, it's just the byte-by-byte nature of `write(const uint8_t*, size_t)` (https://github.com/particle-iot/device-os/blob/develop/wiring/src/spark_wiring_print.cpp#L37) and UART implementation on Gen 3 devices using DMA. The first byte is immediately scheduled for DMA transfer, the other 3 bytes are put into the buffer, the DMA transfer finishes and another DMA transaction is started for those remaining two bytes. It's just the additional latency of restarting the DMA that is seen.

If you use `HAL_USART_Write()` in monolithic on Gen 3 (because it's not exported), you'll see a single DMA transfer, however even with that implementation it is not guaranteed on the edges of the circular buffer for obvious reasons.

avtolstoy, 2020-08-28:
This should be resolved as of 2.0.0-rc.1

closed 2020-08-28 by avtolstoy