particle-iot/device-os #1962: mocked GPIO pins not initialized before STARTUP() block in GCC build

elcojacobs, 2019-11-11:
## Bug Report
When digital pins are set in a STARTUP block, which I guess is the main use case, the mocked pins for the gcc build are not initialized yet. This causes a segfault.

### Expected Behavior
IO mocks should be ready to use before the STARTUP block executes, so the simulator/gcc build does not need to be handled differently from the hardware target.
calling pinMode should not cause a segfault.

The pins are initialized in the hal pin map constructor, which is a global object. I'm not sure how the STARTUP block is implemented, but apparently it runs before globals are constructed. 

Perhaps wrapping the GpioPinMap class with a function that initializes it on first use is a solution, something like this:

```c
// instead of:
GpioPinMap PIN_MAP;

// use:

GpioPinMap& _PIN_MAP() {
  static GpioPinMap map;
  return map;
}

#define PIN_MAP _PIN_MAP()
```

### Observed Behavior
Segfault

What did it actually do?

### Steps to Reproduce

I'm pretty sure the app below should reproduce the behavior when compiled with PLATFORM=gcc

### Test App

```c
STARTUP(
pinMode(0, OUTPUT);
);

void setup() {
  // A minimal test case to reproduce the issue is super helpful!
}

void loop() {

}
```

sergeuz, 2019-11-11:
@elcojacobs Thanks for reporting this. Given that you tracked down the issue already, may I ask you to submit a PR with the fix?

elcojacobs, 2019-11-11:
Done.

sergeuz, 2019-11-12:
Fixed in https://github.com/particle-iot/device-os/pull/1963.

closed 2019-11-12 by sergeuz