particle-iot/device-os #1964: build modules: functional: No such file or directory

Noah-Huppert, 2019-11-12:
## Bug Report

### Expected Behavior

Successfully run the `all` Make target in the `modules` directory.

### Observed Behavior

Target failed with an error related to not being able to find the `functional` header included in a file:

```
% cd modules
% make all PLATFORM=photon
  /home/noah/libs/particle-os/modules/photon/system-part1/makefile /home/noah/libs/particle-os/modules/photon/system-part2/makefile /home/noah/libs/particle-os/modules/photon/user-part/makefile
In file included from src/spark_protocol_functions.cpp:20:
./inc/protocol_defs.h:3:10: fatal error: functional: No such file or directory
    3 | #include <functional>
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [../build/module.mk:277: ../build/target/communication/platform-6-m-prod-6/src/spark_protocol_functions.o] Error 1
make[1]: *** [../../../build/recurse.mk:12: communication] Error 2
make: *** [makefile:86: /home/noah/libs/particle-os/modules/photon/system-part1/makefile] Error 2
```

### Steps to Reproduce

Steps:

```
% git clone https://github.com/particle-iot/device-os.git
% cd device-os
% git checkout release/stable
% git submodule update --init --recursive
% cd modules
% make all PLATFORM=photon -s
```

System Info:

```
% lsb_release -a
lsb_release -a
LSB Version:    1.0
Distributor ID: VoidLinux
Description:    Void Linux
Release:        rolling
Codename:       void
% uname -a
Linux apollo 5.3.8_2 #1 SMP PREEMPT Fri Nov 1 01:39:56 UTC 2019 x86_64 GNU/Linux
```

### Test App

N/A

### References

N/A

---

avtolstoy, 2019-11-12:
@Noah-Huppert Thanks for submitting the issue. Could you please also provide `arm-none-eabi-gcc --version` output?

Noah-Huppert, 2019-11-12:
Sure thing, here are the versions of all the dependencies mentioned in [the docs I am following](https://docs.particle.io/support/particle-tools-faq/local-build/).

```
% arm-none-eabi-gcc --version
arm-none-eabi-gcc (GCC) 9.1.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
```
% git --version
git version 2.23.0
```
```
% dfu-util --version
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

avtolstoy, 2019-11-12:
@Noah-Huppert Unfortunately, we currently don't support any version of the compiler except for gcc 5.3.1.

closed 2019-11-15 by avtolstoy