particle-iot/device-os #1967: How to debug hardfault in system part 2 (probably in WICED source)

elcojacobs, 2019-11-13:
I am trying to find the cause of a hardfault that I can trigger consistently, but I think it is in the system layer or WICED.
When I trigger setup mode in my current build (modular with SWD), the hard fault occurs predictably, but not immediately.

I figured out how to debug including loading the dynamic system module symbols in openocd, so I was able to set a breakpoint in the hardfault handler to get at least some info about the location.

[image]

From what I think I can deduce from those registers:

Program counter seems to be halfway a memcpy. memcpy is at 0x80981ce according to the system part 2 map file.
Link register is at `0x80840b9 is in xQueueGenericReceive (WICED/RTOS/FreeRTOS/ver8.2.1/Source/queue.c:1445).`

The WICED source is not included or mapped to a different location, so I can't put a breakpoint there.

This is much deeper in the system layers that I wanted to go, but I'm in the rabbit hole now. Please help.

elcojacobs, 2019-11-13:
The file reported by the debugger with the command `list *134758585` seems to have version 8.2.1 in the path, which is older than the queue.c in `third_party/freertos`, which is 10.2.1.

I can imagine mixing different RTOS versions could cause issues.

avtolstoy, 2019-11-14:
@elcojacobs Thanks for reporting an issue. Is there an small app I can use to try to reproduce this? I'm not sure we'll be able to give much help without a lot of poking around the threads/stacks for these types of issues.

> The file reported by the debugger with the command list *134758585 seems to have version 8.2.1 in the path, which is older than the queue.c in third_party/freertos, which is 10.2.1.

Photon/P1 uses FreeRTOS bundled with WICED (8.2.1), we haven't upgraded it, so that's expected, we are not mixing versions.

elcojacobs, 2019-11-14:
Here you go:
https://github.com/elcojacobs/brewblox-firmware/tree/fix/setup-hard-fault/app/wifi-test

I was able to narrow it down to a call to WiFi.localIP() I think.

https://github.com/elcojacobs/brewblox-firmware/blob/fix/setup-hard-fault/app/wifi-test/application.cpp#L208

The app includes a python script to report on WiFi/TCP status over serial. Same as the last time I reported networking issues. You can use pipenv to easily run it.

avtolstoy, 2019-11-14:
@elcojacobs Thanks for additional info.

I believe this is the same issue as #1858. As a workaround for now I might suggest trying to make sure that `WiFi.ready()` is true before any `WiFi.xxx()` calls that rely on `wifi_config()`, so `localIP`, `macAddress`, `subnetMask`, `gatewayIP`, `dnsServerIP`, `dhcpServerIP`, `BSSID`, `SSID`.

elcojacobs, 2019-11-14:
And wrap them in a single threaded block? In theory, the WiFi could drop between the WiFi.ready() call and the WiFi.xxx() call.

elcojacobs, 2019-11-14:
Nope, that seems to deadlock in this function:
```
static inline ALWAYS_INLINE wiced_result_t wiced_wifi_get_mac_address( wiced_mac_t* mac )
{
    return (wiced_result_t) wwd_wifi_get_mac_address( mac, WWD_STA_INTERFACE );
}
```

[image]

elcojacobs, 2019-11-14:
Note the call stack, the part I cannot access also has xQueueueGenericReceive.

Deadlocking here:
``` 
SINGLE_THREADED_BLOCK() // avoid hardfault
            {        if (!wifiDetailsUpdated) {
            SINGLE_THREADED_BLOCK() // avoid hardfault
            {
                if (spark::WiFi.ready()) { // check again to be sure
                    IPAddress ip = spark::WiFi.localIP().raw().ipv4;
                    snprintf(ipString, 16, "%d.%d.%d.%d", ip[0], ip[1], ip[2], ip[3]);
                    strncpy(ssidString, spark::WiFi.SSID(), 16);
                    wifiDetailsUpdated = true;
                }
            }
        }

```

avtolstoy, 2019-11-15:
@elcojacobs You shouldn't be using `SINGLE_THREADED_BLOCK()`. WICED API calls (as well as our APIs) use various synchronization primitives, which require the scheduler to be running. I agree that there is still possibility that the `ready()` becomes false in-between the calls, however at this point that's the best we can do. And AFAIK the key point here is to make sure that WICED connectivity is initialized (i.e. `WiFi.on()` has been called and internal WICED network stack state is initialized).

closed 2020-03-03 by avtolstoy

elcojacobs, 2020-04-06:
When trying to chase a hardfault again, I came across this issue I created myself, saying "I figured out how to debug including loading the dynamic system module symbols in openocd". I never wrote down how here and lost how I did it. :rofl: 

Well, I figured it out. So for anyone else coming across my unhelpful quote above in a google search, here is how to do it.
This works with a dynamically linked binary. No monolithic build necessary.

```
make clean all program-dfu PLATFORM=p1 PARTICLE_DEVELOP=y USE_SWD=y
```
This will flash an SWD enabled system layer.

Next build and flash your application with SWD:
```
make clean all program-dfu PLATFORM=p1 PARTICLE_DEVELOP=y USE_SWD=y
```

Create this gdb init file (adjust paths relative to your build dir) and save as system.gdbinit
```
define add-symbol-file-auto
  # Parse .text address to temp file
  shell echo set \$text_address=$(readelf -WS $arg0 | grep .text | awk '{ print "0x"$5 }') >/tmp/temp_gdb_text_address.txt

  # Source .text address
  source /tmp/temp_gdb_text_address.txt

  #  Clean tempfile
  shell rm -f /tmp/temp_gdb_text_address.txt

  # Load symbol table
  add-symbol-file $arg0 $text_address
end

add-symbol-file-auto platform/spark/device-os/build/target/system-part1/platform-8-m/system-part1.elf
add-symbol-file-auto platform/spark/device-os/build/target/system-part2/platform-8-m/system-part2.elf
```

In vscode, use extension cortex-debug and use this launch config in 'launch.json', adjust paths for your system.
```
{
      "name": "brewblox p1 openocd debug",
      "type": "cortex-debug",
      "request": "attach",
      "servertype": "openocd",
      "cwd": "${workspaceRoot}",
      "executable": "./build/target/brewblox-p1/brewblox.elf",
      "debuggerArgs": [
        "-command=./tools/system.gdbinit",
      ],
      "rtos": "FreeRTOS",
      "device": "STM32F205RG",
      "configFiles": [
        "/usr/local/share/openocd/scripts/interface/stlink-v2.cfg",
        "/usr/local/share/openocd/scripts/target/stm32f2x.cfg"
      ],
      "armToolchainPath": "/home/elco/source/gcc-arm-none-eabi-5_3-2016q1/bin",
      //      "showDevDebugOutput": true,
      "svdFile": "./tools/STM32F215.svd"
    },
```

svd file is optional. This is for an stlink-v2 with SWD. Ensure that you are not using the SWD pins in your program (D6 and D7)