particle-iot/device-os #1989: Firmware flash: Unused buffer may indicate usage of wrong variable

perotom, 2019-12-12:
After I got some problems writing my OTA update over HTTP to the flash I discovered that after a write to the flash of "Not an aligned data" (https://github.com/particle-iot/device-os/blob/a7aabdfce6a0f036e6493544ff9246f81d2a51c3/platform/MCU/STM32F2xx/SPARK_Firmware_Driver/src/flash_mal.c#L901) subsequent writes will fail. I looked at the corresponding code and found that the array `buf` (https://github.com/particle-iot/device-os/blob/a7aabdfce6a0f036e6493544ff9246f81d2a51c3/platform/MCU/STM32F2xx/SPARK_Firmware_Driver/src/flash_mal.c#L903) is filled but never used. Could it be possible that there is something wrong with this code block?

avtolstoy, 2019-12-12:
@perotom Thanks for reporting the issue. This does seem like a bug.

@artyLee Could you please take a look at this? Thanks.

artyLee, 2019-12-12:
You're welcome! I try to look at this issue.

PR #2036 (2020-03-05): [stm32f2xx] platform: fix FLASH_Update() for unaligned writes
### Problem

#1989

### Solution

Flash the rest of the buffer using `Flash_ProgramByte`.

### Steps to Test

Gen 2 OTA/ymodem updates are not affected.

### Example App

N/A

### References

- Closes #1989 
- [CH46859]

---

### Completeness

- [x] User is totes amazing for contributing!
- [x] Contributor has signed CLA ([Info here](https://github.com/spark/firmware/blob/develop/CONTRIBUTING.md))
- [x] Problem and Solution clearly stated
- [x] Run unit/integration/application tests on device
- [ ] Added documentation
- [ ] Added to [... cut]
closed 2020-03-05 by avtolstoy