particle-iot/device-os #2002: Fix typo in main/makefile

kubark42, 2020-01-26:
I imagine that `.PNONY: st-flash program-dfu clean all` should be `.PHONY: st-flash program-dfu clean all`

https://github.com/particle-iot/device-os/blob/01e3a87255f6c164c4ef11c728ea041fae66b030/main/makefile#L78

Although maybe that's just old cruft and should be cleaned out? That typo has existed for 4 years, so it looks pretty stale.

(Or maybe it's not a typo? And if not, what is it? All I find when searching online is either My Little Pony or references to `.PHONY`)

sergeuz, 2020-01-27:
Thanks for reporting this. Would you want to submit a PR with the fix?

kubark42, 2020-01-27:
I appreciate the invite, but I don't know what the design intent was here nor do I have the skills to faithfully execute it[*]. The code was originally written 5 years ago and then updated 4 years ago in order to resolve a crash (C.f. https://github.com/particle-iot/device-os/issues/986 and https://github.com/particle-iot/device-os/commit/cde57d109105e40821041b9e8ece496a0d18ae7f), so it's got some history to it. 

Questions to answer:

- Should the typo be fixed, or
- should the section be removed as cruft?
  - If there is cruft, how extensive is it?

[*] GNU `make` and Makefiles in general not being my strength, I wouldn't know how to go about testing that this doesn't either regress or introduce a new bug.

sergeuz, 2020-03-03:
Fixed by https://github.com/particle-iot/device-os/commit/e5e3322255f1f450cfa65baba5799f4fb53919f8.

closed 2020-03-03 by sergeuz