particle-iot/device-os #2021: Consider add -nostdlib flag

talybin, 2020-02-11:
## Enhancement Request

### Expected Behavior

Link project with <b>-nostdlib</b> flag.

### Observed Behavior

Flash memory overflow with automatic include of standard library.

### Steps to Reproduce

Include iostream and try to build firmware.

### Test App

```cpp
#include <iostream>

void setup() { }
void loop() { }
```

---

## Feature/Enhancement Request

Using particle tool (particle compile electron) there is this last step of linking:

```
Building target: target/workspace.elf
Invoking: ARM GCC C++ Linker
mkdir -p target/
arm-none-eabi-g++ -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=10 -DPLATFORM_NAME=electron -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD00A -DUSBD_PID_CDC=0xC00A -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb -DINCLUDE_PLATFORM=1 -fno-builtin -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -DSYSTEM_VERSION_STRING=1.4.4 -DRELEASE_BUILD -Werror -I../../../user/inc -I../../../dynalib/inc -I../../../services/inc -I../../../third_party/nanopb/nanopb -I../../../hal/inc -I../../../hal/shared -I../../../hal/src/electron -I../../../hal/src/stm32f2xx -I../../../hal/src/stm32 -I../../../third_party/freertos/freertos/FreeRTOS/Source/include -I../../../third_party/freertos/freertos/FreeRTOS/Source/portable/GCC/ARM_CM3 -I../../../system/inc -I../../../third_party/miniz/miniz -I../../../rt-dynalib/inc -I../../../wiring/inc -I../../../modules/shared/stm32f2xx/inc -I../../../modules/electron/system-part2/inc -I../../../modules/shared/stm32f2xx/inc -I../../../platform/shared/inc -I../../../platform/MCU/STM32F2xx/STM32_USB_Host_Driver/inc -I../../../platform/MCU/STM32F2xx/STM32_USB_OTG_Driver/inc -I../../../platform/MCU/STM32F2xx/STM32_StdPeriph_Driver/inc -I../../../platform/MCU/STM32F2xx/STM32_USB_Device_Driver/inc -I../../../platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc -I../../../platform/MCU/shared/STM32/inc -I../../../platform/MCU/STM32F2xx/CMSIS/Include -I../../../platform/MCU/STM32F2xx/CMSIS/Device/ST/Include -I. -MD -MP -MF target/workspace.elf.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -Wundef -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -fno-builtin-malloc -fno-builtin-free -fno-builtin-realloc -DUSER_FIRMWARE_IMAGE_SIZE=0x20000 -DUSER_FIRMWARE_IMAGE_LOCATION=0x8080000 -DMODULAR_FIRMWARE=1 -DMODULE_VERSION=6 -DMODULE_FUNCTION=5 -DMODULE_INDEX=1 -DMODULE_DEPENDENCY=4,2,1406 -DMODULE_DEPENDENCY2=0,0,0 -D_WINSOCK_H -D_GNU_SOURCE target//obj/src/user_module.o target//obj/src/module_info.o target//obj/src/user_export.o target//obj/src/newlib_stubs.o  --output target/workspace.elf -nostartfiles -Xlinker --gc-sections -L../../../build/arm/linker/stm32f2xx -L../../../build/arm/linker/stm32f2xx/electron -L../../../build/arm/linker/stm32f2xx -L../../../build/arm/linker/stm32f2xx/electron  -L../../../build/target/user/platform-10-m -L../../../build/target/services-dynalib/platform-10-m/ -L../../../build/target/hal-dynalib/platform-10-m/ -L../../../build/target/system-dynalib/platform-10-m/ -L../../../build/target/rt-dynalib/platform-10-m/ -L../../../build/target/wiring/platform-10-m/ -L../../../build/target/communication-dynalib/platform-10-m/ -L../../../build/target/platform/platform-10-m/ -L../../../build/target/wiring_globals/platform-10-m/ -L../../../build/arm/linker -Wl,--whole-archive -luser -lhal-dynalib -lservices-dynalib -lsystem-dynalib -lrt-dynalib -lwiring -lcommunication-dynalib -lplatform -lwiring_globals -Wl,--no-whole-archive  -lnosys --specs=nano.specs -L../../../modules/electron/system-part1 -L../../../modules/electron/system-part3 -L../../../modules/electron/system-part2 -L. -T./linker.ld -Wl,--defsym,USER_FIRMWARE_IMAGE_SIZE=0x20000 -Wl,--defsym,USER_FIRMWARE_IMAGE_LOCATION=0x8080000 -Wl,-Map,target/workspace.map
```

My suggestion is to add one more option, namely <b>-nostdlib</b>. This option makes sure a library called libstdc++ is never included. As for today, it may be included automatically by linker if you, in your code, for example, try to add iostreams.

When this happens the firmware fail to compile with following error:

```
/usr/local/gcc-arm-embedded/bin/../lib/gcc/arm-none-eabi/5.3.1/../../../../arm-none-eabi/bin/ld: target/workspace.elf section `.text' will not fit in region `APP_FLASH'
/usr/local/gcc-arm-embedded/bin/../lib/gcc/arm-none-eabi/5.3.1/../../../../arm-none-eabi/bin/ld: region `APP_FLASH' overflowed by 23268 bytes
```

Since <i>libstdc++</i> is too large for Arduino flash memory, and impossible to link, it would help to disable auto link of it. It will open opportunity for users to create smaller library (that fits in memory) with only definitions needed to compile included part of std library. The code to fill the "gaps" (definitions) can be just copy-pasted from standard library.

Is it possible to add this option (at least for Photon and Electron)?

avtolstoy, 2020-02-11:
We are not Arduino and rely on C++ runtime (which is provided by `libstdc++`) both within the system parts of DeviceOS as well as in application and cannot remove it. Anything unused is normally discarded and not present in the resulting binary file after the linking stage (unless specifically asked for like in your example by including `<iostream>`).

I'm going to close this, as it is not achievable with the current DeviceOS architecture and codebase.

P.S. The main problem with C++ input/output streams is locale support, which is not that easy to disable with our toolchain, otherwise it would potentially be possible to fit in.

closed 2020-02-11 by avtolstoy

talybin, 2020-02-12:
Andrey, you don't see the picture here.

What you call a C++ runtime will still be accessible with _nostdlib_ flag, at least for applications (I'm not asking to add this flag for DeviceOS).
You can see STL as built of two parts. Part (1) that needs to be compiled (by application) and a precompiled part (2) that may or may not be included depending on use of first part.

Part 1 contains the most of library features such as _vector_, _map_, etc.
Part 2 contains some method/member definitions of part 1.

What _nostdlib_ flag do is to tell compiler to never include part 2 at link stage even if it detects that the code needed to build application is partly located in part 2. When this happens it does not matter what you trying to use (iostream, locale, or something else) the linkage will fail because part 2 is too big to fit in flash.

Now, if we add _nostdlib_ flag it will not help much, right? The link stage will still fail not finding definitions this time.
But there is a difference. If we, in our application, provide definitions needed the code will compile. This means that we can provide, for example, definitions needed to compile iostream (these can be copy-pasted from part 2 of STL).

I'm aware of locale size in STL and this can simply be solved by create interface that support single facet or just disable it. This is not that important for most of the projects. I have done something similar [here](https://github.com/talybin/ard-streams).

What it is good for?

1. This allows us to port many great libraries that depend on std streams such as _Boost_ and _Nlohmann Json_ without modification.
2. Customize or provide own locale.
3. Make all features of STL be accessible.

avtolstoy, 2020-02-12:
First of all, I understand where you are coming from, and if it's just allowing this flexibility for making it potentially possible to do this, I'll welcome a PR that allows it either with existing [`EXTRA_CFLAGS`](https://docs.particle.io/support/particle-tools-faq/local-build/#defining-custom-symbols) or potentially introduction of `EXTRA_LDFLAGS`. The problem why this currently doesn't work, is due to recursive make.

`EXTRA_CFLAGS`/`EXTRA_LDFLAGS` will additionally need to be applied to https://github.com/particle-iot/device-os/blob/develop/modules/shared/nRF52840/user_build.mk (for Gen 3) and https://github.com/particle-iot/device-os/blob/develop/modules/shared/stm32f2xx/user_build.mk (Gen 2).

--------------------------------------------------------

As for the arguments provided, even something as simple as `std::shared_ptr` requires libstdc++.

> When this happens it does not matter what you trying to use

It actually does, otherwise we'd always be overflowing.

> But there is a difference. If we, in our application, provide definitions needed the code will compile. This means that we can provide, for example, definitions needed to compile iostream (these can be copy-pasted from part 2 of STL).

And that's exactly what we'd like to avoid, as it's too fragile of an approach. We've for example had the history of trying to support the stdlib internals for `std::thread` and other concurrency-related primitives and when switching the compiler versions this required us to make necessary changes for the new versions and having to support something we should not really need to (while also potentially having to keep support for old versions of gcc/stdlib). This is even worse for the applications that might rely on this approach.

My $0.02:
1. There are plenty of other libraries that could be used instead that don't require io streams :smile: 
2. Agreed, potentially.
3. Which might not be the best choice for the constrained platform that we are. I haven't been following the recent changes, but what about exceptions (we have them disabled)?

talybin, 2020-02-12:
> As for the arguments provided, even something as simple as `std::shared_ptr` requires libstdc++.

Can be done by providing definitions :)

> > When this happens it does not matter what you trying to use
> 
> It actually does, otherwise we'd always be overflowing.

By <i>what you use</i> I meant depending stuff the has definitions in libstdc++. Otherwise correct, header only not require libstdc++.

> And that's exactly what we'd like to avoid, as it's too fragile of an approach.

From user point of view the flag does no difference at all. It does not change anything. You can't link libstdc++ today. You will not be able to link with this flag. There is nothing to support here.

> 
> My $0.02:
> 
>     1. There are plenty of other libraries that could be used instead that don't require io streams smile

As a C++ coder I'm perfectionist and using third-party libraries carefully :)

>     3. Which might not be the best choice for the constrained platform that we are. I haven't been following the recent changes, but what about exceptions (we have them disabled)?

For constrained platform it is exactly why this is the best choice.
Usually exceptions can be disabled by macro def. I know both STL and Boost supports this. Providing definitions may also help to handle error in different ways.
Actually I'm glad you asked. Why exceptions disabled anyway? To minimize the output size? There is too low overhead and wining in size compare to how hard sometimes implement code without them. For example overloading operator[].

talybin, 2020-02-12:
Again, using <i>nostdlib</i> does not mean you can't use standard library. The usage remain the same as with or without it. The only difference that g++ will not try to add the binary itself even if want to.

talybin, 2020-02-12:
About recursion flags, nothing is impossible :)

talybin, 2020-02-13:
@avtolstoy, so how do we proceed with this issue?

avtolstoy, 2020-02-13:
I think I've outlined the potential solution in https://github.com/particle-iot/device-os/issues/2021#issuecomment-585122232. We are not going to enable `-nostdlib` as the default option for the applications for now.

talybin, 2020-02-13:
Ok, using flags is not a good solution as public libraries that contains definitions will not work by default. Anyway could you at least answer my question about exceptions?

> Why exceptions disabled anyway?

avtolstoy, 2020-02-13:
Historical and flash/RAM constraints mainly.

sergeuz, 2020-02-14:
> From user point of view the flag does no difference at all. It does not change anything. You can't link libstdc++ today. You will not be able to link with this flag. There is nothing to support here.

That's not quite right. libstdc++ is already being linked in user applications, and as @avtolstoy have mentioned already, even smart pointers from `<memory>` require what you call the "part 2" of the library. It's just that the linker only pulls in necessary parts of it, so everything builds fine. In other words, when you include `<iostream>`, it's not the standard library as a whole that doesn't fit into flash, it's just the streams library and its dependencies.

talybin, 2020-02-14:
Did a few tests and yes, my bad... I know all about zero-overhead principle but I didn't know that g++ linker is that smart too and able to drop dead **binary** code. I guess it wasn't easy task to implement, but, hey, this is the linker.