particle-iot/device-os #2055: ASSERTION_ERROR with modular build on p1

elcojacobs, 2020-03-19:
##  Bug Report
When compiling the default app for a p1 with MODULAR=y, I get an assertion error SOS.

### Observed Behavior

Panic code of 10 red flashes.

### Steps to Reproduce

cd device-os/main
make clean all program-dfu PLATFORM=p1 MODULAR=n

### Test App

Not needed, juist build the default app
I'm building with 8-2019-q3-update 8.3.1 20190703 and Boost 1.72

Please let me know if this reproduces. I can't get openocd hardware debugging to work anymore, possibly this is related.
I can start the debugger and connect to the target, but it seems to get an exception before I can break anywhere meaningful.

avtolstoy, 2020-03-20:
> I'm building with 8-2019-q3-update 8.3.1 20190703 

We currently do not support any gcc version other than 5.3.1, even if DeviceOS may currently be buildable with it. Is the issue reproducible with 5.3.1?

elcojacobs, 2020-03-21:
No hardfault with 5.3.

Warning, what follows is a bit of a rant. I am getting a bit frustrated. To the point where I think my next hardware generation won't have a chipset from Particle.

So, my take on this:
This is a bug. If it doesn't express itself with 5.3, you are just lucky. Perhaps it is an uninitialized variable that just happens to be zero most of the time. Maybe next release it isn't.

The recent bug with Ymodem not working because of uninitialized member variables, it is just silly and unnecessary. 
What we have seen in our app for months is that a Ymodem firmware sometimes failed with a hardfault. The device would reboot and it would work on the second or third try. Only with 1.5.0-rc.2 it would fail every time. Uninitialized memory is more likely to be zero just after boot, which explains why a reboot often helped. Until 1.5.0-rc.2, in which memory use elsewhere int he code probably increased the change of it not being zero by luck. But this bug was totally unnecessary. Use `-Wmissing-field-initializers` and the compiler will just tell you to fix it.

Using an outdated compiler is annoying because new features cannot be used, it includes on old version of the STL and because the warnings are a lot better in more recent versions.

Try building a minimal app with some extra compiler warnings on GCC 8. May I suggest some:

```
CFLAGS += -ffunction-sections -Wall
CFLAGS += -Wno-deprecated
CFLAGS += -Wnull-dereference 
CPPFLAGS += -fdiagnostics-show-template-tree -fno-elide-type # GCC8 required
CFLAGS += -Wextra
CFLAGS += -Wvla -Wswitch-enum -fvar-tracking-assignments 
CFLAGS += -Wduplicated-cond -Wduplicated-branches

# Flag compiler error for [-Wdeprecated-declarations]
CFLAGS += -Werror=deprecated-declarations
``` 

I'm not even using these warnings below when building against device-os, because there will be just too many warnings.
```
# CPPFLAGS += -Wextra-semi -Wpedantic 
CFLAGS += -Wno-unused-parameter
```
For unused-parameter:
Instead of writing an overload function as:
``` cpp
void foo(int& v){
   // do something without using v
}
```
Write it in a way that makes it clear that the variable is not used intentionally, and not by error:
``` cpp
void foo(int&){
   // do something without using v
}
```

Here is some build output of a minimal app with those flags:
```
In file included from ../wiring/inc/spark_wiring_wifi.h:39,
                 from ../wiring/inc/spark_wiring.h:44,
                 from ./inc/application.h:42,
                 from /home/elco/repos/firmware/app/minimal/application.cpp:2:
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::ssid' [-Wmissing-field-initializers]
     WLanCredentials creds_ = {0};
                                ^
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::ssid_len' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::password' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::password_len' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::security' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::cipher' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::channel' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::flags' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::version' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::eap_type' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::inner_identity' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::inner_identity_len' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::outer_identity' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::outer_identity_len' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::private_key' [-Wmissing-field-initializers]
../wiring/inc/spark_wiring_wifi_credentials.h:160:32: warning: missing initializer for member 'WLanCredentials::private_key_len' [-Wmissing-field-initializers]
[... 61 lines cut ...]
                                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./inc/application.h:69,
                 from /home/elco/repos/firmware/app/minimal/application.cpp:2:
../wiring/inc/spark_wiring_eeprom.h: At global scope:
[... cut ...]
Can't comment on the not handled switch cases, if they are all handled by the default case, perhaps make it explicit.

I would really like to see Particle fix the bugs with the newer compiler instead of sticking to such an old version.

So even if you only support 5.3, this is still a bug worth looking at.
Perhaps it is the cause of random hardfaults and reboots I see happening in my app. Perhaps it 'mostly works' in 5.3. Don't ignore it because you have to use gcc8 to see it express itself more clearly.

avtolstoy, 2020-03-21:
@elcojacobs I would actually like to thank you for raising all these concerns and assure you that in the following months we are going to address most of these issues that come with some of our historical baggage. I'm personally really looking forward to us switching to newer compiler versions and C++ standards, which have much better analysis that will improve our codebase, and I can assure you that this is the plan for the nearest future.

My comment was simply about the current state of affairs. We have tested DeviceOS with GCC 5.3.1 and we can assure that it works as intended when using a particular version of the compiler in most of the cases.

The YModem issue should also be resolved as of 1.5.0 final with #2051.

> So even if you only support 5.3, this is still a bug worth looking at.

Agreed, I'm not going to close this issue, because this is still a valid issue that we need to take a look into and we are taking this seriously.

elcojacobs, 2020-03-21:
Thanks! Exactly the response I was hoping for.

A major issue that I think should go along with this change is to introduce some owning pointers for things like sockets and TCP clients. 
I think some non-owning pointers might be accessed somewhere after they go out of scope causing very hard to debug hard faults if the network has a hiccup.
I quite frequently see hard fault reports at my customers and I am pretty certain that they don't happen in my own code. I run a lot of test code with fsanitize and I have not seen the bugs in the gcc cross compiled firmware either.

It is frustrating to constantly have to chase bugs that I can't really fix myself.
I know it's not easy for you, because you have to work with the Wiced stack and c interfaces of libraries.
You have already improved the code quality and process a lot though, keep up the good work.

elcojacobs, 2020-03-22:
Maybe this is related:
https://github.com/particle-iot/device-os/pull/1953#issuecomment-602217970

If the startup code is overwriting used data here, doing a non-modular build might just be a way to shuffle the memory layout a bit.

I seem to be getting quite random behavior, depending on build flags. When I compile my app with these flags:
```
make clean all program-dfu APP=brewblox PLATFORM=p1 USE_SWD=y MODULAR=n
```

Suddendly my buzzer turns on at full blast.
The buzzer is a static object initialized on first use.

BuzzerClass&
buzzerRef()
{
    static BuzzerClass buzzer;
    return buzzer;
}

This is with gcc 5.3 again.

avtolstoy, 2020-09-23:
This is most likely resolved as of 2.0.0-rc.2 with https://github.com/particle-iot/device-os/pull/2194.