particle-iot/device-os #2065: [Gen 2] TCP socket close triggers 180 sec wait and negativ signal strength

perotom, 2020-03-31:
## Bug Report

### Expected Behavior

TCP sockets should not affect `Modem::getSignalStrength`. 

### Observed Behavior

If your software periodically polls the cellular signal strength with `Cellular.RSSI()` and it happens that the remote host disconnects the device in that exact moment, the modem code will block for 180 sec and the returned signal strength will be negative.

### Root cause of the problem

If you enable AT command logging, the root problem can be easily spotted.
This is how `Cellular.RSSI()` normally looks like:

```
[ Modem::getSignalStrength ] = = = = = = = = = =
 59075.440 AT send       4 "AT\r\n"
 59075.444 AT read OK    6 "\r\nOK\r\n"
 59075.444 AT send      13 "AT+COPS=3,2\r\n"
 59075.449 AT read OK    6 "\r\nOK\r\n"
 59075.449 AT send      10 "AT+COPS?\r\n"
 59075.455 AT read  +   24 "\r\n+COPS: 0,2,\"23201\",2\r\n"
 59075.457 AT read OK    6 "\r\nOK\r\n"
 59075.457 AT send       8 "AT+CSQ\r\n"
 59075.462 AT read  +   14 "\r\n+CSQ: 26,1\r\n"
 59075.463 AT read OK    6 "\r\nOK\r\n"
```

But if a socket gets closed at that moment, it looks like that:

```
[ Modem::getSignalStrength ] = = = = = = = = = =
 59085.448 AT send       4 "AT\r\n"
 59085.451 AT read OK    6 "\r\nOK\r\n"
 59085.451 AT send      13 "AT+COPS=3,2\r\n"
 59114.708 AT read  +   14 "\r\n+UUSOCL: 1\r\n"
Socket 1: handle 1 closed by remote host
socketFree(1)
```
and blocks for 180 seconds before returning a negative value. So if we look at the modem code of `Modem::getSignalStrength`:

https://github.com/particle-iot/device-os/blob/af50fb1b6017412c3719d876f602500a2c47dd23/hal/src/electron/modem/mdm_hal.cpp#L1468-L1470

we see that the `COPS_TIMEOUT` is 180 seconds. So it looks like the parser is not correctly handling the modem's response. 

### Possible solution

### Quick & dirty
Just decrease the timeout for `AT+COPS=3,2` command to something like 1 second. This will probably not work as `AT+COPS?` will be the next command which will have exactly the same problem.

### Clean
Investigate further if or why modem it not sending an answer to the `AT+COPS=3,2` command. Further logging is needed.

### Affected products

Tested on Electron (SARA-U270) with DeviceOS 1.4.2. May affects all cellular products.

### Steps to Reproduce

Any TCP client in combination with `Cellular.RSSI()`. For example take an [MQTT example](https://github.com/hirotakaster/MQTT) and add periodically `Cellular.RSSI()` checks. The error occurs when `Cellular.RSSI().getStrength() < 0`. You can also observe the error based on the AT commands. To enable AT logging add: `SerialLogHandler logHandler(LOG_LEVEL_ALL)`.

avtolstoy, 2020-03-31:
Thanks for a detailed report! I've edited the title, Gen 3 should not be affected, as internal modem TCP/IP stack is not used on such devices.

perotom, 2020-07-17:
Further observations:
This bug not only happens when the connection drops but also when the server sends any responds. Example log:
```
[ Modem::getSignalStrength ] = = = = = = = = = =
  5050.462 AT send       4 "AT\r\n"
  5050.466 AT read OK    6 "\r\nOK\r\n"
  5050.466 AT send      13 "AT+COPS=3,2\r\n"
  5051.272 AT read  +   16 "\r\n+UUSORD: 1,2\r\n"
Socket 1: handle 1 has 2 bytes pending
```

After that it freezes for 180 seconds like described in the initial issue.

keeramis, 2020-09-01:
Hi @perotom ! Thanks for the issue logs. Could you provide a longer version on this log where there is information on what happens after this 180 sec. It would be interesting to check if the radio is completely unresponsive or if it came back up after the 180 sec timeout. We have checked internally, but unable to reproduce this case where there is 180 sec block on the `AT+COPS=3,2` command. Would it be possible to check which cell firmware version that this test was run on? Thanks!
(This AT log is from internal testing)
```
        27.367 AT send      13 "AT+COPS=3,2\r\n"
    27.371 AT read  +   16 "\r\n+UUSORD: 2,1\r\n"
Socket 2: handle 2 has 1 bytes pending
    27.373 AT read OK    6 "\r\nOK\r\n"
```

avtolstoy, 2020-09-23:
Things should recover much better on 2.0.0-rc.2 with https://github.com/particle-iot/device-os/pull/2198 in case of various modem failures.

Since this was originally reported for 2G/3G (U270), please do make sure that the modem is not browning due to absence of the battery.

I'm going to close this issue for now, please do re-open if it still persists.

closed 2020-09-23 by avtolstoy