particle-iot/device-os #2076: TCPClient write is blocking, even with timeout 0

elcojacobs, 2020-04-08:
I'm back with yet another case of "the network stack doesn't do what I expect".

The screenshot below really says it all:

[image]

The TCPClient is set to non-blocking, the non-blocking flags are set in the call, but still, my application hangs on the TCP write, indefinitely.
Look at the local variables at the top left, look at the call stack.

This happened at a client too, causing 500L of fermenting beer to get much colder than it should have been.  It is hurting my company's reputation to have a sometimes unstable product, yet I am powerless in fixing it because the sources in which the error occurs are unavailable.

Now can I please, please, please sign an NDA to get the WICED source to properly debug this?
This is far from the first time that I have reported issues with the network stack. And now I have been debugging this one for days too.

It might be possible that I have a bug in my own code, somehow overwriting the stack of the network thread, but I cannot find it and I am restrained due to the closed sources.

I have run the code in cross-compiled build with -fsanitize and -fundefined. I have also ran it with valgrind. No memory errors detected. So I am back at SWD debugging.

As seen in the debugging screenshot, the app thread yields. 
*Why does it yield in a non-blocking TCP write with zero timeout?*

Do you understand how incredibly frustrating it is for me to have this black box that doesn't work properly?
@avtolstoy I think you're the one at Particle most likely to take this seriously. Please help me find this bug.

I std::move the TCPClient returned by TCPServer.available() into class member, and store the connections in a vector. Does the network stack somehow return a client by value, but keeps references to it? Is the TCP stack not thread-safe?

Our code is all open-source, so I would be happy to give you the steps to reproduce. But I don't have a simple test app for this complex issue.
The only way I can reliably reproduce this is to let you take these steps:
- Check out our code base from GitHub
- Build it with the supplied makefile
- Use st-flash to set EEPROM to a state that reliably triggers the bug in our app

Happy to give you the steps if you are willing to do this, but I'd gladly investigate more myself and save you the time. If the error is due to a memory overwrite somewhere else in the code, it is hard to find. But I will need sources. And maybe a more direct form of communication would help too.

elcojacobs, 2020-04-08:
I might not be able to see the source, but I can see the variables. These 2 are noteworthy:
`
pvBuffer: 0x0
xTicksToWait: 10000
`
[image]

Maybe that's normal behavior, but I wouldn't know. because I cannot see the f*ing source!

elcojacobs, 2020-04-08:
https://github.com/particle-iot/device-os/blob/0140a25e0bd4c644e6911cc452006a701e802e9d/hal/src/photon/socket_hal.cpp#L1045

This line of code. 
```
wiced_tcp_send_flags_t wiced_flags = timeout == 0 ? WICED_TCP_SEND_FLAG_NONBLOCK : WICED_TCP_SEND_FLAG_NONE;
```

It is included done for `is_tcp(socket)`, not for `is_client(socket)`.

is_client is the path that is executed. Related?

elcojacobs, 2020-04-09:
I think I have found a potential cause for the lockup!

I had an out of memory handler, which stupidly tried to send out a message that it was out of memory.
So the result would be a total lockup because the memory handler tried to send something over TCP, which requires memory allocation. This broke the entire thread handler / RTOS.

With the custom handler removed, TCP messages might fail, but at least it doesn't lock up.

I'll close this issue. :blush:

closed 2020-04-09 by elcojacobs

No1089, 2020-04-17:
@elcojacobs I'm glad this issue was resolved - in the future I'd encourage you to file bugs via our support.particle.io portal to ensure we get the right pair of eyes on it if there is a bug in our codebase.

elcojacobs, 2020-04-18:
Are you saying that GitHub issues is not the right place to report issues? Or are you talking about insights in the wicked stack in particular?

No1089, 2020-04-20:
@elcojacobs Yes, we prefer that issue are reported using the support portal. It allows us much better visibility and avoids issues being created that are related to something other than that which Particle has control over.
We have an internal issue escalation process where incoming issues are discussed and prioritised with engineering weekly - GitHub issues are not part of that discussion.

elcojacobs, 2020-04-21:
I find that strange. GitHub issues is the user facing issue tracker for most open source projects. Going to the support page, it is mentioned as the place for software issues. This is what GitHub is for.
It should be part of your discussions.

You are saying, don't post on GitHub or the community forum, but use this less user visible portal instead?

No1089, 2020-04-22:
@elcojacobs As this issue above clearly demonstrates the problem was not with the platform, but with the implementation. GitHub is not a support platform - our support portal/forum filters out individual issues from those affecting the entire platform.

elcojacobs, 2020-04-22:
Yes, I agree with you for this issue. I thought it was a platform issue, but when I found out it was not I closed the issue.

No1089, 2020-04-22:
Which is we would like to filter issues through the portal first, and if an issue is identified you are welcome to open an issue in Github. We will be aware of it internally with confirmed proof/examples from the interaction - and we use that to escalate the issue to the relevant team.

Our goal is to get bugs resolved as quickly as possible, and the support portal is the most efficient way to do so.