particle-iot/device-os #2135: Errors when compiling release/stable: "array subscript 15 is outside array bounds of 'char[1]'" in core_hal.c

Klikini, 2020-06-22:
## Bug Report

I cannot compile the `release/stable` branch.

### Expected Behavior

What should it do?

Unless I am mistaken, `make clean all PLATFORM=photon -s program-dfu` should successfully compile and flash the board.

### Observed Behavior

What did it actually do?

```
$ make clean all PLATFORM=photon -s program-dfu
  /home/andy/photon-fw/modules/photon/system-part1/makefile /home/andy/photon-fw/modules/photon/system-part2/makefile /home/andy/photon-fw/modules/photon/user-part/makefile
src/photon/core_hal.c: In function 'SysTickChain':
src/photon/core_hal.c:110:88: error: array subscript 15 is outside array bounds of 'char[1]' [-Werror=array-bounds]
  110 |     void (*chain)(void) = (void (*)(void))((uint32_t*)&link_interrupt_vectors_location)[IRQN_TO_IDX(SysTick_IRQn)];
      |                                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
src/photon/core_hal.c:40:13: note: while referencing 'link_interrupt_vectors_location'
   40 | extern char link_interrupt_vectors_location;
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/photon/core_hal.c: In function 'Mode_Button_EXTI_irq':
src/photon/core_hal.c:140:88: error: array subscript 39 is outside array bounds of 'char[1]' [-Werror=array-bounds]
  140 |     void (*chain)(void) = (void (*)(void))((uint32_t*)&link_interrupt_vectors_location)[IRQN_TO_IDX(BUTTON1_EXTI_IRQn)];
      |                                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/photon/core_hal.c:40:13: note: while referencing 'link_interrupt_vectors_location'
   40 | extern char link_interrupt_vectors_location;
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/photon/core_hal.c: In function 'HAL_Core_Setup_finalize':
src/photon/core_hal.c:130:9: error: array subscript 15 is outside array bounds of 'char[1]' [-Werror=array-bounds]
  130 |     isrs[IRQN_TO_IDX(SysTick_IRQn)] = (uint32_t)SysTickChain;
      |     ~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
src/photon/core_hal.c:42:13: note: while referencing 'link_ram_interrupt_vectors_location'
   42 | extern char link_ram_interrupt_vectors_location;
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [../build/module.mk:269: ../build/target/hal/platform-6-m/./src/photon/core_hal.o] Error 1
make[1]: *** [../../../build/recurse.mk:12: hal] Error 2
make: *** [makefile:86: /home/andy/photon-fw/modules/photon/system-part1/makefile] Error 2
```

### Steps to Reproduce

How did you make it do that, and which system/user firmware versions were tested?

1. I just did a fresh install of Kubuntu 20.04 (running on a ThinkPad T590, not a VM or anything) for this and I have not done anything else to the system. I'm using gcc 9.3.0-10ubuntu2 and gcc-arm-none-eabi version 15:9-2019-q4-0ubuntu1
1. `sudo apt install gcc-arm-none-eabi dfu-util libarchive-zip-perl git`
1. `git clone --recursive https://github.com/particle-iot/firmware.git ~/photon-fw`
1. `git checkout release/stable` (as of 8pm CDT on June 21, 2020)
1. `cd ~/photon-fw/modules`
1. `make clean all PLATFORM=photon -s program-dfu`

### Test App

This is pretty much the program I'm trying to compile. I did not make any changes to the repo after cloning it.

```c
void setup() {
  // A minimal test case to reproduce the issue is super helpful!
}

void loop() {

}
```

### References

Links to the Community, Docs, Other Issues, etc..

I'm following the instructions at https://github.com/particle-iot/device-os/blob/release/stable/docs/gettingstarted.md

---

Am I doing something wrong? I'm happy to do anything I can to help with this, just let me know what you need from me :smile:

avtolstoy, 2020-06-22:
At the moment only GCC 5.3.1 is supported with existing releases (`release/stable` is 1.5.2). `develop` branch supports GCC 9.

closed 2020-06-22 by avtolstoy

Klikini, 2020-06-22:
`develop` works. Thanks!

Klikini, 2020-06-22:
Or so I thought...`develop` compiles but won't flash due to the following:

```
../../../build/target/system-part2/platform-6-m/system-part2.elf section `.text' will not fit in region `APP_FLASH'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `APP_FLASH' overflowed by 7460 bytes
collect2: error: ld returned 1 exit status
make[1]: *** [../../../build/module.mk:232: ../../../build/target/system-part2/platform-6-m/system-part2.elf] Error 1
make[1]: Leaving directory '/home/andy/photon-fw/modules/photon/system-part2'
make: *** [makefile:86: /home/andy/photon-fw/modules/photon/system-part2/makefile] Error 2
```

I tried compiling `release/stable` using gcc 5.3.1 and got the same error as before (after running `make clean`).

avtolstoy, 2020-06-22:
This is most likely something with your local setup. Photon build on `develop` successfully passed: https://travis-ci.org/github/particle-iot/device-os/jobs/699653709

```
Running runmake DEBUG_BUILD="n" PLATFORM="photon" COMPILE_LTO="n" in modules, build directory /firmware/build/jobs/18
  /firmware/modules/photon/system-part1/makefile /firmware/modules/photon/system-part2/makefile /firmware/modules/photon/user-part/makefile
   text	   data	    bss	    dec	    hex	filename
 240836	    516	    260	 241612	  3afcc	/firmware/build/jobs/18/system-part1/platform-6-m/system-part1.elf
   text	   data	    bss	    dec	    hex	filename
 257776	   2236	  40060	 300072	  49428	/firmware/build/jobs/18/system-part2/platform-6-m/system-part2.elf
   text	   data	    bss	    dec	    hex	filename
   3180	    108	    176	   3464	    d88	/firmware/build/jobs/18/user-part/platform-6-m/user-part.elf
 ✓ SUCCESS 
```