particle-iot/device-os #2192: in STARTUP block, PIN_MAP is uninitialized and fast_pin functions cause hardfault (p1, 2.0.0-rc.1)

elcojacobs, 2020-09-08:
## Bug Report

### Expected Behavior

The STARTUP block main purpose is probably to do early pin initializations. This should not hardfault.

### Observed Behavior
Hard fault after moving function bodies of unrelated code from header to cpp files:
https://github.com/BrewBlox/brewblox-firmware/commit/ea8f9e8e9771c58751d04ebbc3f7eeadf4651c37

When SWD debugging my app I confirmed that PIN_MAP was pointing at 0x0, causing the hardfault on this line:
[image]

value of PIN_V3_TOP3 is 28 (P1S4) at the breakpoint (valid index into PIN_MAP), but PIN_MAP is 0x0.
I don't know why the lines before it didn't cause problems.

pin_map.h has this line:
```
static Hal_Pin_Info* PIN_MAP = HAL_Pin_Map();
```

When is this variable initialized? It is a hard question actually. HAL_Pin_Map() is a dynalib function and this is a global variable initialized in a header that could be included in multiple source files.

Perhaps it could be rewritten to be initialized on first use instead, to avoid the problems of the initialization order of globals.

### Steps to reproduce
I have not tried to confirm the behavior with a simpler test app.
This fixed it:
https://github.com/BrewBlox/brewblox-firmware/commit/88db41eb90c028bc5d5a547a38a0a660819e8fde

avtolstoy, 2020-09-08:
> When is this variable initialized? It is a hard question actually. HAL_Pin_Map() is a dynalib function and this is a global variable initialized in a header that could be included in multiple source files.

It's a static variable that is defined in the header. It's initialized locally in every compilation unit (source file) that includes `fast_pin.h`. The order of initialization of static objects (`STARTUP()` also creates a static object) is the same as they are defined. I assume what happens in your case is similar to this:

```c++
#include "spark_wiring_startup.h"

void* testval = (void*)0x5a5a5a5a;
void test();

STARTUP(test());

#include "application.h"

void test() {
    testval = PIN_MAP;
}
```

If `application.h` is moved above `STARTUP(test());`, the issues goes away.

elcojacobs, 2020-09-08:
Thanks for the quick reply.

I don't use "application.h", but I use the more specific hal headers that I need in each CPP file.
I don't like to include everything in every cpp file, but I understand it works for simpler programs and novice users.

The relevant files are:
https://github.com/BrewBlox/brewblox-firmware/blob/develop/platform/spark/modules/Board/Board.cpp
https://github.com/BrewBlox/brewblox-firmware/blob/develop/app/brewblox/main.cpp

I don't know what exactly is required from "application.h", but instead of the pin map pointer being zero, a compiler error would be nice to catch that whatever "application.h" defines is available.

avtolstoy, 2020-09-08:
I cannot say exactly what goes wrong, but there is some kind of an issue with multiple source files. We should probably remove that `PIN_MAP` declaration from the header and use instead something like:
```c++
inline Hal_Pin_Info* fastPinGetPinmap() {
    static Hal_Pin_Info* pinMap = HAL_Pin_Map();
    return pinMap;
}
```

Which would have a cost of a single dynalib call on first access, making it comparable to what we have right now.

elcojacobs, 2020-09-08:
Yep, that's exactly the pattern that I meant.
Perhaps you can make the pointer const too. constexpr is probably not an option due to c.

I also thought it could be for C/C++ linkage issues, but the headers seem to have the necessary extern "C" wrappers.

PR #2194 (2020-09-09): [wiring] fastpin: use function with static variable to overcome initialization order woes
### Problem

#2192 

### Solution

The initialization order between the translation units is undefined, so the inline static variable which gets a reference to the pin map from the HAL can be initialized at any point in time during traversal of the global ctor list, making it impossible to use fastpin functions within global constructors e.g. `STARTUP()` in a different translation unit that uses a function to call fastpin.

### Steps to Test

### Example App

```c++
// a.cpp
const void* testval = (const v [... cut]
closed 2020-09-09 by avtolstoy