particle-iot/device-os #2244: Low Level Input/Output Slow on OS 2.0.0-rc4

divrajBevie, 2020-11-30:
</details>

## Bug Report
Tested with pinSetFast, the speed at which this is seting the pin high is slower on firmware targeted 2.0.0-rc4 vs the 1.5.2 firmware. Testing on a device running 2.0.0-rc4.
### Expected Behavior

Tested by printing out micros seconds taken to pinsethigh in loop. on 1.5.2 it takes <1 us.
### Observed Behavior

2.0.0rc4 it takes >1us sometimes >2us

### Test App

```c
void setup() {
  // A minimal test case to reproduce the issue is super helpful!
  Serial.begin(9600);
  pinMode(D6, OUTPUT);

}

void loop() {
      unsigned long drawTime = micros();
      pinSetFast(D6);
      drawTime = micros() - drawTime;
      Serial.println(drawTime);
      pinResetFast(D6);
}
```

## Feature/Enhancement Request

Using this to drive a parallel lcd (8-bit), the writing/reseting speed of these low level input/output is considerably slow. Causing the the lcd to run much slower.

avtolstoy, 2020-11-30:
Thanks for the issue report. @divrajBevie could you please clarify which platform in particular?

divrajBevie, 2020-11-30:
@avtolstoy Apologies should of clarified in post. This issue is tested on a P1

avtolstoy, 2020-12-02:
Just wanted to add another point that the performance hit after we've introduced https://github.com/particle-iot/device-os/pull/2194 for https://github.com/particle-iot/device-os/issues/2192 is not microseconds, but 5-10x and we are still below 1us:

```
0000004429 [app] INFO: 2.0.0: set 575 ns / reset 550 ns
0000009575 [app] INFO: 1.5.4-rc.2: set 100 ns / reset 50 ns
```

`micros()` is a heavy call that goes through dynamic linking interface and also requires interrupts to be disabled and re-enabled. `System.ticks()` (that counts at CPU clock) should provide a more accurate timing here e.g.

```c++
void loop() {
    auto t0 = System.ticks();
    pinSetFast(D7);
    auto t1 = System.ticks();
    pinResetFast(D7);
    auto t2 = System.ticks();
    Log.info("%s: set %lu ns / reset %lu ns", System.version().c_str(), (uint32_t)((t1 - t0) * 1000000000ULL / SystemCoreClock), (uint32_t)((t2 - t1) * 1000000000ULL / SystemCoreClock));
}
```

We'll look into optimizing a bit more, however there is still probably going to be some performance hit.

closed 2020-12-09 by avtolstoy

divrajBevie, 2021-02-17:
Hey @avtolstoy 
I noticed that this is a bit more optimised in 3.0.0-rc.1 OS however I was wondering if there was a way for me to revert it back to the settings used in 1.5.2 (if so can you show me what will be required to do so). The current speed is still not to our liking as it really affects the speed of calling these set/resets multiple times.

`0000000877 [app] INFO: 3.0.0-rc.1: set 241 ns / reset 258 ns`

avtolstoy, 2021-02-17:
You can take a look at #2194 diff (https://github.com/particle-iot/device-os/pull/2194/files) and try to replicate the previous behavior:
1. Cache `HAL_Pin_Map()`
2. Use cached pinmap pointer in set/reset routines

PR #2251 (2020-12-09): Always inline fastPinGetPinmap to speed pin sets
<details>
  <summary><i>Always inline fastPinGetPinmap in order to speed pin set</i></summary>

Also add test to catch unintentional slowing of pin set in the future.  
</details>

### Problem

For some users, calling `pinSetFast` or `pinResetFast` was noticeably slower on 2.0.0-rc.4 or later than on earlier releases. 
 
### Solution

The 2.0.0 development line worked around an issue with initialization order by changing the way the HAL pin map was initialized and subsequently accessed.  This introduced some overhead in each call to fastPinGetPi [... cut]