particle-iot/device-os #2282: Reconnection Process Ram Usage. Not Free'd

divrajBevie, 2021-02-17:
<details>
  <summary><i>More Ram Usage Overtime by reconnection Process</i></summary>

When the device disconnects from the Cloud, a chunk of ram is used and never free'd.  If this device keeps connecting and disconnecting, then more and more ram is used.

I was trying to replicate a scenario where a few of my users devices are "Freezing" and the OS is not handling a SOS restart.
## Bug Report

### Expected Behavior
The RAM used by the reconnection process should be freed once the device connects back up to the CLOUD.

### Observed Behavior

More and more and ram is used by the OS until WiFi fully disconnects. (PLEASE SEE ATTACH LOG FILE)

Additionally, on OS 1.5.2 when the memory was supposedly enough (20kb) the controller froze and did not restart (similar to seen by customers)

### Steps to Reproduce
 *Tested on:* 

 - Particle P1
 - Device OS 3.0.0-rc.1 and OS 1.5.2
 - Hotspot

Steps: 

 1. Connect device to only a single hotspot (using a secondary phone)
 2. After connection is established, turn off your Mobile data (not the hotspot)
 3. After connection is disconnected from the cloud, turn the device again
 4. Keep repeating steps 2 and 3.
### Test App

```c
SYSTEM_THREAD(ENABLED);
SYSTEM_MODE(SEMI_AUTOMATIC);
void setup() {
  System.disableUpdates();
  Particle.connect();
}

void loop() {
	fragmentationCheck();
}
void  fragmentationCheck() {
	static  uint32_t  free_memory = 0;
	static  uint32_t  largest_block = 0;
	static  uint32_t  prevCheckMillis = 0;
	static  int  RSSI = 0;
	static  bool  printRightAway = true;
	static  uint32_t  counter = 0;

	if (millis() -  prevCheckMillis  > (2000  *  1) ||  printRightAway) {
		runtime_info_t  info;
		memset(&info, 0, sizeof(info));
		info.size = sizeof(info);
		HAL_Core_Runtime_Info(&info, NULL);
		largest_block = info.largest_free_block_heap;
		free_memory = System.freeMemory();
		RSSI = WiFi.RSSI();
		counter++;
		char  memoryCharArray[65];
		snprintf(memoryCharArray, sizeof(memoryCharArray),
		"%u: %u,%u,%d,%u,%u", counter, free_memory, largest_block, RSSI, Time.now(), (free_memory-largest_block));

		Serial.println(memoryCharArray);
		prevCheckMillis = millis();
		printRightAway = false;
	}
}
```
---

## Feature/Enhancement Request

Users are having issues where their devices are freezing (i.e. the device is not responding to any input, it is not online). And the only way to reset this is to physically reset the device. Hopefully this linked issue is a linked issue.

[Log.txt](https://github.com/particle-iot/device-os/files/5998300/Log.txt)

avtolstoy, 2021-02-19:
@divrajBevie Thanks for the bug report. Is there anything else that needs to be done to reproduce the issue e.g. forcing WiFi to get disconnected etc? We've ran this on both Photon and P1 for a while and the amount of free heap always comes back to a certain level while there are some dips during the cloud handshake.

rtheil, 2021-02-19:
We are having a similar issue with Photon devices running OS <1.5. We cannot figure out where the issue lies, only that it is tied to loss of connectivity. Just not sure if it's associated with loss of wifi or internet. As @avtolstoy said, our results running your script are that repeated loss of internet connectivity results in a recovery of free memory. HOWEVER, it seems that only occurs when the photon semi-restarts (double flash of orange light). Loss of connectivity, by itself, causes a drop in free memory of about 2k, so it makes me think that the constant reconnection and drop must happen before the double-orange-flash. I am working to set up an environment where we can further test this in order to try and reproduce consistently, as we have customers who regularly experience this issue.

avtolstoy, 2021-02-19:
Thanks for additional info.

> photon semi-restarts (double flash of orange light).

Just to clarify that's a cloud connection error: https://docs.particle.io/tutorials/device-os/led/photon/#red-blink-basic-errors

polystyreneMan, 2021-02-19:
I tested using both 1.4.2 and 1.5.2, and can confirm similar. It looks like if I restore the connection before the orange flash, there's a drop in memory for every disconnect. If I let it go through the orange flash, it recovers everything, even from multiple drops due to reconnects without orange light.

The same happens with our own firmware of course, which uses a lot more memory to start. I tend to get hard faults rather than a locked up device though, but we have a lot of customer reports of frozen LEDs.

divrajBevie, 2021-02-21:
I have noticed that when the wifi is disconnected, the Ram is restored. Currently, I have roughly 30KB of free ram and with the steps above (toggling the internet and not the wifi), roughly 2kb ram is used up.
Also, I have noticed that sometimes while the reconnection is happening, the device also freezes up completely. Any idea what could be causing this?

divrajBevie, 2021-02-22:
In my user code, I am calling Particle.publish every second as this is required. I am checking if Particle.connected() is true before doing that. However, when the internet is disconnected (i.e. mobile data is turned off, while the device is connected to phone hotspot), the Particle.publish is still being called atleast for another 10 seconds because Particle.connected() is still returning true. Could this cause the blocking described above?

avtolstoy, 2021-02-24:
@divrajBevie Yes, this could be the source of the problem. Calling `Particle.publish()` while a TCP device is in a quasi-connected state (`Particle.connected()` reports `true`, however the connection is no longer functioning and will be broken in some tens of seconds) will continue to fill up the TX buffers, which may very well cause this behavior.

divrajBevie, 2021-02-24:
Is there a way to monitor the TX buffers? And possibly be used to stop Particle.publish() if TX buffers are filled up.

avtolstoy, 2021-02-24:
@divrajBevie There isn't one, but I can perhaps suggest seeing if using `WITH_ACK` with publishes and capturing the return value (bool, which will cause the calling code to block until the publish is acknowledged by the cloud) or capturing the returned future (https://github.com/particle-iot/app-notes/tree/master/AN009-Firmware-Examples/Stop-Sleep-Cellular#the-future) helps in any way.

polystyreneMan, 2021-02-24:
For what it's worth, our product uses the `PublishQueueAsyncRK` library and uses `WITH_ACK` as well.

@divrajBevie, does the sample app above ever freeze up the device? I think we're getting closer to reproducing device lock-ups on our product firmware more reliably, but it's still going to be tough to find the root of the issue.

avtolstoy, 2021-02-25:
Hey everyone, I wanted to clarify one thing here as there are mentions of 1.4, 1.5 and older releases: we may have a potential leak in Device OS releases for Photon and P1 < 2.0.0 when stopping threads, which comes from newlib (our standard C library implementation) incompatibility between Device OS side and WICED (Cypress WiFi stack used on Photons and P1s). See https://github.com/particle-iot/device-os/pull/2126 and somewhat related https://github.com/particle-iot/device-os/pull/2126.

These two changes should also have a positive impact on the amount of total RAM available (about 10% more for Photons and P1).

A kind ask is to confirm whether the issues raised are reproducible on 2.0 or later releases.

polystyreneMan, 2021-02-25:
I'm seeing similar behavior using the test firmware above on v2.0.1, by toggling mobile data on and off while connected to my phone's hotspot.:

2: 48296,48296,2,1614273933,0
3: 48296,48296,2,1614273935,0
4: 48296,48296,2,1614273937,0
5: 48296,48296,2,1614273939,0
6: 48296,48296,2,1614273941,0
7: 48048,48048,-16,1614273943,0
8: 48032,45488,-17,1614273945,2544
9: 48032,45488,-21,1614273947,2544
10: 48032,45488,-9,1614273949,2544
11: 48032,45488,-17,1614273951,2544
12: 48032,45488,-12,1614273953,2544
13: 48032,45488,-9,1614273955,2544
14: 48032,45488,-17,1614273957,2544
15: 48032,45488,-24,1614273959,2544
16: 48032,45488,-28,1614273961,2544
17: 46768,45488,-26,1614273963,1280
18: 46768,45488,-19,1614273965,1280
19: 46768,45488,-23,1614273967,1280
20: 46768,45488,-28,1614273969,1280
21: 46768,45488,-23,1614273971,1280
22: 42688,42536,-30,1614273973,152
23: 46648,45488,-25,1614273976,1160
24: 46648,45488,-22,1614273978,1160
25: 46648,45488,-29,1614273980,1160
26: 46648,45488,-30,1614273982,1160
27: 46648,45488,-45,1614273984,1160
28: 46648,45488,-32,1614273986,1160
29: 46648,45488,-29,1614273988,1160
30: 46648,45488,-24,1614273990,1160
31: 45384,44224,-27,1614273992,1160
32: 45384,44224,-25,1614273994,1160
33: 45384,44224,-36,1614273996,1160
34: 45384,44224,-31,1614273998,1160
35: 45384,44224,-15,1614274000,1160
36: 45456,44224,-17,1614274002,1232
37: 45272,44224,-22,1614274004,1048
38: 45272,44224,-16,1614274006,1048
39: 45272,44224,-15,1614274008,1048
40: 45272,44224,-13,1614274010,1048
41: 45272,44224,-23,1614274012,1048
42: 45272,44224,-13,1614274014,1048
43: 45272,44224,-15,1614274016,1048
44: 45272,44224,-15,1614274018,1048
45: 45272,44224,-15,1614274020,1048
46: 44008,42960,-20,1614274022,1048
47: 44008,42960,-22,1614274024,1048
48: 44008,42960,-16,1614274026,1048
49: 44008,42960,-23,1614274028,1048
50: 44008,42960,-23,1614274030,1048
51: 40160,40040,-23,1614274032,120
52: 43896,42960,-19,1614274034,936
53: 43896,42960,-23,1614274036,936
54: 43896,42960,-27,1614274038,936
55: 43896,42960,-23,1614274040,936
56: 43896,42960,-21,1614274042,936
57: 43896,42960,-25,1614274044,936
58: 43896,42960,-25,1614274046,936
59: 43896,42960,-14,1614274048,936
60: 42632,41696,-21,1614274050,936
61: 42632,41696,-20,1614274052,936
62: 42632,41696,-24,1614274054,936
63: 42632,41696,-16,1614274056,936
64: 42632,41696,-24,1614274058,936
65: 42632,41696,-24,1614274060,936
66: 42520,41696,-28,1614274062,824
67: 42520,41696,-31,1614274064,824
68: 42520,41696,-21,1614274066,824
69: 42520,41696,-19,1614274068,824
70: 42520,41696,-21,1614274070,824
71: 42520,41696,-17,1614274072,824
72: 42520,41696,-22,1614274074,824
73: 42520,41696,-25,1614274076,824
74: 42520,41696,-22,1614274078,824