particle-iot/device-os #2294: BLE: Utilize the Service changed characteristic

maxarndt, 2021-03-04:
## Feature/Enhancement Request

### Use case

We are currently developing a data logger which is run by a Particle Boron.
Our customer wants to communicate with the data logger with a custom iOS app via BLE. 
The customer must be able to change the name of each of these data loggers via the iOS app.

### Current state

#### Our Particle firmware

We provided an writable BLE characteristic which calls `BLE.setDeviceName(...)` API in its callback which has been introduced with PR #1847. 
And of course we do advertise our service and append a `localName` in this packet.

#### iOS behavior

When a BLE peripheral is first discovered by an iOS device the `localName` from an advertisement packet is shown to the user.
After connecting to a device the GAP name will be discovered and cached.
For subsequent discoveries the cached GAP name is shown to the user instead of the data in an advertisement packet.

This is what we have discovered with `iOS 14.4` and what has been reported a few years ago in the [Apple Developer Forum](https://developer.apple.com/forums/thread/72343?answerId=210896022#210896022).

### Problem

Unfortunately, after the user changes the name of our product via the iOS app the new name is not shown in the next connection attempt.
The reason is the caching of the GAP name which seems to be refreshed right after the BLE connection gets established.
As a result, any change of the device name initiated by an user is performed after the cache gets refreshed. 
Finally, the new name will only be displayed correctly when the next but one connection attempt is made.

### Solution

Utilize the `Service changed characteristic (0x2A05)` which is present in the `Generic Attribute Service (0x1801)` after the device name has been changed.
From my point of view it would be ideal that the device-os sends a proper indication via the `Service changed characteristic` to inform the iOS client to refresh any changed data.

I can see in the PacketLogger app on macOS that our Particle devices do set up the `Service changed characteristic (0x2A05)` and that our iOS devices seem to subscribe/activate the indications of this characteristic after connecting to the device.
But as far as I can see this characteristic is not used to inform the client about corresponding changes.
I tried to send the indications my own using the `sd_ble_gattas_service_changed(…)` function but I couldn't get it work because the required `conn_handle` is managed by the Softdevice which is as far I know not accessible from our code.

Is it possible to handle this internally in the device-os to better meet the BLE spec?

XuGuohui, 2021-03-10:
After sending indication of service changed the iOS will disconnect the peripheral actively and the device name is updated still only after the next connection is established. In that case, sending service changed indication doesn't help fix the issue.

From the context, your customer is using a custom APP to communication with Particle device. In the device application we can update the advertising data with new device name and let the APP filter scanned device with the new device name in the advertising data and show it to user.

XuGuohui, 2021-03-15:
@maxarndt I'm gonna close this issue. Feel free to re-open it if there is any question.

closed 2021-03-15 by XuGuohui