particle-iot/device-os #2296: Nanopb v0.4.0 Conflict

Awbmilne, 2021-03-09:
## Nanopb v0.4.0 conflict

Forum post: https://community.particle.io/t/nanopb-v0-4-4-conflict/59604

Currently, the Particle device OS seems to use Nanopb internally for serializing of data sent to/received from the Particle cloud. This is super cool! Great to see open source implementations of open standards in other open source code.

**The issue arrives when I want to use a different version of Nanopb with my application.**\
I have an existing app that I'm looking to move to Particle, But the Nanopb-0.4.0 that i'm using butts heads with the Nanopb-0.3.9.3 that is built into the Particle Device OS.

I have made some attempts at using Nanopb 0.3.9.3 for re-compiling my Proto files, but the 2 versions of Nanopb rely on different Proto file standards, meaning I have to change the proto files, the protoc compiler version for the non-embedded ends, and the code for the other systems, etc.

Of course, this is not a trivial thing, neither for my app or the device OS.

**What Im looking for:**\
I'm looking for a way to include the Nanopb-0.4.0 in my app, but leave the Nanopb-0.3.9.3 for the Particle Device OS to use. Is there a way to make this work?\
There have been similar issues in the past with Nanopb being included, namely  #1502. This issue could also be resolved by segregating the 2 instances, leaving the user's app to do its own thing.

This isn't ideal since it means there are 2 versions of the same library in the final binary, but its far better than having to rewrite an entire application on the user end or the Particle end.

I may be talking out my back pockets, but is there any way to accomplish this? Or something else to the same effect?

Awbmilne, 2021-03-09:
## Temporary Fix
_(Also noted in Forum)_

Thanks to the Nanopb folks being stellar C devs, it is easy to rename every symbol and preprocessor directive with a little regex.

**Regex:**
find: `(pb_)`
replace `[leader]_$1`
(_Note: the search is case agnostic. Or just run with_ `(pb_|PB_)`)

Where `[leader]` can be whatever you like, ex: `MY` for `MY_pb_...`\
This replace needs to be run in the source files of your project, and any files that rely on Nanopb. Also run this in your generated NanoPB files (ex. `my_file.pb.c` `my_file.pb.h`)

Then, You also need to rename the headers and source files for the Nanobp files:
`pb_common.c` --> `MY_pb_common.c`
`pb_common.h` --> `MY_pb_common.h`
`pb_decode.c` --> `MY_pb_decode.c`
`pb_decode.h` --> `MY_pb_decode.h`
`pb_encode.c` --> `MY_pb_encode.c`
`pb_encode.h` --> `MY_pb_encode.h`
`pb.h`        --> `MY_pb.h`

Make sure the includes of your files match the new names, and you should be good to go!
_(You can fix the includes with easy search/replace as well)_

elcojacobs, 2021-03-12:
I am currently using the particle version, but with some workarounds for functions that are not included in dynalib.
The problem:
- Some functions are included in the system layer through dynalib and some are not. If you include the nanopb files in your build, you will have duplicate functions. If you don't, you'll miss functions.

I have this c file to fill the gap:

https://github.com/BrewBlox/brewblox-firmware/blob/develop/app/brewblox/nanopb_not_in_particle_dynalib.c

If you don't want to use the dynalib functions and completely separate them, perhaps using namespaces would be a nicer solution. I think it would be best if particle put their dependency in a namespace.

Awbmilne, 2021-03-13:
@elcojacobs The issue I am having is that the NanoPB _version_ is wrong. I found your solution before, but it solves the issue of there being functions missing. My issue is that the 2 versions of NanoPB are not compatible (v0.3.9.3 and v0.4.0), So I cannot switch my v0.4.0 based code to use to v0.3.9.3

Namespaces would be a great solution. I tried that, but ran into issues since the Nanopb code is C not C++ (no user-defined namespace). This _could_ be remedied on Particle's side by changing there version of Nanopb to use `.cpp` files and and then implementing a `particle_nanopb` namespace (for example).