particle-iot/device-os #2343: BLE scan returns -270

natanaelmgomes, 2021-07-29:
<details>
  <summary><i>submission notes</i></summary>

```
**Important:** Please sanitize/remove any confidential info like usernames, passwords, org names, product names/ids, access tokens, client ids/secrets, or anything else you don't wish to share.

Please choose one of the following templates **Bug Report** or **Feature/Enhancement Request** and delete the other.

You may also delete this submission notes header if you'd like. Thank you for contributing!
```
</details>

## Bug Report

BLE scan returns -270.

### Expected Behavior

BLE scan should return the number of devices in the scanResults.

### Observed Behavior

BLE scan returns -270 (I don't know wich error is this), and the scanResults is empty.
This also broke the BLE_Group library.

### Steps to Reproduce

Updated to DeviceOS@3.1.0 compiled locally with Vidual Studio Code

### Test App

```c
void loop() {
    int count = BLE.scan(scanResults, SCAN_RESULT_MAX);
    Log.info("Count: %d", count);
    delay(3000);
}
```

rickkas7, 2021-07-29:
This is a bug in Device OS 3.1.0 only. You can work around this by adding this to setup():

```
#if SYSTEM_VERSION == SYSTEM_VERSION_v310
	// This is required with 3.1.0 only
	BLE.setScanPhy(BlePhy::BLE_PHYS_AUTO);
#endif
```

natanaelmgomes, 2021-07-29:
It works!
Thank you rickkas.

closed 2021-07-29 by natanaelmgomes