particle-iot/device-os #2380: WiFi dropout causes hardfault, hardfault handler is broken and P1 doesn't SOS and reset

elcojacobs, 2021-12-04:
## Bug Report

### Expected Behavior

Hardfault handler should blink RED status LED and reboot

### Observed Behavior

Hardfault results in complete system hangup and system doesn't reset.

I built my app with a non-modular build with SWD enabled in an attempt to find the cause of complete system hangups when the wifi becomes unavailable.
The cause of the hangup is inaccessible to me because it is in the closed source WICED files.
However, I noticed that I couldn't break the debugger on hardfaults anymore in 'prvGetRegistersFromStack'. I did looked at the history of the hardfault handler and reverted a change. This made SOS and hardfault work again.

### Steps to Reproduce
- Clone our repo
- In the <repo root>/build directory, run `make program-dfu MODULAR=n PLATFORM=p1 PARTICLE_DEVELOP=y USE_SWD=y APP=brewblox`
- Turn WiFi on and off on router until turning it off causes a hardfault

### Test App
https://github.com/BrewBlox/brewblox-firmware/tree/develop/app/brewblox-particle

### Analysis
Since recently, I had issues with complete hangups of our temperature controllers. Even the watchdog didn't reset the system anymore.

While trying to step debug using SWD, I noticed the system hung in the `HardFault_Handler` function, without jumping into `prvGetRegistersFromStack.

In this commit:

https://github.com/particle-iot/device-os/commit/7b6e2086af907668d395c63404ecc754ee36fe41#diff-cd3f7ac0caef6976249cf63ff8b1dbcfe1c3150d09e3b694a45cfbc3354c85d7

```cpp
void HardFault_Handler(void)
{
    __asm volatile
    (
        " tst lr, #4                                                \n"
        " ite eq                                                    \n"
        " mrseq r0, msp                                             \n"
        " mrsne r0, psp                                             \n"
        " ldr r1, [r0, #24]                                         \n"
        " ldr r2, handler2_address_const                           \n"
        " bx r2                                                     \n"
        " handler2_address_const: .word prvGetRegistersFromStack    \n"
    );
}
```

was changed into:

```
void HardFault_Handler(void)
{
    __asm volatile
    (
        " tst lr, #4                                                \n"
        " ite eq                                                    \n"
        " mrseq r0, msp                                             \n"
        " mrsne r0, psp                                             \n"
        " ldr r1, [r0, #24]                                         \n"
        " ldr r2, =handler2_address_const                           \n"
        " bx r2                                                     \n"
        " handler2_address_const: .word prvGetRegistersFromStack    \n"
    );
}
```

When I revert that change and rebuild the firmware, the `prvGetRegistersFromStack` is entered again, the LED shows SOS flashes and the system reboots instead of completely locking up.

Reading the registers in pulFaultStackAddress, I get the following:
link register is at 
```
0x802ec35 is in worker_thread_main (WICED/RTOS/FreeRTOS/WICED/../../wiced_rtos_common.c:173).
```
pc is 0.

Is this perhaps a function pointer that is zero? See the screenshot below.

[image]

If my analysis is correct, this is a double bug report:
- Hardfault handling is broken
- There is a bug in WICED causing the hardfaults

DaliborFarny, 2021-12-07:
We also observe that bug on our side, it appeared when we started using 2.0.0. It never happened when using 1.4.4, we have a fleet of 1000 devices, so we would notice it.

No1089, 2021-12-07:
Hi folks,

Thank you for the tickets submitted. The bug reports will be dealt with via the tickets.

closed 2021-12-07 by No1089

DaliborFarny, 2021-12-07:
> Hi folks,
> 
> Thank you for the tickets submitted. The bug reports will be dealt with via the tickets.

This is a serious bug, please dont underestimate it. We don’t need personal communication over tickets, we need someone to delete that “=“ sign in the code and release it again, so we have our photons running normally again.

avtolstoy, 2021-12-07:
@DaliborFarny Just want to clarify that the change/issue being talked about here is 3.1- and 3.2-specific and as you mentioned you are on LTS.

DaliborFarny, 2021-12-07:
> 

@avtolstoy the issue is not specific to 3.X. I am sure it was not present in 1.4.4, then we switched to LTS 2.0.0 (straight from 1.4.4) and this is where we started having the problems. We tested 2.1, 2.0, and 3.0, the problem was on all these versions. I did A/B testing with 1.4.4 (without modifying our code) and concluded that the problem only happen from 2.0 and above. Not on 1.4.4. The Wifi was not bulletproof on 1.4.4 either, but let's first solve this issue. Thank you!

elcojacobs, 2021-12-07:
The hardfault bug was first merged in 3.1.0, so if you had a freeze before that version it had a different cause.

The wiced line causing the hardfault has been around for a long time, but before 3.1.0 it would sos.

DaliborFarny, 2021-12-07:
> The hardfault bug was first merged in 3.1.0, so if you had a freeze before that version it had a different cause.
> 
> The wiced line causing the hardfault has been around for a long time, but before 3.1.0 it would sos.

OK, there must be another issue. But let's get this fixed and try the latest version.